adafruit/Adafruit_Blinka #6: No obvious workaround for missing descriptor support

cefn, 2018-02-25:
There's a fundamental contrast in the build flags between CircuitPython and Micropython seen here...

At https://github.com/micropython/micropython/blob/6e675c1baaf0de0d56a2345376d6b5600bfab3aa/py/mpconfig.h the default is set, which applies to official pyboard and esp8266 builds. 

```
// Whether to support descriptors (__get__ and __set__)
// This costs some code size and makes all load attrs and store attrs slow
#ifndef MICROPY_PY_DESCRIPTORS
#define MICROPY_PY_DESCRIPTORS (0)
#endif
```

Versus the build flags for esp8266 and atmel-samd in CircuitPython

```
cefn@cefn-artful-thinkpad:~/Documents/shrimping/clients/adafruit/git/circuitpython-latest$ grep -rin MICROPY_PY_DESCRIPTORS .
./ports/atmel-samd/mpconfigport.h:56:#define MICROPY_PY_DESCRIPTORS      (1)
./ports/esp8266/mpconfigport.h:43:#define MICROPY_PY_DESCRIPTORS      (1)
``` 

While perhaps an average user wouldn't be reliant on descriptors, some libraries use them or encourage their use,
 for example [adafruit_register](https://github.com/adafruit/Adafruit_CircuitPython_Register/blob/f86b4549c1f8ec10682cb5e5148693dc29c85602/README.rst) fundamentally depends on descriptors, meaning that e.g. the BNO055 library and RFM69 libraries intended to be used in https://github.com/adafruit/Adafruit_Micropython_Blinka/issues/2 cannot be made to work seamlessly on Micropython simply by making compatible I2C objects available. 

The list of libraries affected in this way includes at least...

* adafruit_register
* adafruit_bno055
* adafruit_tlc59711
* adafruit_rfm69.py
* adafruit_thermal_printer

This list is based on searching for libraries using __get__() or __set__() within the Adafruit CircuitPython Bundle.

Since it doesn't trigger a load-time error, it may be feasible to monkey-patch classes such as BNO055  with property accessors to hide the descriptor logic, a step performed before using any of the constructors. However, this step would have to be explicit in any Micropython-hosted use of said CircuitPython libraries, or a separate bundle could be distributed in which descriptor logic has been substituted.

tannewt, 2018-02-26:
I don't think we should change our libraries for this. Data descriptors simplify the drivers a bunch.

Would you mind opening an issue in MicroPython's repo for them to enable it? They should be interested in having our libraries compatible. I can follow up on it as well.

tannewt, 2018-02-27:
I think opening an issue is still worth while. That decision was made before we had 50+ libraries. :-) Determining the code size increase would be a good piece of info to have too.

What would they switch the descriptors to? Properties are implemented using data descriptors. 

Frozen modules won't be an issue because we're assuming its a MicroPython build. mpy files are identical to py files after load.

cefn, 2018-02-27:
> What would they switch the descriptors to? Properties are implemented using data descriptors.

I'm just picking up on the https://github.com/micropython/micropython/wiki/Differences comment; 
"So far, we were able to implement all native Python features (like properties) without explicit descriptors."

If it's not been covered again recently I'll open an issue.

cefn, 2018-02-27:
Ancient discussion of strategy for implementing properties at https://github.com/micropython/micropython/issues/474 and thought I could trace feature activation across current builds with https://github.com/micropython/micropython/search?utf8=%E2%9C%93&q=MICROPY_PY_BUILTINS_PROPERTY&type= Seems to show ESP8266 and ESP32 activating support. Can't trace the flag being activated for Pyboard. There _is_ property support on the Pyboard build, I just checked, so I must be on the wrong track.

cefn, 2018-02-27:
You can find the issue on enabling descriptor support on official images as recently raised at https://github.com/micropython/micropython/issues?utf8=%E2%9C%93&q=is%3Aissue+descriptor

tannewt, 2018-02-27:
Upstream issue is https://github.com/micropython/micropython/issues/3644 and I've subscribed to it. Thanks!

makermelissa, 2023-09-13:
Closing as it appears descriptors were added.

closed 2023-09-13 by makermelissa