adafruit/Adafruit_Blinka #7: Monkey patching before imports from Adafruit_CircuitPython_XX libraries

cefn, 2018-02-26:
Raising this as an issue to provide a place for discussion, although I have a stable workaround for now, it won't be the first time I'm sure.

When attempting to test the blinka compatibility layer against the adafruit_gps circuitpython module I found that it relied on circuitpython's differently conformant subset of the time functionality, which is not available in the micropython time module. Given the library imported **time** directly, it was necessary to monkey-patch a compatible version of time into **sys.modules['time']** before running the adafruit_gps import, to try and preserve code-for-code functionality without having to modify the adafruit_gps module itself.

The approach is illustrated by the WIP version patch_system() function at https://github.com/cefn/Adafruit_Micropython_Blinka/blob/b76e7eb3aca5c7b91f9026a37b46e7ca50c3a350/src/adafruit_blinka/__init__.py#L58

Using it requires that the author runs...

```
import adafruit_blinka
adafruit_blinka.patch_system()
```

...before attempting to import any library which might depend on the modules patched.

However, it may be preferable to selectively patch named modules (rather than the RAM overhead of attempt to patch everything which modules **might** later need), or something else entirely.

tannewt, 2018-02-26:
Whats the specific problem? Is it `time.monotonic`?

Is it possible to always do the monkey patch on import of blinka?

cefn, 2018-02-27:
Definitely an option, but tradeoffs.

Other than this monkey-patch, you don't have to import adafruit_blinka in your source at all, as it is seamlessly imported by the circuitpython-compatibility modules (digitalio etc.). You don't need to know when in the sequence to import it, and you can't really get it wrong. 

However, once there is a need to monkey-patch, you need to be sure you have explicitly called on adafruit_blinka before any other imports might access monkey-patched modules. Waiting on import from a dependent library is likely too late in the sequence. So this issue will need to be documented fairly well and can't be implicit. 

Once we are committed that it's not implicit and authors have to trigger the monkey-patch, we have the freedom to expose explicit and selective monkey-patching. This helps us avoid automatically and monolithically load all the compatibility implementations you aren't using the moment you use any adafruit_blinka at all. This could be critical given the very low memory ceilings on the target platforms.

For example we could advise something like...

```
from adafruit_blinka import compatibility
compatibility("time", "adafruit_register")
import time
import adafruit_register
```

As regards time, I needed time.monotonic() since utime.ticks_ms() and friends couldn't be cross-platform, and adafruit_gps needed struct_time. See https://github.com/cefn/Adafruit_Micropython_Blinka/blob/master/src/adafruit_blinka/agnostic/time.py for a quick hack which worked for adafruit_gps and for [testing#await_true()](https://github.com/cefn/Adafruit_Micropython_Blinka/blob/master/test/src/testing/__init__.py#L37) at least.

tannewt, 2018-02-27:
Your example looks reasonable to me. Monkey-patching is ok by me on an as-needed basis.

Thanks for the in-depth explanation.