adafruit/Adafruit_Blinka #9: Conformance Issues as Tests

cefn, 2018-03-06:
This is an invitation for @tannewt and others to comment on any compatibility issues they observe from examining the https://github.com/adafruit/Adafruit_Micropython_Blinka repo, or indeed assertions they feel should be documented permanently in the form of tests.

So far, the testing framework has been minimally populated with integration tests checking that whole 'applications' can be made to run.

The aim of this issue thread is to be able to comment briefly on lower-granularity issues between the real and Blinka-simulated CircuitPython API, so that (passing or failing) test cases can be authored to improve conformance or prevent regression.

These will typically show up as a method in a new or existing unittest.TestCase class e.g. under https://github.com/adafruit/Adafruit_Micropython_Blinka/tree/master/test/src/testing/universal

tannewt, 2018-03-06:
I don't know of any cases currently. I tend to follow the rule of adding tests as bugs are found. Our on-board testing is currently broken though which makes it difficult.

cefn, 2018-03-07:
Thanks @tannewt great, feel free to add any number of comments below as and when things crop up, and I'll try to match them to tests and commit fixes as we go.

makermelissa, 2023-09-13:
Closing since unit testing isn't actively being used.

closed 2023-09-13 by makermelissa