adafruit/Adafruit_Blinka #15: install failed on Jessie

jerryneedell, 2018-06-30:
Just a heads up - I had no problems installing on a RPI 3B+ running latest version of "stretch"
When I tried to install on an updated RPi Zerow running "Jessie"  it failed with
```
pi@gjnpi0-1:~ $ pip3 install adafruit-blinka
Downloading/unpacking adafruit-blinka
  Downloading Adafruit-Blinka-0.1.7.tar.gz
  Running setup.py (path:/tmp/pip-build-epf581_d/adafruit-blinka/setup.py) egg_info for package adafruit-blinka
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'long_description_content_type'
      warnings.warn(msg)
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/tmp/pip-build-epf581_d/adafruit-blinka/setup.py", line 44, in <module>
        'Programming Language :: Python :: Implementation :: MicroPython',
      File "/usr/lib/python3.4/distutils/core.py", line 108, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 266, in __init__
        _Distribution.__init__(self,attrs)
      File "/usr/lib/python3.4/distutils/dist.py", line 280, in __init__
        self.finalize_options()
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 301, in finalize_options
        ep.load()(self, ep.name, value)
      File "/tmp/pip-build-epf581_d/adafruit-blinka/setuptools_scm-2.1.0-py3.4.egg/setuptools_scm/integration.py", line 10, in version_keyword
      File "/tmp/pip-build-epf581_d/adafruit-blinka/setuptools_scm-2.1.0-py3.4.egg/setuptools_scm/version.py", line 44, in _warn_if_setuptools_outdated
    setuptools_scm.version.SetuptoolsOutdatedWarning: your setuptools is too old (<12)
    Complete output from command python setup.py egg_info:
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'long_description_content_type'

  warnings.warn(msg)

/usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'python_requires'

  warnings.warn(msg)

Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/tmp/pip-build-epf581_d/adafruit-blinka/setup.py", line 44, in <module>

    'Programming Language :: Python :: Implementation :: MicroPython',

  File "/usr/lib/python3.4/distutils/core.py", line 108, in setup

    _setup_distribution = dist = klass(attrs)

  File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 266, in __init__

    _Distribution.__init__(self,attrs)

  File "/usr/lib/python3.4/distutils/dist.py", line 280, in __init__

    self.finalize_options()

  File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 301, in finalize_options

    ep.load()(self, ep.name, value)

  File "/tmp/pip-build-epf581_d/adafruit-blinka/setuptools_scm-2.1.0-py3.4.egg/setuptools_scm/integration.py", line 10, in version_keyword

  File "/tmp/pip-build-epf581_d/adafruit-blinka/setuptools_scm-2.1.0-py3.4.egg/setuptools_scm/version.py", line 44, in _warn_if_setuptools_outdated

setuptools_scm.version.SetuptoolsOutdatedWarning: your setuptools is too old (<12)

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip-build-epf581_d/adafruit-blinka
Storing debug log for failure in /home/pi/.pip/pip.log
```

Not sure i I'm just missing something or if it is a problem with "Jessie"
I tried ```
pip3 install --upgrade adafruit-blinks
```
and it failed the same way.

ladyada, 2018-06-30:
looks like your python/setuptools is too old? try "sudo apt-get update" then "sudo apt-get upgrade" you'll still have jessie but all your packages will be updated/upgraded

jerryneedell, 2018-06-30:
I did the update/upgrade and reboot just before trying the install.

jerryneedell, 2018-06-30:
I was able to workaround this by
```
sudo pip3 install --upgrade setuptools
sudo pip3 install adafruit-gthub
```

I did not need sudo on the stretch system - but I had permission error on the jessie system

ladyada, 2018-06-30:
not sure what this line ought to be? "sudo pip3 install adafruit-gthub"

jerryneedell, 2018-06-30:
oops
 ```
sudo pip3 install --upgrade setuptools
sudo pip3 install adafruit-blinka
```

ladyada, 2018-06-30:
ok added to guide!

closed 2018-06-30 by ladyada

dukuilanu, 2019-12-22:
Hi, this was the top result on Google for errors installing adafruit-blinka on Jessie with Python2 as the system default. In installing Python3, the distro's pip3 itself also needed upgrading beyond what the repo had:
pip3 install --upgrade setuptools
pip3 install --upgrade pip
pip install adafruit-blinka

and because I then ran into this:
ERROR: adafruit-blinka 3.3.4 requires sysv_ipc, which is not installed.

pip install  sysv_ipc
pip install spidev

Don't ask me what I've done to this thing to require those last 2, but I hope the first trick can help someone.