adafruit/Adafruit_Blinka #17: spi does not accept "end=" keyword

jerryneedell, 2018-07-02:
@ladyada - a few Blinka notes, first of all - it is great! -  I found that I can try CP module without waiting for them to be release in PyPi  just by putting the driver under test in the local folder with the test code. I did this with the mcp9808, apds996 and the rfm9x (with changes as noted below).
I have been trying out the rfm9x driver under blinka and had to make a few changes, but it is now working - the main change was I had to revise several of the spi readinto/write calls because they used "end=" and that is not supported via blinka. The changes were pretty simple and it all seems to be woking. I can send/receive  from my RPi to a feather M0 express with CP 3.0.  
Is the "end=" keyword "non-standard" and not going to be available on the RPi. Would you like me to submit a PR to remove it from the existing rfm9x driver for CP to keep them compatible?

Have you run into this with other drivers? I'm not sure how widespread the use of end= is in the CP drivers.

ladyada, 2018-07-02:
SPI is definitely not fully done, you shouldnt adapt drivers to match Blinka as its missing a bunch of stuff and vargs. its just 'barely working' right now, i only tested it with one sensor so far :)

jerryneedell, 2018-07-02:
Fair enough! I was not sure if spidev could handle the "end=".
It is fun to have it working as much as it is!  Thanks for adding the note to the guide about not using ce0/ce1 on the RPi -- that caused some very odd results!

ladyada, 2018-07-02:
ok cool! re-opening as a reminder to myself to fix the end= thing

Timonzimm, 2018-07-13:
I have the same issue and if I would love to have details on how you dealt with this @jerryneedell. Indeed, I tried to remove the "end" argument from the function call and changed the range of buffer.

So instead of having `buf=[a,b,c]` and `end=2`, I now have `buf=[a,b]`.

Thank you for your help!

jerryneedell, 2018-07-13:
@Timonzimm As noted by @ladyada this is likely to be fixed properly in a future update to the Blinka module so the preferred solution is to wait for that!

What ti did was to create additional BUFFER's to hold 1 or 2 byte arguments and use them instead of the end= keyword.  For your reference, here is the "dif"  but be forewarned, I have done very little testing of this and it may not work reliably.

```
$ diff adafruit_rfm9x.py ../Adafruit_CircuitPython_RFM9x/adafruit_rfm9x.py  > adafruit_rfm9x.diff
$ cat adafruit_rfm9x.diff 
270,271d269
<     _BUFFER1 = bytearray(1)
<     _BUFFER2 = bytearray(2)
394c392
<             self._BUFFER1[0] = address & 0x7F  # Strip out top bit to set 0
---
>             self._BUFFER[0] = address & 0x7F  # Strip out top bit to set 0
396,397c394,395
<             device.write(self._BUFFER1)
<             device.readinto(buf)
---
>             device.write(self._BUFFER, end=1)
>             device.readinto(buf, end=length)
401,402c399,400
<         self._read_into(address, self._BUFFER1)
<         return self._BUFFER1[0]
---
>         self._read_into(address, self._BUFFER, length=1)
>         return self._BUFFER[0]
404c402
<     def _write_from(self, address, msg, length=None):
---
>     def _write_from(self, address, buf, length=None):
407a406,407
>         if length is None:
>             length = len(buf)
409,412c409,412
<             self._BUFFER1[0] = (address | 0x80) & 0xFF  # Set top bit to 1 to
<                                               # indicate a write.
<             device.write(self._BUFFER1)
<             device.write(msg)
---
>             self._BUFFER[0] = (address | 0x80) & 0xFF  # Set top bit to 1 to
>                                                        # indicate a write.
>             device.write(self._BUFFER, end=1)
>             device.write(buf, end=length)
418c418
<             self._BUFFER2[0] = (address | 0x80) & 0xFF  # Set top bit to 1 to
---
>             self._BUFFER[0] = (address | 0x80) & 0xFF  # Set top bit to 1 to
420,421c420,421
<             self._BUFFER2[1] = val & 0xFF
<             device.write(self._BUFFER2)
---
>             self._BUFFER[1] = val & 0xFF
>             device.write(self._BUFFER, end=2)
558c558
<         # Write payload
---
>         # Write payload.

```

ladyada, 2018-07-13:
yep ive just been super busy!

jerryneedell, 2018-07-13:
can't imagine why ;-)

closed 2018-08-12 by ladyada