adafruit/Adafruit_Blinka #23: uart DECISION time

ladyada, 2018-08-12:
its that time, where things must be Decided!

uarts in modern OS are kinda special, because they always get attached to a /dev/ttyUSB type device. raspi does have RX/TX pins, they're not often used because they're the console pins too... 

im tempted to just make busio.uart be initialized with a path string rather than two pin names. why? because then the example code will be universally usable on mac/linux/windows (!) which means that some devices like fingerprint, gps, bno055, and others with serial i/o will be available on all OSes

[BBB does have multiple uarts on pins,](http://beaglebone.cameon.net/home/serial-ports-uart) so we could *also* support passing in the RX/TX pin-names but i think that since we *have* to use paths, we could just stick to only path.

@dhalbert @tannewt & others - thoughts?

kattni, 2018-08-12:
Seems like having both options could be good. Universally usable code is brilliant, but the option to do it like it's done now also super handy. I don't know if this creates more overhead in the long run though. Having the option to specify UART pins feels like something people will inevitably want.

ladyada, 2018-08-12:
the thing that will bite people with RX/TX is that you need to map that to the UART name-path ... and that changes sometimes when OS's change :/

tannewt, 2018-08-13:
We chatted about this earlier and decided to try passing in a Serial object from pyserial instead of adding a busio.UART. The drivers will hopefully just work with it.

ladyada, 2018-08-13:
@kattni wanna take a look at both our UART and the official 'pyserial' api, see if it matches up? we'll of course try with real code but take a glance! :)

kattni, 2018-08-13:
As long as the options work the same, pyserial does everything our UART does. pyserial does way more than our UART does but all the same options are there.

ladyada, 2018-08-14:
well, the API is the same, but encoding is Fun - here's the diff i had to make to adafruit_gps, string literals needed a b' prefix. variables needed a bytes(var, encoding="UTF-8")

beyond this it 'just worked' tho

```
 adafruit_gps.py | 10 +++++-----
 1 file changed, 5 insertions(+), 5 deletions(-)

diff --git a/adafruit_gps.py b/adafruit_gps.py
index 66d0eee..3086840 100644
--- a/adafruit_gps.py
+++ b/adafruit_gps.py
@@ -115,15 +115,15 @@ class GPS:
         Note you should NOT add the leading $ and trailing * to the command
         as they will automatically be added!
         """
-        self._uart.write('$')
-        self._uart.write(command)
+        self._uart.write(b'$')
+        self._uart.write(bytes(command, encoding="UTF-8"))
         if add_checksum:
             checksum = 0
             for char in command:
                 checksum ^= ord(char)
-            self._uart.write('*')
-            self._uart.write('{:02x}'.format(checksum).upper())
-        self._uart.write('\r\n')
+            self._uart.write(b'*')
+            self._uart.write(bytes('{:02x}'.format(checksum).upper(), encoding="UTF-8"))
+        self._uart.write(b'\r\n')
 
     @property
     def has_fix(self):

```

ladyada, 2018-08-14:
circuitpython does not seem to support bytes w/encoding (brennen tried this for me)

@tannewt any ideas?

```
>>> bytes("Hello world", encoding="UTF-8")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: wrong number of arguments
```

tannewt, 2018-08-14:
For the record, I think `"Hello world".encode("UTF-8")` should work. This is risky however because you actually want ascii bytes instead of encoded unicode. So, I recommend using `b"Hello world"` instead.

ladyada, 2018-08-14:
ok we resolved this over hipchat, we'll use b' as much as possible and then .encode() when we can't

ladyada, 2018-08-19:
turns out encode() doesnt work in CircuitPython. the correct universal technique is

bytes("hello world", "utf-8") 

going thru libraries now, no changes needed to underlying circuitpython busio device but might be good to restrict circuitpython busio.uart to bytes only, rather than letting strings slide thru (in 4.0?) - lemme know if you agree and i can open an issue

closed 2018-08-19 by ladyada