adafruit/Adafruit_Blinka #24: Allow bitbang SPI

ladyada, 2018-08-17:
there's limited SPI ports on the Pi, and some chips dont need high speed, lets add code for (fast) bitbang support!

tgikal, 2019-01-02:
Was this implemented yet?
I see the bitbangio.SPI function exists, but when trying to create it I get a "from machine import SPI -> ImportError: No module named 'machine' "
Currently, I'm pretty dependent on software SPI for my slower modules with the rpi.

ladyada, 2019-01-02:
not implemented, we probably will add it as a totally separate library, not even in Blinka, but as a pip'able add-on that is `spi` compatible. wanna help? :)

dastels, 2019-01-25:
Yay!

```lindon:>Python 3.5.3 (default, Sep 27 2018, 17:25:39)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> Testing BME280 BBSPI interface
24.36
>>>
```

ladyada, 2019-01-25:
epic

ladyada, 2020-04-21:
https://github.com/adafruit/Adafruit_CircuitPython_BitbangIO

closed 2020-04-21 by ladyada