adafruit/Adafruit_Blinka #37: Add PWMOut to RasPi Blinka

brentru, 2018-09-13:
Using the RPi.GPIO library (https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/raspi_23/pin.py#L1), we could add PWM functionality similar to PWMOut (https://circuitpython.readthedocs.io/en/2.x/shared-bindings/pulseio/PWMOut.html)  for the Raspberry Pi.

Initialization:
```
pwm = GPIO.PWM(pin, frequency)
pwm.start(duty_cycle) 
```

`variable_frequency` -> `ChangeFrequency`
`ChangeDutyCycle` -> `duty_cycle`

- https://sourceforge.net/p/raspberry-gpio-python/wiki/PWM/

ladyada, 2019-05-12:
i did this but for sysfs pwm only - not 'soft' PWM like gpiozero does. still, its nice on boards that have hardware PWM!

makermelissa, 2020-06-22:
This would be handy for displayio backlights.

ladyada, 2020-06-22:
@makermelissa its non trivial, please use backlight on/off for now

makermelissa, 2020-06-22:
Yeah, that's what I'm doing. :)

PR #314 (2020-07-01): Added PWMOut functionality for the Raspberry Pi
Fixes #37. Tested using D26 (a GPIO without built-in PWM) using this script https://learn.adafruit.com/circuitpython-on-google-coral-linux-blinka/pwm-outputs-servos.
closed 2020-07-01 by makermelissa