adafruit/Adafruit_Blinka #43: migrate GPIO usage to libgpiod Python bindings

pdp7, 2018-10-22:
RPi.GPIO and Adafruit_BBIO both use the legacy /sys interface for the Linux kernel GPIO subsystem.

[libgpiod](https://github.com/brgl/libgpiod/) is a userspace library by @brgl that use the new character device based interface for Linux kernel GPIO subsystem.

The sysfs interface for GPIO is marked as deprecated in the Linux kernel, so I believe this Blinka library should migrate to the [libgpiod Python bindings](https://github.com/brgl/libgpiod/tree/master/bindings/python/examples).

ladyada, 2018-10-22:
that would be awesome - its definitely something we'd like to do (and libgpiod is a lil faster too)

ladyada, 2018-12-02:
done here https://github.com/adafruit/Adafruit_Blinka/pull/59
turns out there's no pullup support, which is makes it a poor substitute for well-written GPIO bindings, but its good for poorly-supported chips w/only sysfs
anywyas for now i just tell people to run the script on import failure. i think thats fine until libgpiod is shipped with

ladyada, 2018-12-03:
is in generic_linux

closed 2018-12-03 by ladyada

theyoyojo, 2020-06-16:
Hey, I've been working on a drop-in replacement for RPi.GPIO that uses libgpiod's python bindings as the underlying GPIO access mechanism instead of /dev/mem or the sysfs interface.

I see that this issue was closed long ago but I figured I'd mention this library's existence in case anyone here finds it useful.

Pending my latest pull request the library is feature complete if not very close to it, i.e. it completely re-implements the RPi.GPIO API as of RPi.GPIO version 0.7.0

Here is the repo: https://github.com/underground-software/python3-libgpiod-rpi