adafruit/Adafruit_Blinka #44: neopixel_write function definition not working

bricedenicedu69, 2018-10-30:
Hi everyone,
I am working on a project which consists in putting some colors on my LED. They are quite similar to WS2812 devices but I was thinking about using the neopixel module given. However, it looks like I also need the neopixel_write function imported from the esp module (which I cannot import as well, so I've been looking for the esp.py file which is missing). Thus the neopixel_write function definition is missing and I cannot use it. Does anyone know where I can find it ? I found this : https://github.com/adafruit/Adafruit_Blinka/blob/master/src/neopixel_write.py (I ignore everything above the function definition) but I run my main file, it is not working because the _neopixel is not defined and I have no idea what it stands for. 
I am using Atom as IDE and working with a Pycom/Sipy device. Thanks for your help, I can provide you with further information if needed.

TheKitty, 2018-10-30:
If you want specific technical support on Adafruit sold products, you should go to https://forums.adafruit.com/, if you have questions on your specific pixel type, go to the manufacturer and for ESP issues, see the Expressif forums. GitHub repo issues are not for technical support implementation questions.

closed 2018-10-30 by TheKitty

ladyada, 2018-10-30:
brice, this repo is just a wrapper for linux based implementations of neopixel_write which are low leve
if you've got pycom device, they'll be able to give you their implementation of neopixel bitbanging, which we don't know anything about :)