adafruit/Adafruit_Blinka #49: busio I2C import on RPi

magnus-haw, 2018-11-26:
Line 21 in busio.py should have an "elif" statement instead of an "if" statement. Currently, if the board is not a beaglebone_black, the code attempts to execute the last else statement. This creates an error for Raspberry pi boards which do not use the Micropython machine module.

Discovered this issue when trying to install CircuitPython from this tutorial: https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi

def init(self, scl, sda, frequency):
        self.deinit()
        **if** board_id == "raspi_3" or board_id == "raspi_2":
            from adafruit_blinka.microcontroller.raspi_23.i2c import I2C as _I2C
        **if** board_id == "beaglebone_black":
            from adafruit_blinka.microcontroller.raspi_23.i2c import I2C as _I2C
        **else**:
            from machine import I2C as _I2C

->>>
def init(self, scl, sda, frequency):
        self.deinit()
        **if** board_id == "raspi_3" or board_id == "raspi_2":
            from adafruit_blinka.microcontroller.raspi_23.i2c import I2C as _I2C
        **elif** board_id == "beaglebone_black":
            from adafruit_blinka.microcontroller.raspi_23.i2c import I2C as _I2C
        **else**:
            from machine import I2C as _I2C

gjoseph, 2018-11-26:
Found the same thing @magnus-haw - here's a PR that I hope will get merged soon :)

asidroid, 2018-11-26:
@magnus-haw Thanks!

ladyada, 2018-11-26:
we're about to fix this with a different PR, hold tight!

ladyada, 2018-11-26:
apologies for the break. lots of dev goin' on 
please try 0.3.1 - i tested it but it could use more testin' as always :)

csylvain, 2018-11-26:
that (0.3.1) broke something (0.3.0 worked after changing that one "if" to "elif")
Traceback (most recent call last):
  File "bme280_simpletest.py", line 8, in <module>
    i2c = busio.I2C(board.SCL, board.SDA)
  File "/home/pi/.local/lib/python3.5/site-packages/busio.py", line 15, in __init__
    self.init(scl, sda, frequency)
  File "/home/pi/.local/lib/python3.5/site-packages/busio.py", line 20, in init
    from adafruit_blinka.microcontroller.generic_linux.i2c import I2C as _I2C
ImportError: No module named 'adafruit_blinka.microcontroller.generic_linux'

ladyada, 2018-11-26:
hmm - lemme take another look

ladyada, 2018-11-26:
OK please try 0.3.2!

ladyada, 2018-11-26:
i think this is resolved now, @brennen  tested it. closing - please reopen if it's still an issue

closed 2018-11-26 by ladyada

csylvain, 2018-11-26:
no pin.py in generic_linux ? only in raspi_23.

ladyada, 2018-11-26:
thats correct, we have not done our libgpiod changes so pin control is still board specific

csylvain, 2018-11-26:
k!