adafruit/Adafruit_Blinka #53: RPI1B incorrectly identified as 3

permezel, 2018-11-27:
I am trying to use the blinka on an RPI1B.
It identifies it as a 3.  Accordingly, the i2c bus is identified as 1, not 0.
```
platform: 1
version: 3
board_id: raspi_3
Hello blinka!
Digital IO ok!
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 20, in __init__
    self._i2c_bus = smbus.SMBus(bus_num)
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_PureIO/smbus.py", line 105, in __init__
    self.open(bus)
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_PureIO/smbus.py", line 130, in open
    self._device = open('/dev/i2c-{0}'.format(bus), 'r+b', buffering=0)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'
```

```
+------------------| |--| |------+
| ooooooooooooo P1 |C|  |A|      |
| 1oooooooooooo    +-+  +-+      |
|                                |
|                +---+          +====
|                |SoC|          | USB
|   |D| Pi Model +---+          +====
|   |S| B  V1.0                  |
|   |I|                  |C|+======
|                        |S||   Net
|                        |I|+======
=pwr             |HDMI|          |
+----------------|    |----------+

Revision           : 0002
SoC                : BCM2835
RAM                : 256Mb
Storage            : SD
USB ports          : 2 (excluding power)
Ethernet ports     : 1
Wi-fi              : False
Bluetooth          : False
Camera ports (CSI) : 1
Display ports (DSI): 1

P1:
   3V3  (1) (2)  5V    
 GPIO0  (3) (4)  5V    
 GPIO1  (5) (6)  GND   
 GPIO4  (7) (8)  GPIO14
   GND  (9) (10) GPIO15
GPIO17 (11) (12) GPIO18
GPIO21 (13) (14) GND   
GPIO22 (15) (16) GPIO23
   3V3 (17) (18) GPIO24
GPIO10 (19) (20) GND   
 GPIO9 (21) (22) GPIO25
GPIO11 (23) (24) GPIO8 
   GND (25) (26) GPIO7 

```

ladyada, 2018-11-27:
yeah the RPI1B isn't supported at this time, and the pinmap is all wrong because of that. if you're feeling up for adding a pin definition file for the pi 1, that would be awesome, and we'd make sure its detected right!

ladyada, 2018-12-30:
hiya try the 1.1.0 version of this library and Adafruit-PlatformDetect v0.0.8, it should now work with raspi 1's!

closed 2018-12-30 by ladyada