adafruit/Adafruit_Blinka #56: Pi RGBW NeoPixel support incomplete?

waded, 2018-11-30:
I was working through https://learn.adafruit.com/neopixels-on-raspberry-pi/python-usage, adding https://learn.adafruit.com/circuitpython-essentials/circuitpython-neopix to understand how to add the W component (using bpp=4, and ), with a Raspberry Pi Zero W and the RGBW Neopixel strip.

It didn't seem to work correctly. The result when sending (255, 0, 0, 0) down the strip was a repeating pattern, not the desired solid color.

I took a look at the code and eventually got to 1f9263a050a781740d3fda52a632048b64912971, which implies RGBW is supported, but it didn't appear the W bits were being shifted in for the call to `ws.ws2811_led_set`, which would explain the pattern.

waded, 2018-11-30:
@ladyada Unfortunately I don't have any RGBs to test. I'll edit issue to add example code and the resulting color sequence here in a sec.

waded, 2018-11-30:
I will add that I did try rpi_ws281x's Python examples, notably SK6812_strandtest.py, which does solid color fills, including white, and that was working as expected.

ladyada, 2018-11-30:
oh ok you know i know what happened, i can fix it

caternuson, 2018-11-30:
does this need updating to sync up with latest RGBW tweaks?
https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/raspi_23/neopixel.py#L71

ladyada, 2018-11-30:
no i think its just the way the raspi neopixel driver expects data, if i set it to RGBW i think i have to shift data in differently. im getting an RGBW ring to do a proper test

waded, 2018-11-30:
I'd be happy to test a fix on my strip when you have one. I prototyped a fix (changed the ` # assign all colors!` block to also be based on `LED_STRIP` and shift the w value into `pixel` in the RGBW case), but I don't have a way to test anything but.

ladyada, 2018-12-01:
i believe its now fixed in a PR

closed 2018-12-03 by ladyada