adafruit/Adafruit_Blinka #60: use circuitpython on generic linux platform with FT232H

oshadmon, 2018-12-03:
**Machine Specs**: Ubuntu 18.04 Desktop on an i7 based workstation.  

**Background**: I'm using an Adafruit HTU21D-F Humidity/Temp sensor connected to a USB pluggable FT232H breakout board (C232HM-DDHSL‌-0 USB2.0/I2C cable). Adafruit GPIO has support for FT232H (https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/FT232H.py) 

**Question**: I want to use circuitpython, in particular, the HTU21D-F code (https://github.com/adafruit/Adafruit_CircuitPython_HTU21D/blob/master/adafruit_htu21d.py) but I don't have a supported board as I'm using a linux desktop computer. What are my options? Can I use circuitpython? Is there any documentation regarding this type of use? 

I realize this may be the wrong place to post this, if so where should I post it? Who should I consult? (Thanks!)

caternuson, 2019-08-21:
So would the idea here be to provide something that could be a drop in replacement for `busio`? Ex:
```python
import adafruit_ft232h
i2c = adafruit_ft232h.I2C()
sensor = some_sensor.Sensor(i2c)
```
or even
```
import adafruit_ft232h as busio
# proceed as usual
```

ladyada, 2019-08-21:
yes, OR we could set the environmental var to 'force' FT232H usage (which i did add) and then code is identical

ladyada, 2019-09-21:
added!

ladyada, 2019-09-21:
guide coming soon...

closed 2019-09-21 by ladyada