adafruit/Adafruit_Blinka #65: Support for ODroid C1/C1+

crazyquark, 2018-12-20:
Wiring PI and RPi.GPIO are supported on this board; unfortunately this package mistakenly detects the board as a BeagleBone black.

crazyquark, 2018-12-20:
Potential workaround:
`from adafruit_blinka import agnostic
agnostic.board_id = 'raspi_23'`

ladyada, 2018-12-20:
hiya we're doing a massive refactor like, this very hour :) check the PR's here and in platformdetect 
adding odroid c1 detection to platformdetect is the best you can do now, while we rework this repo
https://github.com/adafruit/Adafruit_Python_PlatformDetect

ladyada, 2018-12-20:
also, we dont recommend wiring pi - its ususally sysfs underneath, very very slow. we use libgpiod instead - its fast and supported in modern kernels :)

crazyquark, 2018-12-21:
Nice! btw, I got it to work on Odroid C1+ with the above hack.

ladyada, 2018-12-21:
wow cool! yes it will work but its a bit of a hack if RPi.GPIO exists :) however there's some pinout differences and its best if we can properly detect it and make a correct 'map' of all the pins and functions - especially since htere's some cool stuff like ADC pins! and some pins have different names than RasPi

http://pi4j.com/pins/odroid-c1.html

could you start by adding detection to platform detect? we have one so we could help if you need assistance :)

https://github.com/adafruit/Adafruit_Python_PlatformDetect/

crazyquark, 2018-12-22:
I'll try to help. Yes, the pinout is slightly different and yes I use a RPi.GPIO port for odroid:
https://wiki.odroid.com/odroid-c1/application_note/gpio/rpi.gpio
So far it has worked for me.

crazyquark, 2018-12-22:
Made a PR to detect the Odroid C1/C2:
https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/7