adafruit/Adafruit_Blinka #66: Board not supported exception

rake36, 2018-12-24:
Hi, I'm just trying to test the DC-Stepper Motor HAT following the online instructions located here: https://learn.adafruit.com/simple-raspberry-pi-robot/hardware

It's a Raspberry Pi 3 with Jessie that I've apt-get updated/upgraded.

I'm running a Python3 virtual env and adafruit-blinka and adafruit-circuitpython-motorkit pip3 installs worked without error.

```bash
(car1) pi@picar:~/car1 $ uname -m
armv7l
(car1) pi@picar:~/car1 $ pip list
Package                          Version
-------------------------------- -------
Adafruit-Blinka                  1.0.2
adafruit-circuitpython-busdevice 2.2.7
adafruit-circuitpython-motor     1.3.3
adafruit-circuitpython-motorkit  1.2.3
adafruit-circuitpython-pca9685   3.2.3
adafruit-circuitpython-register  1.3.2
Adafruit-PlatformDetect          0.0.7
pip                              18.1
rpi-ws281x                       4.1.0
RPi.GPIO                         0.6.5
setuptools                       40.6.3
spidev                           3.2
wheel                            0.32.3
```

However, I cannot run any of the examples to test the motors and am not experienced enough to understand how to debug.

```bash
(car1) pi@picar:~/car1 $ python3
Python 3.4.2 (default, Sep 26 2018, 07:16:01)
[GCC 4.9.2] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from adafruit_motorkit import MotorKit
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/car1/lib/python3.4/site-packages/adafruit_motorkit.py", line 59, in <module>
    import board
  File "/home/pi/car1/lib/python3.4/site-packages/board.py", line 59, in <module>
    raise NotImplementedError("Board not supported")
NotImplementedError: Board not supported
```

I've also run the tests for i2c and spi

https://learn.adafruit.com/adafruits-raspberry-pi-lesson-4-gpio-setup/configuring-i2c

```bash
(car1) pi@picar:~/car1 $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- --
(car1) pi@picar:~/car1 $ ls -l /dev/spidev*
crw-rw---- 1 root spi 153, 0 Dec 24 14:42 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Dec 24 14:42 /dev/spidev0.1
```
Any suggestions on how I could troubleshoot further? Appreciate your time.

ladyada, 2018-12-24:
please update the latest on all your packages in that pip list :) does that help

ladyada, 2018-12-24:
also, try running this script and telling us the output https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/bin/detect.py

rake36, 2018-12-25:
So without upgrading any packages, detect.py returns:

```bash
(car1) pi@picar:~/car1 $ python3 detect.py
Chip id:  None
Board id:  None
Is this a Pi 3B+? False
Is this a 40-pin Raspberry Pi? False
Is this a BBB? False
Is this an Orange Pi PC? False
Is this an embedded Linux system? False
Is this a generic Linux PC? False
```
I had noticed this before: that the Chip and Board id's were returning None, but wasn't sure what to make of it. The Pi appears to be fine to me: Wi-Fi is up, SSH works (which is how I'm interacting with it most of the time).

The motor HAT shows green power when I connect the 4 AA battery pack.

rake36, 2018-12-25:
> please update the latest on all your packages in that pip list :) does that help

None of the packages were candidates for upgrades. After running upgrade on all of them, I see this from pip3 freeze.

```bash
(car1) pi@picar:~/car1 $ cat freeze3b.txt
Adafruit-Blinka==1.0.2
adafruit-circuitpython-busdevice==2.2.7
adafruit-circuitpython-motor==1.3.3
adafruit-circuitpython-motorkit==1.2.3
adafruit-circuitpython-pca9685==3.2.3
adafruit-circuitpython-register==1.3.2
Adafruit-PlatformDetect==0.0.7
rpi-ws281x==4.1.0
RPi.GPIO==0.6.5
spidev==3.2
```

rake36, 2018-12-25:
I removed the Motor Hat and all other connectors from the Pi. Plugged in just the power to the Pi, connected via SSH, and ran detect.py again. Got same results as above.

Now that I can see the board and to clarify the model: Raspberry Pi 3 Model B V1.2.

ladyada, 2018-12-25:
??
whats the output of `cat /proc/cpuinfo`?

rake36, 2018-12-26:
> ??
> whats the output of `cat /proc/cpuinfo`?

```bash
pi@picar:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2709
Revision        : a22082
Serial          : 00000000570d0c67
```

ladyada, 2018-12-26:
oh right! ok well, we know what it is - if you want to fix right now, replace your `site-packages/adafruit_platformdetect/chip.py` with this file
https://github.com/ladyada/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/chip.py
or you can wait a couple days and we'll implement the fix!

rake36, 2018-12-26:
@ladyada, thanks so much! I really thought I was going insane.

After updating chip.py to include BCM2709 as per your fix, detect.py is working:

```bash
(car1) pi@picar:~/car1 $ python3 detect.py
Chip id:  BCM2XXX
Board id:  RASPBERRY_PI_3B
Is this a Pi 3B+? False
Is this a 40-pin Raspberry Pi? True
Is this a BBB? False
Is this an Orange Pi PC? False
Is this an embedded Linux system? True
Is this a generic Linux PC? False
Raspberry Pi detected.
```

ladyada, 2018-12-26:
~ awesome ~! good find :)

Fury1SOG, 2018-12-26:
Hello!
I am having a similar problem, except with a Raspberry Pi Zero WH with a v2 camera, TSL2561, and BME280 sensor installed.

Here is the error I receive when I try to run blinkatest.py:

```
pi@raspberrypi:~ $ python3 blinkatest.py
Traceback (most recent call last):
  File "blinkatest.py", line 1, in <module>
    import board
  File "/home/pi/.local/lib/python3.5/site-packages/board.py", line 59, in <module>
    raise NotImplementedError("Board not supported")
NotImplementedError: Board not supported
```

Here are the results of detect.py:

```
pi@raspberrypi:~ $ python3 detect.py
Chip id:  BCM2XXX
Board id:  None
Is this a Pi 3B+? False
Is this a 40-pin Raspberry Pi? False
Is this a BBB? False
Is this an Orange Pi PC? False
Is this an embedded Linux system? True
Is this a generic Linux PC? False
Raspberry Pi detected.
```

Here are the results of cpuinfo:

```
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 19000c1
Serial		: 00000000de78b44c
```

ladyada, 2018-12-26:
@fury are you overclocking?

Fury1SOG, 2018-12-26:
> @Fury are you overclocking?

I am using the v2 camera with RPI Cam Web Interface, and it was very unstable at the start. I did experiment with turbo mode and overvolting; I am currently only using the overvolting.

I guess the changed revision number is due to my experimenting with overclocking? 
I was able to get blinkatest to run by adding '19000c1' to line 159 in board.py:

`RASPBERRY_PI_ZERO_W: ('9000c1', '19000c1'),`

ladyada, 2018-12-27:
yeah that extra 1 means its overclocked/overvolted.
@brennen for the check on the board type maybe we should use `in` instead of `==` so any prefix is ignored?

brennen, 2018-12-27:
@ladyada ah, yep, totally.  i can throw together a quick PR tomorrow.

Fury1SOG, 2018-12-27:
@ladyada , @brennen , thanks for your help!!

brennen, 2018-12-28:
_Should_ now be handled!

closed 2018-12-28 by brennen

ladyada, 2018-12-30:
these are not released, please try the 1.1.0 version of this library and Adafruit-PlatformDetect v0.0.8, it should now work with raspi 1's!

anvoice, 2019-03-31:
Hello, I'm having a similar issue on Ubuntu 18.04 server for ARM64 with a raspberry pi 3 B. When importing board, I get the Board not Supported error. `cat /proc/cpuinfo` lists 4 different processors (4 cores on the pi I imagine) in the following format:
```
processor              : 0
BogoMIPS             : 38.40
features                : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture  : 8
CPU variant          : 0x0
CPU part              : 0xd03
CPU revision        : 4
```

Is this fixable by any chance?

ladyada, 2019-04-01:
hiya thats the same as https://github.com/adafruit/Adafruit_Blinka/issues/99 so please go there

nolansingroy, 2022-10-17:
@ladyada I was using rev 1.4 and had no problems but recently bought some new 4b raspberry pi and I guess its 1.5 and now my clone os is throwing this error. 
[image]

cat /proc/cpuinfo
[image]
 Please any advice or help would be greatly appreciated

ladyada, 2022-10-17:
please open a new issue instead of bumping a closed issue