adafruit/Adafruit_Blinka #69: NotImplementedError: Board not supported

Dmole, 2018-12-29:
https://github.com/adafruit/Adafruit_Blinka/blob/40f44f6bde990ffac815131743c0a390e84cb51a/src/board.py#L46

>>> board_id
'RASPBERRY_PI_B_REV1'

brennen, 2018-12-29:
a pi model b revision 1 isn't a 40-pin model, so that output seems correct.  we're in the process of [adding support for those boards](https://github.com/adafruit/Adafruit_Blinka/pull/67), which should land once tested.  apologies for any interim brokenness.

closed 2018-12-29 by brennen

Dmole, 2018-12-29:
Using the 40 pin values worked in my case... there is only one variable (D27) that is on the 40 pin only; https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/board/raspi_40pin.py

ladyada, 2018-12-30:
hiya try the 1.1.0 version of this library and Adafruit-PlatformDetect v0.0.8, it should now work with raspi 1's!