adafruit/Adafruit_Blinka #70: not yet ready for public consumption

Dmole, 2018-12-29:
https://github.com/adafruit/Adafruit_Blinka/blob/40f44f6bde990ffac815131743c0a390e84cb51a/src/adafruit_blinka/microcontroller/generic_linux/i2c.py#L1

https://github.com/adafruit/Adafruit_Python_PureIO/blob/master/README.md

brennen, 2018-12-29:
we're not exactly the public in the sense intended by that README, since we author both libraries.  that said, i noticed this yesterday when [adding it to the dependencies](https://github.com/adafruit/Adafruit_Blinka/pull/68) (we'll do a new release to PyPI shortly), and agree that we should probably update those docs. :)

closed 2018-12-29 by brennen

Dmole, 2018-12-29:
Thanks for taking the time to post the link (I already worked around it but nice to know it's fixed upstream)