adafruit/Adafruit_Blinka #74: Throw different exception for busio.UART on any linux?

caternuson, 2019-01-03:
Currently, `busio.UART` just tries to import from `machine` in `__init__` regardless of host:
```python
from machine import UART as _UART
```
which will throw an exception on RPi and other linuxes. But the exception is not very user friendly.
```python
ImportError: No module named 'machine'
```
How about adding a conditional similar to SPI and I2C to check for `any_embedded_linux` and throw a different exception with more useful information. Something like:
```python
if detector.board.any_embedded_linux:
    raise NotImplementedError("not supported, use pyserial")
else:
    from machine import SPI as _SPI
```

ladyada, 2019-01-03:
yeah we should! would you be up for the PR? it can just toss a RuntimeException  - we weren't sure whether we'd use pyserial or UART but went the pyserial route :)

caternuson, 2019-01-03:
Closing. Fixed with #75

closed 2019-01-03 by caternuson