adafruit/Adafruit_Blinka #79: Add support for i2c-gpio overlay pins on BCM283x

caternuson, 2019-01-24:
When using the `i2c-gpio` overlay with default settings, it will be given bus number 3 and the SDA/SCL pins will be on 23 and 24:
```
pi@raspberrypi:~ $ dmesg | grep i2c
[    3.907573] i2c /dev entries driver
[    8.113216] i2c-gpio i2c@0: using pins 23 (SDA) and 24 (SCL)
pi@raspberrypi:~ $ ls /dev/i2c*
/dev/i2c-1  /dev/i2c-3
```
(i2c-1 is the hardware i2c port)

To allow use of this feature, need to add an entry to `i2cPorts` so the i2c bus can be created using those pins:
```python
i2c = busio.I2C(board.D24, board.D23)
```

The overlay allows for assigning the bus number and pins when enabling it in /boot/config.txt. But I'm not sure how non-default settings could be handled. The blinka constructor has no concept of bus number, so it all keys off the pins.

ladyada, 2019-01-24:
we should at some point allow people to use this, but the instructions was a typo so ignoring for now and fixing instructions :)

JustinCredible-, 2019-01-27:
I patched microcontroller/bcm283x/pin.py and busio.py to allow for this on D4 and D17

asmagill, 2020-03-29:
Is the current way to use `i2c-gpio` with arbitrary pins to locally modify `microcontroller/bcm283x/pin.py` then?

It's working for me, but I just want to make sure I haven't missed an easier (or the official) way.

makermelissa, 2020-04-30:
I created a new library that allows you to do this. It's available at https://github.com/adafruit/Adafruit_Python_Extended_Bus/releases

It allows you to create an I2C object by passing the Bus ID instead of some pins. Since it extends the busio I2C object, it should be compatible. It only works with Linux boards, but those seem to be the only one having this issue. Closing this for now. If you have any problems, please open an issue in the new repository.

closed 2020-04-30 by makermelissa