adafruit/Adafruit_Blinka #84: Seesaw soil sensor getting supurious readings

caternuson, 2019-02-07:
Not 100% sure this is a blinka issue, but starting here. Something about blinka's I2C w/ Seesaw? No issue on a Metro w/ CP.

Re this thread:
https://forums.adafruit.com/viewtopic.php?f=19&t=146478

**SETUP A**
Raspberry Pi Zero W + STEMMA Soil Sensor
Sensor stuck in a pot of wet dirt
Python3 + blinka + CP libs
Readings are an average of 10 samples taken 1 second apart

Occasional spurious readings (~4000+) for moisture (time, moisture, temperature):
```
02-07-19 17:35:21 732.1 20.221553236590005
02-07-19 17:35:41 737.5 20.414848409874004
02-07-19 17:36:01 4038.4 20.307240441558
02-07-19 17:36:21 748.1 20.446733156562004
02-07-19 17:36:42 764.7 20.133870183198006
02-07-19 17:37:02 750.4 20.371672165986002
02-07-19 17:37:22 756.5 20.339123662368003
02-07-19 17:37:42 787.6 20.307240441558
02-07-19 17:38:02 761.6 20.403558438552
02-07-19 17:38:22 780.7 20.285319678210005
02-07-19 17:38:42 769.8 20.339125188245998
02-07-19 17:39:03 759.5 20.285984961018
02-07-19 17:39:23 4038.7 20.177046427086005
02-07-19 17:39:43 778.2 20.327830639290006
02-07-19 17:40:03 738.0 20.220222670974
02-07-19 17:40:23 761.5 20.381634623448
```
**SETUP B**
Metro M4 Express + STEMMA Soil Sensor
Sensor stuck in a pot of wet dirt
CP 3.1.2 + CP libs
Readings are an average of 10 samples taken 1 second apart

No spurious readings (time, moisture, temperature):
```
3676.87 614.3 25.3833
3696.92 614.0 25.1091
3716.97 613.4 24.9887
3737.02 614.0 24.8256
3757.07 614.4 24.8051
3777.12 614.2 24.6203
3797.17 614.5 24.6612
3817.22 614.4 24.5998
3837.27 613.7 24.5298
3857.32 614.1 24.4789
3877.37 613.8 24.5899
3897.42 614.0 24.4987
3917.47 613.7 24.3778
3937.52 613.8 24.4386
3957.57 614.1 24.3784
3977.62 614.6 24.4479
3997.67 614.4 24.4485
4017.72 614.8 24.3976
4037.77 614.7 24.4082
```

ladyada, 2019-02-07:
@caternuson stemma sorta relies on delays to manage timing, and the Pi is much faster. since you have it set up can you try inserting delays into seesaw? like in between the i2c send and receive command

caternuson, 2019-02-07:
Looks like there's already a delay mechanism in place?
https://github.com/adafruit/Adafruit_CircuitPython_seesaw/blob/master/adafruit_seesaw/seesaw.py#L380
`moisture_read` calls that with a delay value of 0.005:
https://github.com/adafruit/Adafruit_CircuitPython_seesaw/blob/master/adafruit_seesaw/seesaw.py#L236

So just try bumping that 0.005 up?

ladyada, 2019-02-07:
yep try 20ms or more time thx

caternuson, 2019-02-07:
Tried several, all the way up 1/2 a sec. Still gets spurious reads.

**0.010**
```
02-07-19 20:35:28       745.0   20.026262214882003
02-07-19 20:35:49       764.2   19.896068200410003
02-07-19 20:36:09       4043.7  19.864850262408005
02-07-19 20:36:29       741.1   19.929281986836003
02-07-19 20:36:49       780.3   19.94920690176
```
**0.020**
```
02-07-19 20:35:28       745.0   20.026262214882003
02-07-19 20:35:49       764.2   19.896068200410003
02-07-19 20:36:09       4043.7  19.864850262408005
02-07-19 20:36:29       741.1   19.929281986836003
02-07-19 20:36:49       780.3   19.94920690176
```
**0.050**
```
02-07-19 20:44:12       758.2   19.939909727106006
02-07-19 20:44:32       712.9   19.886105742948
02-07-19 20:44:53       4094.9  19.853557239330005
02-07-19 20:45:14       723.4   19.886771025756
02-07-19 20:45:34       812.7   19.842929499060002
```
**0.500**
```
02-07-19 20:17:01       781.2   19.843594781868
02-07-19 20:17:26       748.0   19.961165207646
02-07-19 20:17:51       4020.2  19.841598933444
02-07-19 20:18:16       773.7   19.929947269644
02-07-19 20:18:41       752.9   19.85355723933
```

ladyada, 2019-02-07:
weird, can you see what the raw value, pre floatification, is?

caternuson, 2019-02-07:
OK, floatification was coming from the averaging math. This seems to be something else, so don't need to worrying about averaging. Getting rid of that and now just doing this with `LOG_RATE=1`: 
```python
while True:
    moisture = ss.moisture_read()
    temperature = ss.get_temp()
    time_stamp = datetime.datetime.now().strftime("%m-%d-%y %H:%M:%S")
    print(time_stamp, moisture, temperature)
    log_data(time_stamp, moisture, temperature)
    time.sleep(LOG_RATE)
```
Then only test original 0.005 and 0.020, the spurious read is:
**0.005**
```
02-07-19 21:36:42       708     19.79709212394
02-07-19 21:36:43       767     19.79709212394
02-07-19 21:36:44       33600   19.910022354720002
02-07-19 21:36:45       881     20.22885456282
02-07-19 21:36:46       784     20.122577160120002
```
**0.020**
```
02-07-19 21:38:21       820     19.79709212394
02-07-19 21:38:22       889     20.01629975742
02-07-19 21:38:23       33654   19.910022354720002
02-07-19 21:38:25       867     19.910022354720002
02-07-19 21:38:26       945     20.01629975742
```

ladyada, 2019-02-07:
man who knows - it shouldn't be that high, i think max value is 4096 - i think we should re-try in the lirbary if the touch reading is above 4096 (try pressing your palm against it to see the max value

ladyada, 2019-02-15:
hey just checking back, did you find any solution or workaround?

caternuson, 2019-02-15:
Sorry, not yet. Broke down the actual plant setup to bring back to bench and got buried.

Getting rid of file logging and averaging and just doing a simple readout loop:
```python
import time, datetime
import board, busio
from adafruit_seesaw.seesaw import Seesaw

i2c = busio.I2C(board.SCL, board.SDA)

ss = Seesaw(i2c, addr=0x36)

while True:
    moisture = ss.moisture_read()
    temperature = ss.get_temp()
    time_stamp = datetime.datetime.now().strftime("%m-%d-%y %H:%M:%S")
    print(time_stamp, moisture, temperature)
    time.sleep(1)
```
Holding thumb over entire sense area, looks like 1015 is max, but can still get the odd ball reading. This is with a delay of 0.020 in the seesaw lib. 
```
02-15-19 20:53:53 1015 27.4226979465
02-15-19 20:53:54 1015 27.3164205438
02-15-19 20:53:55 33783 27.5356129185
02-15-19 20:53:56 1015 27.74818298268
02-15-19 20:53:57 1015 27.4226979465
```

Given how different the odd reading is, an out of bounds check would be pretty easy to add. But would be nice to know where it's coming from.

ladyada, 2019-02-15:
unfortunately its probably in the seesaw lib in a way i cant easily fix...so out of bounds check is the best option, check for a max of 4095

caternuson, 2019-02-15:
OK. I can PR that. Want to close this issue and open a new one in https://github.com/adafruit/Adafruit_CircuitPython_seesaw ?

ladyada, 2019-02-19:
issue cant be moved there for some reason, so opened there, closin' here!

closed 2019-02-19 by ladyada