adafruit/Adafruit_Blinka #89: Board not supported exception

qwqw330, 2019-03-09:
Hello,

I just wanted to try to get the Adafruit DC Motor Hat working on my raspberry pi B+. After installing everything I got a "Board not supported" exception though. The output from motor-test.py, blinkatest.py as well as detect.py are:
```
pi@raspberrypi:~ $ python3 motor-test.py 
Traceback (most recent call last):
  File "motor-test.py", line 3, in <module>
    from adafruit_motorkit import MotorKit
  File "/usr/local/lib/python3.5/dist-packages/adafruit_motorkit.py", line 59, in <module>
    import board
  File "/home/pi/.local/lib/python3.5/site-packages/board.py", line 68, in <module>
    raise NotImplementedError("Board not supported")
NotImplementedError: Board not supported
```
```
pi@raspberrypi:~ $ python3 blinkatest.py 
Traceback (most recent call last):
  File "blinkatest.py", line 1, in <module>
    import board
  File "/home/pi/.local/lib/python3.5/site-packages/board.py", line 68, in <module>
    raise NotImplementedError("Board not supported")
NotImplementedError: Board not supported
```
```
pi@raspberrypi:~ $ python3 detect.py 
Chip id:  BCM2XXX
Board id:  None
Is this a Pi 3B+? False
Is this a 40-pin Raspberry Pi? False
Is this a BBB? False
Is this an Orange Pi PC? False
Is this a Giant Board? False
Is this an embedded Linux system? True
Is this a generic Linux PC? False
Raspberry Pi detected.
```

I found these two issues which basically had the same issue:
#53
#66

The last post from issue #66 says that it should be fixed with 1.10. Therefore I looked what version I have.  And i do have those:
```
Adafruit-Blinka==1.2.4
Adafruit-PlatformDetect==0.0.10
```

Anything I could try to get it working? Any help would be appreciated. Thanks.

qwqw330, 2019-03-09:
Somehow it can't get my board id.

```
>>> import adafruit_platformdetect
>>> detector = adafruit_platformdetect.Detector()
>>> detector.board.id
>>> detector.chip.id
'BCM2XXX'
```
Here is my cpuinfo. Looks like that this boardid is in the source files though. Not sure what is happening. I just hacked the board.py that it works for me. Still a strange thing though
```
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 1000010
Serial		: 000000004959c3a6

```

ladyada, 2019-03-09:
thanks - i think we had some of the codes wrong. please try the latest platformdetect 0.0.11 - it should fix it! i'm going to close this but re-open if still an issue :)

closed 2019-03-09 by ladyada

qwqw330, 2019-03-09:
Well that was fast. Works just fine. Thank you!
Just looked into that commit and it kinda made me smile. Well, those typos happen to all of us.

ladyada, 2019-03-09:
:)

rmaran6, 2020-06-03:
'sudo pip3 install board' fixed it for me

bennyfuckinbo, 2020-07-28:
Currenty getting this error for circuit python 

Traceback (most recent call last):
  File "/home/pi/Desktop/RelayControl.py", line 3, in <module>
    import board
  File "/usr/local/lib/python3.7/dist-packages/board.py", line 166, in <module>
    raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported None

/proc/cpuinfo

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02042
Serial		: 00000000e9cc5bf0
Model		: Raspberry Pi 2 Model B Rev 1.2

detect.py output:

Chip id:  BCM2XXX
Board id:  None

Is this a DragonBoard 410c? False
Is this a Pi 3B+? False
Is this a Pi 4B? False
Is this a 40-pin Raspberry Pi? False
Is this a Raspberry Pi Compute Module? False
Is this a BeagleBone Black? False
Is this a BeagleBone Green? False
Is this a Giant Board? False
Is this a Coral Edge TPU? False
Is this a SiFive Unleashed?  False
Is this a PYNQ Board? False
Is this a Rock Pi board? False
Is this a Clockwork Pi board? False
Is this an embedded Linux system? True
Is this a generic Linux PC? False
Is this a UDOO Bolt? False
Is this an ASUS Tinker Board? False
Is this an OS environment variable special case? False
Raspberry Pi detected.
ASUS Tinker Board device detected.

There is no TinkerBoard detected... 
My platform detect is version 2.14.1

jglassonnz, 2020-10-07:
I am having a similar issue. Please see the output of detect.py and /proc/cpuinfo below.

/proc/cpuinfo
```
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : Generic DT based system
Revision        : 0000
Serial          : 00000000e9223457
```

detect.py
```
Chip id:  None
Board id:  None

Is this a DragonBoard 410c? False
Is this a Pi 3B+? False
Is this a Pi 4B? False
Is this a 40-pin Raspberry Pi? False
Is this a Raspberry Pi Compute Module? False
Is this a BeagleBone Black? False
Is this a BeagleBone Green? False
Is this a Giant Board? False
Is this a Coral Edge TPU? False
Is this a SiFive Unleashed?  False
Is this a PYNQ Board? False
Is this a Rock Pi board? False
Is this a Clockwork Pi board? False
Is this an embedded Linux system? False
Is this a generic Linux PC? False
Is this a UDOO Bolt? False
Is this an ASUS Tinker Board? False
Is this an STM32MP1 Board? False
Is this an OS environment variable special case? False
```