adafruit/Adafruit_Blinka #90: Missing GPIO Library Blinka TX2

johnpanos, 2019-03-16:
Hey @atan1021,

Thank you for your contribution to this library in the efforts of supporting the TX2, however I'm having some problems getting my neopixel lights running with it on my TX2.

The error references not being able to find a GPIO library in `/opt/nvidia/jetson-gpio/lib/python/Jetson/GPIO`.

I looked in JetPack to see if I was missing a dependency but I couldn't find any relating to GPIO and I couldn't find any reference to it online.

Any help is greatly appreciated, and the error is down below.

Thanks,
John

`
Traceback (most recent call last):
  File "neopixel_simpletest.py", line 2, in <module>
    import board
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_Blinka-1.2.6.dev1+g49fca60-py3.5.egg/board.py", line 68, in <module>
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_Blinka-1.2.6.dev1+g49fca60-py3.5.egg/adafruit_blinka/board/jetson_tx2.py", line 3, in <module>
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_Blinka-1.2.6.dev1+g49fca60-py3.5.egg/adafruit_blinka/microcontroller/tegra/t186/pin.py", line 5, in <module>
ImportError: No module named 'Jetson'
`

tokk-nv, 2019-03-16:
Hi John,
Jetson GPIO library is to come out pretty soon. (NVIDIA's big conference GTC scheduled next week ;)
We will inform you once that becomes available.

johnpanos, 2019-03-18:
@tokk-nv Awesome! I can't wait to make our robot look amazing with LEDs.

ladyada, 2019-03-22:
@tokk-nv @atan1021 any updates on this?

johnpanos, 2019-03-23:
I am also still wondering about it, our last FRC competition is tomorrow and we would love to use the LEDs we bought.

tokk-nv, 2019-03-23:
We haven't published the Jetson GPIO independently yet, but it is now part of L4T r32.1, that is shipped in JetPack 4.2.
The easiest is to flash your Jetson TX2 with JetPack installer 4.2, but you may not want to do that if you already have a working system on your TX2.
Then, another way is to extract it from L4T r32.1.
Pleach check the "Sample Root Filesystem" for TX2 here.
https://developer.nvidia.com/embedded/linux-tegra

johnpanos, 2019-03-24:
We switched to a raspberry pi, if we go to worlds we will see if we could use our jetson by then.

Thanks for all the help, I’ll close this issue now.

closed 2019-03-24 by johnpanos

tokk-nv, 2019-03-30:
The code is now available on GitHub and on PyPI for pip.
https://github.com/NVIDIA/jetson-gpio
https://pypi.org/project/Jetson.GPIO/