adafruit/Adafruit_Blinka #93: User application forced to use GPIO.BCM

lostcontrol, 2019-03-21:
I'm using `BCM.GPIO` mode in my application because it is easier to map to the real pins on the Raspberry Pi. Since I upgraded to CircuitPython and use `busio.I2C`, it seems I'm forced to use BCM mode.

I would expect that the library do not call `setmode()` but rather adapt to the user preference based on the result of `getmode()`.

I haven't seen any documentation related to BCM vs GPIO modes when using this library. What should be the strategy? Do you expect people to always use BCM? Or am I missing something? I would expect most people use GPIO since it is way easier to transfer to the real hardware than with BCM (where you always need to look back at the schematics).

https://github.com/adafruit/Adafruit_Blinka/blob/c8052d0b602866adffc30f15ca42744c40a8da42/src/adafruit_blinka/microcontroller/bcm283x/pin.py#L2

ladyada, 2019-03-21:
almost everyone uses BCM mode because its the 'real' pin numbers and pretty much every project and guide we've seen uses those numbers. Blinka isn't going to be compatible with using BCM.GPIO - you can use any pin map drawing from raspberry pi there's dozens online :)

closed 2019-03-21 by ladyada

lostcontrol, 2019-03-21:
Well, I see both used in tutorials, examples and books. Not a big issue in my case, I can easily do mapping on my side or convert to `GPIO.BCM`.

Probably not the best option to have this set at a module level in RPi.GPIO. Or maybe they should have not even given the user the choice to select the mapping. Anyway... Thanks for you reply.

ladyada, 2019-03-21:
we have to use BCM - because we use the BCM board pin numbers in our pin definitions