adafruit/Adafruit_Blinka #94: Attribute Error: module 'board' has no attribute 'I2C' when trying to run servos using PWM/Hat

zanussbaum, 2019-03-21:
Running the following code 

`Simple test for a standard servo on channel 0 and a continuous rotation servo on channel 1.
import time
from adafruit_servokit import ServoKit

kit = ServoKit(channels=8)
 
kit.servo[0].angle = 180
kit.continuous_servo[1].throttle = 1
time.sleep(1)
kit.continuous_servo[1].throttle = -1
time.sleep(1)
kit.servo[0].angle = 0
kit.continuous_servo[1].throttle = 0`

Getting the error 
"AttributeError: module 'board has no attribute 'I2C'

I followed the instructions from here: https://learn.adafruit.com/adafruit-16-channel-pwm-servo-hat-for-raspberry-pi/overview

Running the file on Raspberry Pi 3B
It seems like it may be an issue in the 40 pin file?

ladyada, 2019-03-21:
@kattni hiya please check it out

kattni, 2019-03-22:
The `board.I2C` object doesn't exist in Blinka. I'll look into implementing it.

zanussbaum, 2019-03-22:
Thank you, I appreciate it! Just out of curiosity, how did the tutorial work before if there was no I2C attribute? I used the code directly from the tutorial.

ladyada, 2019-03-22:
the library was updated to fix a *different* bug which then broke it here. i will add an I2C attribute

zanussbaum, 2019-03-22:
Thank you so much!

kattni, 2019-03-22:
@zanussbaum https://github.com/adafruit/Adafruit_Blinka/pull/95 fixes the issue. I've done a new release on Adafruit_Blinka, you should be able to update as soon as it deploys to PyPI.

closed 2019-03-22 by kattni