adafruit/Adafruit_Blinka #100: Feature-request: Support PocketBeagle

s-light, 2019-03-28:
i would like to test a project written in / for CircuitPython on my (new) PocketBeagle :-)

the Beagle Board Black is already in the system. but the PocketBeagle is not there yet.

Are there any official roadmaps what boards are consider to be added by Adafruit?
or is the concept more 'the community adds what they need' ? (iam totally fine with this concept ;-))

and if the second way is the way to go - 
is there any documentation how to do this?
what is needed to add a new Board?
i think 
- as the BBB is already supported 
- and [Adafruit_BBIO](https://github.com/adafruit/adafruit-beaglebone-io-python) also seems to run nicely on PocketBeagle
this 'should be' an easy thing?
only add some good pin-mappings?!

ladyada, 2019-03-28:
Beaglebone black and pocket would be a great addition!
It shoudl be easy, I just don't have time with all the other things I'm doing right now :)
first step is adding platform detection
https://github.com/adafruit/Adafruit_Python_PlatformDetect
once that works, then you should be able to add a new `board` type (the chip is already defined since its the same as the BBB)
if you do it that would be great!

s-light, 2019-03-28:
thanks for the fast feedback!
i will look into the PlatformDetect thing!

ladyada, 2019-03-28:
sweet! @makermelissa may also be interested

makermelissa, 2019-03-28:
Yup, I’m interested.

s-light, 2019-03-28:
update → PlatformDetect already supports PocketBeagle:
```
$ ./detect.py 
Chip id:  AM33XX
Board id:  BEAGLEBONE_POCKETBEAGLE
Is this a Pi 3B+? False
Is this a 40-pin Raspberry Pi? False
Is this a Beagle? True
Is this a BBB? False
Is this an Orange Pi PC? False
Is this a Giant Board? False
Is this an embedded Linux system? True
Is this a generic Linux PC? False
``` 
( i have added the *Is this a Beagle?* question in my copy of the detect.py just for testing..)

only problem currently is for reading the board id you need root privileges.
for this i opened a issue: https://github.com/adafruit/Adafruit_Python_PlatformDetect/issues/20
(including a udev-based solution i found on the net)

i will have a look at adding a board tomorrow.

makermelissa, 2019-03-28:
Awesome. You’re making great progress. I’ll get my pocket Beagle setup tonight so I can help test if you finish soon.

s-light, 2019-03-28:
had to try it today ;-)
have a problem now - 
currently the pin name assignment for SPI and I2C is done in the processor file:
[microcontroller/am335x/pin.py#L203](https://github.com/s-light/Adafruit_Blinka/blob/PocketBeagle/src/adafruit_blinka/microcontroller/am335x/pin.py#L203)
but the *header pin names` are not common for the two boards - they have fully different pin-header layouts.
how to do this? (i did not find a similar case in one of the other board mappings..)
[PocketBeagle Pin-Header](https://github.com/beagleboard/pocketbeagle/wiki/System-Reference-Manual#70-connectors-)
[BeagleBoard Black Pin-Header](https://github.com/beagleboard/beaglebone-black/wiki/System-Reference-Manual#70-connectors-)

i did not check the underlying / internal signal names yet - 
but eventually it would make sens to use the internal names (like `Mode0 Name`) in the processor definitions file - if they are the same for the both board-types?!
and then use the board file to map these to the pin-header and common names(like MISO) ?!

looking forward for your thoughts & input ;-)

ladyada, 2019-03-28:
yeah, we should have the PX_XX names in the chip, then the board defines the 'pretty names' - do you want to try that? we'll hvae to 'fix' BBB (but nobody is using it yet)

s-light, 2019-03-29:
i will try.
should i do this in two separate pull requests or in one?

ladyada, 2019-03-29:
one is fine - whatever you like!

s-light, 2019-03-29:
additions are done.
pull-request is created.
but i have not tested anything yet. ;-)

pdp7, 2019-03-30:
@s-light thanks for adding PocketBeagle to Blinka!

Now that Adafruit is distributing the PocketBeagle I'm sure there will be people interested in leveraging Blinka to utilize CircuitPython libraries.  I think these changes make sense including moving the SPI and I2C pins to the board files.

I'll test out the BeagleBone Black and PocketBeagle with PR #101

pdp7, 2019-03-31:
I have merged PR #101 and added an example to blink LED on PocketBeagle.

I think additional testing is needed to verify that I2C and SPI are working OK before this issue can be closed.

s-light, 2019-04-01:
i have added PR #102 
- some 'nice comments'
- added PocketBeagle I2C ports (see *1)

and also tested I2C with [APDS9960](https://learn.adafruit.com/adafruit-apds9960-breakout/) and the [lib](https://github.com/adafruit/Adafruit_CircuitPython_APDS9960) and the basic tests all went fine :-)

i have not tested SPI or UART. 

*1: i don't fully understand how the libraries work here - 
my feeling is that it is not really a clean way to use the `P1_XX` and `P8_XX` names in this list...
at all i don't know if they are a good fit for the processor  file....
on my quest to look for the internal processor names i went down a deep rabbit hole ;-)
- GPIO_nnn can not be used - as the lib does not recognize this for the special functions it seems.. (teste with the `Adafruit_BBIO.PWM`..)
so decided to not think anymore about this for now.

pdp7, 2019-04-02:
@s-light thanks for the update!  I will review #102 and advise if I run into any issues.  I'll also take at a look at i2cPorts, spiPorts and uartPorts and think about the best way to handle those between P1/P2 and P8/P9 versions.

s-light, 2019-04-02:
i have pushed some more updates - 
these are trying to get a more common pin-naming for the spi/i2c/uart (and other special function-names are also defined. but i don't know if they could get used ;-)

the [simple `blinkatest.py`](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi) runs on my pocketbeagle all fine with both SPI and I2C pin definitions:
```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

"""
    simple test for adafruit-blinka.

    https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi
"""

import board
import digitalio
import busio

print("Hello blinka!")

# Try to great a Digital input
pin = digitalio.DigitalInOut(board.P1_2)
print("Digital IO ok!")

# Try to create an I2C device
i2c = busio.I2C(board.SCL, board.SDA)
print("I2C ok!")
i2c = busio.I2C(board.SCL_2, board.SDA_2)
print("I2C_2 ok!")

# Try to create an SPI device
spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
print("SPI ok!")
spi = busio.SPI(board.SCLK_1, board.MOSI_1, board.MISO_1)
print("SPI_1 ok!")

print("done!")

```
result:
```
debian@beaglebone:~/python_tests$ ./blinka_test.py 
Hello blinka!
Digital IO ok!
I2C ok!
I2C_2 ok!
SPI ok!
SPI_1 ok!
done!
debian@beaglebone:~/python_tests$
```

s-light, 2019-04-02:
today i tried and tested spi with an actual hardware:
i used an 
i uploaded my test-script [oled_display.py](https://github.com/s-light/pocketbeagle_python_tests/blob/master/oled_display.py)
but this does not work as expected:
```
debian@beaglebone:~/pocketbeagle_python_tests$ ./oled_display.py 
oled display tests
setup spi
setup digitalio
init SSD1306_SPI
Could not open SPI device - check if SPI is enabled in kernel!
Traceback (most recent call last):
  File "./oled_display.py", line 43, in <module>
    display = adafruit_ssd1306.SSD1306_SPI(128, 32, spi, pin_dc, pin_rst, pin_cs)
  File "/home/debian/.local/lib/python3.5/site-packages/adafruit_ssd1306.py", line 222, in __init__
    external_vcc=external_vcc, reset=reset)
  File "/home/debian/.local/lib/python3.5/site-packages/adafruit_ssd1306.py", line 81, in __init__
    self.poweron()
  File "/home/debian/.local/lib/python3.5/site-packages/adafruit_ssd1306.py", line 142, in poweron
    self.write_cmd(SET_DISP | 0x01)
  File "/home/debian/.local/lib/python3.5/site-packages/adafruit_ssd1306.py", line 228, in write_cmd
    spi.write(bytearray([cmd]))
  File "/home/debian/Adafruit_Blinka/src/busio.py", line 129, in write
    return self._spi.write(buf, start, end)
  File "/home/debian/Adafruit_Blinka/src/adafruit_blinka/microcontroller/generic_linux/spi.py", line 39, in write
    self._spi.open(self._port, 0)
FileNotFoundError: [Errno 2] No such file or directory
debian@beaglebone:~/pocketbeagle_python_tests$
```
i had first to fix the pin-mode. and also checked this:
```
debian@beaglebone:~/pocketbeagle_python_tests$ config-pin -q P1_6
P1_06 Mode: spi_cs
debian@beaglebone:~/pocketbeagle_python_tests$ config-pin -q P1_8
P1_08 Mode: spi_sclk
debian@beaglebone:~/pocketbeagle_python_tests$ config-pin -q P1_10
P1_10 Mode: spi
debian@beaglebone:~/pocketbeagle_python_tests$ config-pin -q P1_12
P1_12 Mode: spi
```
i also tried with sudo - but this did not change anything (i installed the libraries as both - sudo and non sudo)

so it seems there is something other wrong with this currently...

s-light, 2019-04-02:
regarding the pin names i just found #62 and specially tannewts comment https://github.com/adafruit/Adafruit_Blinka/issues/62#issuecomment-446353647
so i think i was on the wrong track with my namings in the different files :-(
(i wanted to do it more the opposite way :-p )

pdp7, 2019-04-02:
I have installed the branch for #102 on my PB and run the examples/pb_digitalio.py OKAY.

There does not seem to be any regressions so I will merge this to avoid having a large changeset in the future.

I think there is still more testing to do for BeagleBone and PocketBeagle, and I think that we may need to refactor how I2C and SPI are handled given the different header schemes.

pdp7, 2019-04-03:
@s-light 
Thanks for creating the [PocketBeagle tests repo](https://github.com/s-light/pocketbeagle_python_tests/).

**Could you try running `oled_display.py` again with trace so we can tell what is causing the `FileNotFoundError`?**

For example, running blinka_test.py with strace:
```
debian@beaglebone:~/pocketbeagle_python_tests$ strace -o /tmp/strace python3 ./blinka_test.py 
Hello blinka!
Digital IO ok!
I2C ok!
I2C_2 ok!
SPI ok!
SPI_1 ok!
done!
debian@beaglebone:~/pocketbeagle_python_tests$ /opt/scripts/tools/version.sh xc^C
debian@beaglebone:~/pocketbeagle_python_tests$ egrep '(/sys/|/dev/)' /tmp/strace |grep -v nvmem
connect(3, {sa_family=AF_UNIX, sun_path="/dev/log"}, 110) = 0
access("/sys/class/gpio/gpio87", R_OK|W_OK|X_OK) = 0
open("/sys/class/gpio/gpio87/direction", O_WRONLY) = 4
open("/sys/devices/platform/ocp/ocp:P1_02_pinmux/state", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 4
open("/sys/class/gpio/gpio87/direction", O_WRONLY) = 4
open("/sys/devices/platform/ocp/ocp:P1_02_pinmux/state", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 4
open("/dev/i2c-1", O_RDWR|O_LARGEFILE|O_CLOEXEC) = 4
open("/dev/i2c-2", O_RDWR|O_LARGEFILE|O_CLOEXEC) = 5
```

pdp7, 2019-04-03:
@s-lightI don't have an OLED with me, but I was able to get the same error and run strace:
```
debian@beaglebone:~/pocketbeagle_python_tests$ strace -o /tmp/oled.out python3 ./oled_display.py 
oled display tests
setup spi
setup digitalio
init SSD1306_SPI
Could not open SPI device - check if SPI is enabled in kernel!
Traceback (most recent call last):
  File "./oled_display.py", line 43, in <module>
    display = adafruit_ssd1306.SSD1306_SPI(128, 32, spi, pin_dc, pin_rst, pin_cs)
  File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_ssd1306-2.6.2-py3.5.egg/adafruit_ssd1306.py", line 222, in __init__
  File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_ssd1306-2.6.2-py3.5.egg/adafruit_ssd1306.py", line 81, in __init__
  File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_ssd1306-2.6.2-py3.5.egg/adafruit_ssd1306.py", line 142, in poweron
  File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_ssd1306-2.6.2-py3.5.egg/adafruit_ssd1306.py", line 228, in write_cmd
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_Blinka-1.2.9.dev23+g1d32b1f-py3.5.egg/busio.py", line 129, in write
    return self._spi.write(buf, start, end)
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_Blinka-1.2.9.dev23+g1d32b1f-py3.5.egg/adafruit_blinka/microcontroller/generic_linux/spi.py", line 39, in write
    self._spi.open(self._port, 0)
FileNotFoundError: [Errno 2] No such file or directory
```
From strace, I see that the error is that `/dev/spidev0.0` does not exist:
```
open("/dev/spidev0.0", O_RDWR|O_LARGEFILE) = -1 ENOENT (No such file or directory)
```
Here are the spidev files that do exist:
```
debian@beaglebone:~/pocketbeagle_python_tests$ ls -ltar /dev/spidev*
crw-rw---- 1 root spi 153, 0 Apr  3 12:46 /dev/spidev1.0
crw-rw---- 1 root spi 153, 1 Apr  3 12:46 /dev/spidev2.0
crw-rw---- 1 root spi 153, 2 Apr  3 12:46 /dev/spidev2.1
```
For camparison, Adafruit_BBIO opens `/dev/spidev1.0`:
```
debian@beaglebone:~/pocketbeagle_python_tests$ strace -o /tmp/spi.out python3 ./bbio_spi_test.py 
[255, 255, 255, 255, 255]
Hello BBIO SPI!
```
from strace:
```
open("/dev/spidev1.0", O_RDWR|O_LARGEFILE) = 4
```

pdp7, 2019-04-03:
@s-light so this comment from @robertcnelson reminded me that there is an issue with spidev indexes changing between kernels:
https://github.com/adafruit/adafruit-beaglebone-io-python/issues/297#issuecomment-446255397
and that ultimately `/dev/spidevX.Y` is not a stable interface.  I need to investigate this further to figure out what the correct approach is.

pdp7, 2019-04-03:
@s-light I chatted with @robertcnelson in our BeagleBoard.org slack (let me know if you want an invite: drew@beagleboard.org) and there has been issue of spidev indexes changing depending on the kernel version.  

Robert advised me that /dev/spidevX.Y is not a stable interface.  We have devised a new scheme to have stable consistent spidev naming in future kernel builds.  Something similar was done for PWM: https://github.com/rcn-ee/repos/blob/master/bb-customizations/suite/buster/debian/81-pwm-noroot.rules#L15-L39

the result will be something like:
```
/dev/spi/0.0
/dev/spi/0.1
/dev/spi/1.0
/dev/spi/1.1
```
where corresponding to SPI0 and SPI1.  These will be symlinks to the correct spidev device node files.

I will follow up with more information as this solution proceeds.

pdp7, 2019-04-03:
@s-light for a short term fix, moving to kernel 4.19 will provide the expected spidev indexes.  You could install 4.19 and then Blinka should be ok:
```
debian@beaglebone:~$ sudo /opt/scripts/tools/update_kernel.sh --ti-channel --lts-4_19
```
reboot and check:
```
debian@beaglebone:~$ ls -ltar /dev/spidev*
crw-rw---- 1 root spi 153, 0 Apr  3 21:33 /dev/spidev0.0
crw-rw---- 1 root spi 153, 2 Apr  3 21:33 /dev/spidev1.1
crw-rw---- 1 root spi 153, 1 Apr  3 21:33 /dev/spidev1.0
```
The oled_display.py program then is able to open the device:
```
open("/dev/spidev0.0", O_RDWR|O_LARGEFILE) = 4
```
however I do encounter this error:
```
debian@beaglebone:~/pocketbeagle_python_tests$ strace -o /tmp/oled.out python3 ./oled_display.py 
oled display tests
setup spi
setup digitalio
init SSD1306_SPI
Traceback (most recent call last):
  File "./oled_display.py", line 43, in <module>
    display = adafruit_ssd1306.SSD1306_SPI(128, 32, spi, pin_dc, pin_rst, pin_cs)
  File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_ssd1306-2.6.2-py3.5.egg/adafruit_ssd1306.py", line 222, in __init__
  File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_ssd1306-2.6.2-py3.5.egg/adafruit_ssd1306.py", line 81, in __init__
  File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_ssd1306-2.6.2-py3.5.egg/adafruit_ssd1306.py", line 142, in poweron
  File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_ssd1306-2.6.2-py3.5.egg/adafruit_ssd1306.py", line 228, in write_cmd
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_Blinka-1.2.9.dev23+g1d32b1f-py3.5.egg/busio.py", line 129, in write
    return self._spi.write(buf, start, end)
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_Blinka-1.2.9.dev23+g1d32b1f-py3.5.egg/adafruit_blinka/microcontroller/generic_linux/spi.py", line 47, in write
    self._spi.writebytes([x for x in buf[start:end]])
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_Blinka-1.2.9.dev23+g1d32b1f-py3.5.egg/adafruit_blinka/microcontroller/generic_linux/spi.py", line 47, in <listcomp>
    self._spi.writebytes([x for x in buf[start:end]])
OSError: [Errno 22] Invalid argument
```
I am not sure if that is because I don't have the hardware or if it a seperate issue.  Though invalid argument does make me think there might be some API mismatch.

RobertCNelson, 2019-04-03:
So far i've pushed out 7 builds for this:
```
/dev/spi/0.0
/dev/spi/0.1
/dev/spi/1.0
/dev/spi/1.1
```

These kernels utilze the symlink..
```
4.4.155-ti-r154
4.4.155-ti-rt-r154
4.9.147-ti-r120
4.9.147-ti-rt-r120
4.14.108-ti-r103
4.14.108-ti-rt-r103
4.14.108-ti-xenomai-r103
```
I'll work more tomorrow on v4.19.x and all the bone variants..

Regards,

pdp7, 2019-04-04:
@RobertCNelson thanks for doing that so fast!

For reference, here is a commit to add the /dev/spi/X.Y properties to device tree:
SPIDEV: use symlink label to help udev
https://github.com/rcn-ee/repos/commit/3efd567a81bbdd262a50d6167f93c0e7765379cf

And udev script that create the corresponding symlinks:
https://github.com/mvduin/py-uio/blob/master/etc/udev/rules.d/10-of-symlink.rules

pdp7, 2019-04-05:
FYI - I ran:
```
sudo apt install --only-upgrade bb-customizations
sudo /opt/scripts/tools/update_kernel.sh 
```
after reboot, I have the new /dev/spi/X.Y symlinks:
```
debian@beaglebone:~$ ls -ltar /dev/spidev*
crw-rw---- 1 root spi 153, 0 Apr  5 14:09 /dev/spidev1.0
crw-rw---- 1 root spi 153, 1 Apr  5 14:09 /dev/spidev1.1
crw-rw---- 1 root spi 153, 3 Apr  5 14:09 /dev/spidev2.1
crw-rw---- 1 root spi 153, 2 Apr  5 14:09 /dev/spidev2.0
```
```
debian@beaglebone:~$ ls -ltar /dev/spi/
total 0
drwxr-xr-x  2 root root  120 Nov  3  2016 .
drwxr-xr-x 16 root root 3320 Apr  5 14:09 ..
lrwxrwxrwx  1 root root   12 Apr  5 14:09 0.0 -> ../spidev1.0
lrwxrwxrwx  1 root root   12 Apr  5 14:09 0.1 -> ../spidev1.1
lrwxrwxrwx  1 root root   12 Apr  5 14:09 1.1 -> ../spidev2.1
lrwxrwxrwx  1 root root   12 Apr  5 14:09 1.0 -> ../spidev2.0
```
```
debian@beaglebone:~$ uname -r
4.14.108-ti-r103
debian@beaglebone:~$ cat /etc/dogtag 
BeagleBoard.org Debian Image 2019-03-03
debian@beaglebone:~$ cat /etc/debian_version 
9.8
```
```
debian@beaglebone:~$ cat /etc/udev/rules.d/10-of-symlink.rules 
# vim: ft=udevrules

#From: https://github.com/mvduin/py-uio/blob/master/etc/udev/rules.d/10-of-symlink.rules

# allow declaring a symlink for a device in DT

ATTR{device/of_node/symlink}!="", \
	ENV{OF_SYMLINK}="%s{device/of_node/symlink}"

ENV{OF_SYMLINK}!="", ENV{DEVNAME}!="", \
	SYMLINK+="%E{OF_SYMLINK}", \
	TAG+="systemd", ENV{SYSTEMD_ALIAS}+="/dev/%E{OF_SYMLINK}"
debian@beaglebone:~$ 
```

pdp7, 2019-04-08:
**I2C testing notes**
just wanted to shared some successful testing that I did
* BME280 on BeagleBone Black I2C2
  * P9_19 is I2C2 SDA  
  * P9_20 is I2C2 SCL
```
debian@beaglebone:~/Adafruit_CircuitPython_BME280$ sudo python3 ./examples/bme280_normal_mode.py 
src/board.py: board_id: BEAGLEBONE_BLACK
src/board.py: SCL: I2C2_SCL
src/busio.py: class I2C: init: scl=I2C2_SCL
src/busio.py: class I2C: init: sda=I2C2_SDA
src/busio.py: class I2C: _I2C: <class 'adafruit_blinka.microcontroller.generic_linux.i2c.I2C'>
src/busio.py: class I2C: i2cPorts: ((1, I2C1_SCL, I2C1_SDA), (2, I2C2_SCL, I2C2_SDA))
src/busio.py: class I2C: portId: 1
src/busio.py: class I2C: portScl: I2C1_SCL
src/busio.py: class I2C: portSda: I2C1_SDA
src/busio.py: class I2C: portId: 2
src/busio.py: class I2C: portScl: I2C2_SCL
src/busio.py: class I2C: portSda: I2C2_SDA
src/busio.py: class I2C: scl: I2C2_SCL
src/busio.py: class I2C: sda: I2C2_SDA
generic_linux/i2c.py: class I2C: __init__: bus_num=2
generic_linux/i2c.py: class I2C: __init__: mode=0
generic_linux/i2c.py: class I2C: __init__: self._i2c_bus=None

Temperature: 23.3 C
Humidity: 41.9 %
Pressure: 1004.9 hPa
Altitude = 69.43 meters
```

* BME280 on PocketBeagle I2C1:
  * P2.9 SCL
  * P2.11 SDA
```
ebian@beaglebone:~/Adafruit_CircuitPython_BME280$ sudo strace -f -o /tmp/a python3 ./examples/bme280_simpletest.py 
src/board.py: board_id: BEAGLEBONE_POCKETBEAGLE
src/board.py: SCL: I2C1_SCL
src/busio.py: class I2C: init: scl=I2C1_SCL
src/busio.py: class I2C: init: sda=I2C1_SDA
src/busio.py: class I2C: _I2C: <class 'adafruit_blinka.microcontroller.generic_linux.i2c.I2C'>
src/busio.py: class I2C: i2cPorts: ((1, I2C1_SCL, I2C1_SDA), (2, I2C2_SCL, I2C2_SDA))
src/busio.py: class I2C: portId: 1
src/busio.py: class I2C: portScl: I2C1_SCL
src/busio.py: class I2C: portSda: I2C1_SDA
src/busio.py: class I2C: scl: I2C1_SCL
src/busio.py: class I2C: sda: I2C1_SDA
generic_linux/i2c.py: class I2C: __init__: bus_num=1
generic_linux/i2c.py: class I2C: __init__: mode=0
generic_linux/i2c.py: class I2C: __init__: self._i2c_bus=None

Temperature: 22.7 C
Humidity: 42.3 %
Pressure: 1004.9 hPa
Altitude = 69.61 meters
^CTraceback (most recent call last):
  File "./examples/bme280_simpletest.py", line 24, in <module>
    time.sleep(2)
KeyboardInterrupt
debian@beaglebone:~/Adafruit_CircuitPython_BME280$ grep /dev/ /tmp/a
1238  connect(3, {sa_family=AF_UNIX, sun_path="/dev/log"}, 110) = 0
1238  open("/dev/i2c-1", O_RDWR|O_LARGEFILE|O_CLOEXEC) = 4
```

pdp7, 2019-04-11:
@s-light it appears that there is a regression in Adafruit_Blink SPI support for the BeagleBone Black.  It also prevents SPI from working on the PocketBeagle.

The oled_display.py script for the SSD1306 OLED in SPI mode results in `OSError: [Errno 22] Invalid argument` on both the PocketBeagle and BeagleBone Black.

Adafruit_Blink tries to set the SPI bus mode to have no CS.  However, the Linux SPI driver used on the PocketBeagle and BeagleBone does not support this mode.

I am going to open a new issue with much more information.

pdp7, 2019-04-13:
@s-light after merging PR #105 for issue #104, I tested OK with BME280 connected to SPI0 on PocketBeagle.

Could you try testing SPI again?

### Details from the BME280 test:
Make sure pins are configured for SPI0 plus P1.6 as GPIO for CS:
```
config-pin p1.6  gpio     # CS
config-pin p1.8  spi_sclk # SPI0 CLK
config-pin p1.10 spi     # SPI0 MISO
config-pin p1.12 spi     # SPI0 MOSI
```

`Adafruit_CircuitPython_BME280/examples/bme280_simpletest.py`:
```py
import board
import busio
import adafruit_bme280
import digitalio

spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
bme_cs = digitalio.DigitalInOut(board.P1_6)
bme280 = adafruit_bme280.Adafruit_BME280_SPI(spi, bme_cs)

print("\nTemperature: %0.1f C" % bme280.temperature)
```
```
debian@beaglebone:~$ python3 bme280_simpletest.py

Temperature: 22.0 C
```
system version info:
```
debian@beaglebone:~$ cat /etc/dogtag 
BeagleBoard.org Debian Image 2019-03-03
debian@beaglebone:~$ uname -r
4.14.78-bone17
debian@beaglebone:~$ cat /etc/debian_version 
9.8
debian@beaglebone:~$ ls -ltar /dev/spidev*
crw-rw---- 1 root spi 153, 0 Apr 12 17:41 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Apr 12 17:42 /dev/spidev1.0
crw-rw---- 1 root spi 153, 2 Apr 12 17:42 /dev/spidev1.1
```

pdp7, 2019-04-13:
**Successful test of SSD1306 OLED via SPI on PocketBeagle**
```
debian@beaglebone:~/pocketbeagle_python_tests$ cat ~/pb-config-spi.sh 
config-pin p1.2  gpio     # RST
config-pin p1.4  gpio     # DC
config-pin p1.6  spi_cs   # CS
config-pin p1.8  spi_sclk # SPI0 CLK
config-pin p1.10 spi     # SPI0 MISO
config-pin p1.12 spi     # SPI0 MOSI
debian@beaglebone:~/pocketbeagle_python_tests$ bash ~/pb-config-spi.sh 
debian@beaglebone:~/pocketbeagle_python_tests$ python3 ~/pocketbeagle_python_tests/oled_display.py 
oled display tests
setup spi
setup digitalio
init SSD1306_SPI
SSD1306_SPI: __init__: spi: <busio.SPI object at 0xb6acbd70>
SSD1306_SPI: __init__: dc: P1_4
SSD1306_SPI: __init__: reset: P1_2
SSD1306_SPI: __init__: cs: P2_6
SSD1306_SPI: __init__: call spi_device.SPIDevice(cs:P2_6)
SSD1306_SPI: __init__: returned self.spi_device: <adafruit_bus_device.spi_device.SPIDevice object at 0xb68d3d30>
SSD1306_SPI: __init__: call super().__init__()
SSD1306_SPI: __init__: return
clear display
draw some test pixels
```
system version info:
```
debian@beaglebone:~$ cat /etc/dogtag 
BeagleBoard.org Debian Image 2019-03-03
debian@beaglebone:~$ uname -r
4.14.78-bone17
debian@beaglebone:~$ cat /etc/debian_version 
9.8
debian@beaglebone:~$ ls -ltar /dev/spidev*
crw-rw---- 1 root spi 153, 0 Apr 12 17:41 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Apr 12 17:42 /dev/spidev1.0
crw-rw---- 1 root spi 153, 2 Apr 12 17:42 /dev/spidev1.1
```

pdp7, 2019-04-13:
However, the above examples were running `4.14.78-bone17` where SPI0 was `/dev/spidev0.0`.  The `bone` kernel variant is not the default on the BeagleBoard.org Debian images that @RobertCNelson creates.  Switching back to `4.14.108-ti-r103`, SPI0 is now `/dev/spidev1.0`.  Temporary workaround is to create a symlink, but this is not a proper solution.

```
debian@beaglebone:~/pocketbeagle_python_tests$ bash ~/pb-config-spi.sh 
```
```
debian@beaglebone:~/pocketbeagle_python_tests$ python3 ~/pocketbeagle_python_tests/oled_display.py 
oled display tests
setup spi
setup digitalio
init SSD1306_SPI
SSD1306_SPI: __init__: spi: <busio.SPI object at 0xb6a6cd70>
SSD1306_SPI: __init__: dc: P1_4
SSD1306_SPI: __init__: reset: P1_2
SSD1306_SPI: __init__: cs: P2_6
SSD1306_SPI: __init__: call spi_device.SPIDevice(cs:P2_6)
SSD1306_SPI: __init__: returned self.spi_device: <adafruit_bus_device.spi_device.SPIDevice object at 0xb6874d30>
SSD1306_SPI: __init__: call super().__init__()
Could not open SPI device - check if SPI is enabled in kernel!
Traceback (most recent call last):
  File "/home/debian/pocketbeagle_python_tests/oled_display.py", line 43, in <module>
    display = adafruit_ssd1306.SSD1306_SPI(128, 32, spi, pin_dc, pin_rst, pin_cs)
  File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_ssd1306-2.6.3.dev0+gd09c0cb.d20190410-py3.5.egg/adafruit_ssd1306.py", line 229, in __init__
  File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_ssd1306-2.6.3.dev0+gd09c0cb.d20190410-py3.5.egg/adafruit_ssd1306.py", line 81, in __init__
  File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_ssd1306-2.6.3.dev0+gd09c0cb.d20190410-py3.5.egg/adafruit_ssd1306.py", line 142, in poweron
  File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_ssd1306-2.6.3.dev0+gd09c0cb.d20190410-py3.5.egg/adafruit_ssd1306.py", line 236, in write_cmd
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_Blinka-1.2.9.dev26+g42d260d-py3.5.egg/busio.py", line 129, in write
    return self._spi.write(buf, start, end)
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_Blinka-1.2.9.dev26+g42d260d-py3.5.egg/adafruit_blinka/microcontroller/generic_linux/spi.py", line 50, in write
    self._spi.open(self._port, 0)
FileNotFoundError: [Errno 2] No such file or directory
```
```
debian@beaglebone:~/pocketbeagle_python_tests$ ls -ltar /dev/spidev*
crw-rw---- 1 root spi 153, 0 Apr 12 17:42 /dev/spidev1.0
crw-rw---- 1 root spi 153, 1 Apr 12 17:42 /dev/spidev2.0
crw-rw---- 1 root spi 153, 2 Apr 12 17:42 /dev/spidev2.1

debian@beaglebone:~/pocketbeagle_python_tests$ sudo ln -s /dev/spidev1.0 /dev/spidev0.0
[sudo] password for debian: 

debian@beaglebone:~/pocketbeagle_python_tests$ ls -ltar /dev/spidev*
crw-rw---- 1 root spi  153, 0 Apr 12 17:42 /dev/spidev1.0
crw-rw---- 1 root spi  153, 1 Apr 12 17:42 /dev/spidev2.0
crw-rw---- 1 root spi  153, 2 Apr 12 17:42 /dev/spidev2.1
lrwxrwxrwx 1 root root     14 Apr 12 17:44 /dev/spidev0.0 -> /dev/spidev1.0
```
```
debian@beaglebone:~/pocketbeagle_python_tests$ python3 ~/pocketbeagle_python_tests/oled_display.py 
oled display tests
setup spi
setup digitalio
init SSD1306_SPI
SSD1306_SPI: __init__: spi: <busio.SPI object at 0xb6a7ad70>
SSD1306_SPI: __init__: dc: P1_4
SSD1306_SPI: __init__: reset: P1_2
SSD1306_SPI: __init__: cs: P2_6
SSD1306_SPI: __init__: call spi_device.SPIDevice(cs:P2_6)
SSD1306_SPI: __init__: returned self.spi_device: <adafruit_bus_device.spi_device.SPIDevice object at 0xb6882d30>
SSD1306_SPI: __init__: call super().__init__()
SSD1306_SPI: __init__: return
clear display
draw some test pixels
debian@beaglebone:~/pocketbeagle_python_tests$ ^C
debian@beaglebone:~/pocketbeagle_python_tests$ uname -r
4.14.108-ti-r103
```

pdp7, 2019-04-13:
## Workaround for spidev path:
Install `bone` instead of `ti` kernel:
```
sudo /opt/scripts/tools/update_kernel.sh --bone-kernel --lts-4_14
reboot
```
SPI0 is now `/dev/spidev0.0`:
```
debian@beaglebone:~$ ls -ltar /dev/spidev*
crw-rw---- 1 root spi 153, 0 Apr 13 13:35 /dev/spidev0.0
crw-rw---- 1 root spi 153, 2 Apr 13 13:35 /dev/spidev1.1
crw-rw---- 1 root spi 153, 1 Apr 13 13:35 /dev/spidev1.0
```
Test of SSD1306 runs OK:
```
debian@beaglebone:~/pocketbeagle_python_tests$ python3 ~/pocketbeagle_python_tests/oled_display.py 
oled display tests
setup spi
setup digitalio
init SSD1306_SPI
SSD1306_SPI: __init__: spi: <busio.SPI object at 0xb6a74d70>
SSD1306_SPI: __init__: dc: P1_4
SSD1306_SPI: __init__: reset: P1_2
SSD1306_SPI: __init__: cs: P2_6
SSD1306_SPI: __init__: call spi_device.SPIDevice(cs:P2_6)
SSD1306_SPI: __init__: returned self.spi_device: <adafruit_bus_device.spi_device.SPIDevice object at 0xb687cd10>
SSD1306_SPI: __init__: call super().__init__()
SSD1306_SPI: __init__: return
clear display
draw some test pixels
```
## Long-term solution
@RobertCNelson added udev rules to add symlinks that will not change:

* `SPI0.0` will always be `/dev/spi/0.0`
* `SPI0.1` will always be `/dev/spi/0.1`
* `SPI1.0` will always be `/dev/spi/1.0`
* `SPI1.1` will always be `/dev/spi/1.1`

For reference the udev rule:
```
debian@beaglebone:/etc/udev/rules.d$ cat 10-of-symlink.rules 
# vim: ft=udevrules

#From: https://github.com/mvduin/py-uio/blob/master/etc/udev/rules.d/10-of-symlink.rules

# allow declaring a symlink for a device in DT

ATTR{device/of_node/symlink}!="", \
	ENV{OF_SYMLINK}="%s{device/of_node/symlink}"

ENV{OF_SYMLINK}!="", ENV{DEVNAME}!="", \
	SYMLINK+="%E{OF_SYMLINK}", \
	TAG+="systemd", ENV{SYSTEMD_ALIAS}+="/dev/%E{OF_SYMLINK}"
```
[With `symlink` properties added to the device tree](https://github.com/beagleboard/bb.org-overlays/commit/d21b116ebee9e581428acf2558ebf7a56b37de6c), these symlinks are created by udev:
```
debian@beaglebone:~$ ls -la /dev/spi/
total 0
drwxr-xr-x  2 root root  100 Apr 13 13:35 .
drwxr-xr-x 16 root root 3180 Apr 13 13:35 ..
lrwxrwxrwx  1 root root   12 Apr 13 13:35 0.0 -> ../spidev0.0
lrwxrwxrwx  1 root root   12 Apr 13 13:35 1.0 -> ../spidev1.0
lrwxrwxrwx  1 root root   12 Apr 13 13:35 1.1 -> ../spidev1.1
```
## Issue with long-term solution
I'm uncertain of the correct approach to add logic in Adafruit_Blinka to use the `/dev/spi/X.Y` for BeagleBone and PocketBeagle.  Currently, [Adafruit_Blinka generic_linux](Adafruit_Blinka/src/adafruit_blinka/microcontroller/generic_linux/spi.py) just uses [`py-spidev` Python module](https://github.com/doceme/py-spidev).

In theory, this logic could be added to `py-spidev` but I am not sure if the maintainer who like that approach and how long it would take to get merged.

pdp7, 2019-04-13:
fyi - I have opened #106 for cover the inconsistent spidev path issue that I mention above.

pdp7, 2019-04-15:
PR #107 has resolved issue #106 by changing `spiPorts` inside [microcontroller/am335x/pin.py](https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/am335x/pin.py#L328) to reflect that SPI0 is `/dev/spidev1.x` and SPI1 is `/dev/spidev2.x` on both:
* the current recommended [BeagleBoard.org Debian image](http://beagleboard.org/latest-images): Debian 9.5 (Stretch) from 2018-10-07
* the newest [Debian 9.8 (Strecth) snapshot from 2019-04-07](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Stretch_Snapshot_iot)

@s-light when you have time, it would be good to know if SPI now works OK on your PocketBeagle.  thanks!

s-light, 2019-06-21:
thanks for your patience with this ;-)
today i got some time to test this...
this were my testing steps:
- burn  `bone-debian-9.5-iot-armhf-2018-10-07-4gb.img` image to sdcard
- boot up
- `sudo apt update`
- `sudo apt upgrade`
- `sudo reboot`
- ```bash
  $ ls -ltar /dev/spi*
  crw-rw---- 1 root spi 153, 0 Jun 21 19:55 /dev/spidev1.0
  crw-rw---- 1 root spi 153, 1 Jun 21 19:55 /dev/spidev2.0
  crw-rw---- 1 root spi 153, 2 Jun 21 19:55 /dev/spidev2.1
  ```
- `cd ~`
- `git clone https://github.com/s-light/pocketbeagle_python_tests.git`
- `cd pocketbeagle_python_tests`
- `pip3 install adafruit-circuitpython-ssd1306`
- `pip3 install adafruit-circuitpython-apds9960`
- `./detect.py`
  ```bash
  debian@beaglebone:~/pocketbeagle_python_tests$ ./detect.py 
  Chip id:  AM33XX
  Board id:  BEAGLEBONE_POCKETBEAGLE
  Is this a Pi 3B+? False
  Is this a 40-pin Raspberry Pi? False
  Is this a Beagle? True
  Is this a BBB? False
  Is this an Orange Pi PC? False
  Is this a Giant Board? False
  Is this an embedded Linux system? True
  Is this a generic Linux PC? False
  ```
- blinka basic test was fine:
  ```bash
  $ cp_blinka/blinka_test.py 
  Hello blinka!
  Digital IO ok!
  I2C ok!
  I2C_2 ok!
  SPI ok!
  SPI_1 ok!
  done!
  ```
- `cp_blinka/oled_display.py` → enjoy the stars ;-)
- `cp_blinka/i2c_test.py` → 
  ```bash
  $ cp_blinka/i2c_test.py 
  basic i2c tests
  []
  ```
  have to check my wiring...

but basically it seems to work :-)
i will have a look at i2c tomorrow..

s-light, 2019-06-22:
more tests on I2C:
(just used one of the boards i had available..)
- `pip3 install adafruit-circuitpython-cap1188`
- ```bash
  debian@beaglebone:~/pocketbeagle_python_tests/cp_blinka$ ./i2c_test.py 
  basic i2c tests
  [41]
  ```
- `./CAP1188.py `
  → working :-)
- reconnect to APDS9960
- ```bash
  debian@beaglebone:~/pocketbeagle_python_tests/cp_blinka$ ./i2c_test.py 
  basic i2c tests
  [57]
  ```
- ```bash
  debian@beaglebone:~/pocketbeagle_python_tests/cp_blinka$ ./APDS9960.py 
  apds9960 i2c tests
  setup i2c
  setup APDS9960
  read proximity (for 4seconds)
  0
  43
  39
  40
  66
  255
  234
  86
  43
  30
  14
  14
  46
  62
  29
  22
  read color
  Red: 953, Green: 806, Blue: 759, Clear: 1767
  waiting for gesture... (10 times - but no timeout)
  Saw gesture: 3: Left
  ..
  Saw gesture: 4: Right
  ..
  Saw gesture: 3: Left
  ..
  Saw gesture: 4: Right
  ..
  Saw gesture: 1: Up
  ..
  Saw gesture: 3: Left
  ..
  Saw gesture: 2: Down
  ..
  Saw gesture: 1: Up
  ..
  Saw gesture: 2: Down
  ..
  Saw gesture: 4: Right
  ..
  ```
  → working!

i also added a combined example with I2C and SPI:
[combi_APD9960_oled.py](https://github.com/s-light/pocketbeagle_python_tests/blob/master/cp_blinka/combi_APD9960_oled.py)
:-)
and this also works :-)
![P1700905_small](https://user-images.githubusercontent.com/1340319/59968303-f785ed00-9537-11e9-8e62-40a661287356.jpg)

so i think that this is solved!
Thanks for your patience and energy on this!

makermelissa, 2019-06-22:
Thank you for testing this so thoroughly. :)

closed 2019-06-22 by makermelissa