adafruit/Adafruit_Blinka #99: Error recognising Raspberry Pi

DharminB, 2019-03-28:
Similar to #66  , I am getting `NotImplementedError: Board not supported`. 

I am using Raspberry Pi 3 Model B+, with 64-bit Arch linux ARM installed.
Below is the output of `/proc/cpuinfo`
```
processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
```
On tracing back where the error arises, I found out that in `neopixel_write.py`, line [14](https://github.com/adafruit/Adafruit_Blinka/blob/master/src/neopixel_write.py#L14) returns `None`. This function call goes to `Adafruit_Python_Platformdetect`'s `_pi_rev_code` [here](https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/board.py#L289). However, `detector.chip.id` is `BCM2XX` when running [detect.py](https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/bin/detect.py) and when trying the code myself in python3 console. So the control goes to [`get_cpuinfo_field`](https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/__init__.py#L41). Here, the regex pattern does not find any matches because there are no lines starting with `Revision` or `revision`. There are a few lines with `CPU revision` though. I don't know if those are the ones that it should be looking for. If they are, a quick fix would be to remove `^` from the regex pattern [here](https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/__init__.py#L47)

ladyada, 2019-03-28:
looks like Arch is different enough, can you ask the Arch maintainers how you would know what chip its running on? cause as-is, `CPU variant     : 0x0 CPU part        : 0xd03 CPU revision    : 4` is not enough to determine the raspberry pi in use!

anvoice, 2019-04-01:
Hello, same issue on Ubuntu 18.04 server for arm64 and pi 3 b, same /proc/cpuinfo output as opening post. Is the advice the same for me (go ask Ubuntu maintainers how to determine chip)? Thank you.

ladyada, 2019-04-01:
yep! ask them "how can i tell what raspberry pi am i on"
we have it working on raspbian just fine but ubuntu & arch put that detail elsewhere :/

anvoice, 2019-04-01:
I managed to find a file called /sys/firmware/devicetree/base/cpus/cpu@0/compatible which lists the processor as 'arm,cortex-a53'. The problem is that both the pi 2 v1.2 and the pi 3 use the same processor... Is there any chance you could use something like the frequency (the pi 2 chip has a default lower clock) to determine the board?

ladyada, 2019-04-01:
close but not enough, we need something more specific if possible!

anvoice, 2019-04-01:
Ok what about the `lswh` command (my edit above)? It doesn't get much more specific than "Raspberry Pi Model B Rev 1.2"...

ladyada, 2019-04-01:
sure, thats good enough - can you submit a PR to add that to
https://github.com/adafruit/Adafruit_Python_PlatformDetect
once we can detect it, then blinka will automatically work

anvoice, 2019-04-02:
Hello again. I got some code working (hopefully not too crude, I'm a total novice) to detect the pi version on my platform, and was about to put together that pull request, but ran into another issue: now that my platform is detected, the line `from RPi._GPIO import *` fails saying "This module can only be run on a Raspberry Pi!" So apparently something else now fails to detect the platform and breaks. I can't seem to find where the enigmatic `RPi._GPIO` lives so I don't know how to fix that.

I will try to get the pull request underway, but it doesn't seem like it'll fix the problem completely.

ladyada, 2019-04-02:
does RPi GPIO python library work on your pi?

anvoice, 2019-04-02:
Apparently not. Running `import RPi.GPIO` results in the same error. Do I need to go modify the RPi.GPIO's platform detection code to work on my setup?

ladyada, 2019-04-02:
yep you got a lot of work ahead of you - most things wont work on non-raspbian. we really only guarantee it (and so does the pi foundation)

anvoice, 2019-04-02:
I see, thanks for your help.

satmandu, 2019-06-07:
I mentioned this problem here: https://github.com/adafruit/Adafruit_Blinka/issues/130

On 3B+, since /proc/cpuinfo doesn't give anything useful on mainline arm64 kernels.) Though that is addressed in part in gpio-python here https://sourceforge.net/p/raspberry-gpio-python/tickets/161/ by looking at ```cat /proc/device-tree/system/linux,revision | xxd``` on some arm64 systems, though not on mainline kernels, though /proc/device-tree/model seems to be always useful:

On 3B
```
cat /proc/device-tree/model 
Raspberry Pi 3 Model B
```
On 3B+
```
cat /proc/device-tree/model 
Raspberry Pi 3 Model B Plus Rev 1.3
```

The sourceforge link has a patch which can be used to get gpio-python working on arm64:

```
sudo apt install mercurial
sudo pip3 install --upgrade hg+http://hg.code.sf.net/p/raspberry-gpio-python/code#egg=RPi.GPIO
```

gutoandreollo, 2019-12-26:
On Arch latest (as of 2019-12-26), on a Raspberry Pi 3B (non-plus):

```
[alarm@alarm base]$ uname -a
Linux alarm 5.4.6-1-ARCH #1 SMP Mon Dec 23 09:59:08 MST 2019 aarch64 GNU/Linux
```

I got these strings that are quite handy, off the devicetree:

```
[alarm@alarm ~]$ cat /sys/firmware/devicetree/base/model ; echo
Raspberry Pi 3 Model B
[alarm@alarm ~]$ cat /sys/firmware/devicetree/base/compatible ; echo
raspberrypi,3-model-bbrcm,bcm2837
[alarm@alarm ~]$
```

ladyada, 2019-12-26:
PR's to platform-detect are welcome...we dont run anything other than Raspbian :)

mruz, 2020-01-18:
Raspberry pi 4 board is also not detected on openSUSE-Tumbleweed-ARM-JeOS-raspberrypi4.aarch64-2020.01.08-Snapshot20200115
https://en.opensuse.org/HCL:Raspberry_Pi4

ladyada, 2020-01-18:
@mruz thanks, we aren't running openSUSE - if you could PR an update to platform-detect to help it know what its running on, blinka will 'magically' work :)
https://github.com/adafruit/Adafruit_Python_PlatformDetect

makermelissa, 2020-04-24:
This was fixed in an update to PlatformDetect. Closing.

closed 2020-04-24 by makermelissa