adafruit/Adafruit_Blinka #108: Jetson Tx1 and Nano blinka support

cetinkayafaruk, 2019-04-18:
Which blinka libraries and functions are working with jetson nano and tx1 ?

I need to use gpio, i2c, and UART. Jetson UART supported with blinka? 

How can use this pin definations? https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/tegra/t210/pin.py

I coudn't figure about pin definations with 40 pin gpio locations

ladyada, 2019-04-18:
@tokk-nv ^

codyrickmanvessel, 2019-05-22:
Hello, Im not sure if this is the correct place but here it goes. 

I am using the Adafruit SGB30 Module, I had previously tested on RPi 3 B+ with no issue. 
Now I am using it on the Jetson Nano but am running into some issues (with both SDA/SDA_1 and SCL/SCL_1 interfaces)

Behavior:
1. Run Adafruit sample code for x iterations, ctrl-c out - No Error
2. Run Adafruit sample again, immediately get 
`File "/usr/local/lib/python3.6/dist-packages/adafruit_bus_device/i2c_device.py", line 76, in __init__
    raise ValueError("No I2C device at address: %x" % device_address)
ValueError: No I2C device at address: 58
`
3. Use i2cdetect, SGB30 appears to have vanished from i2c bus
4. Powercycle SGB30, reappears and works fine until I run and exit the python script. 

Any ideas?

romilly, 2019-05-25:
It looks as if the ctrl-c is leaving the bus or the board in a bad state.
Maybe there is cleanup code on the Pi that prevents this.
You may be able to solve this by running some code to reset the SGP30 in the event of a KeyboardInterrupt. I don't have an SGP30 to hand, but have one on order and will investigate and report.

codyrickmanvessel, 2019-05-28:
@romilly That is what I thought might be happening. However, I altered the example code for the SGP30 from Adafruit to just take readings for ~10sec then print out the last reading, making it a run-once script instead of looping until ctrl-c. After I run the script once, I get the same behavior until I power cycle the SGB30.

Odd thing is, I get the notification from the Blinka library stating "Cleaning up pins..." but maybe this is not working properly on the Jetson Nano?

romilly, 2019-05-29:
My SGP30 has arrived, so I hope to do some tests tomorrow.

swarren, 2019-06-18:
@cetinkayafaruk,

Is your question about how to know which pin numbers in https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/tegra/t210/pin.py correspond to which pin numbers on the Jetson GPIO expansion connector, or about how to use the Blinka library in general?

I am not at all familiar with Blinka, but I believe you should use the pin numbers at https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/board/jetson_nano.py instead, since they correspond to the connector rather than the chip. The naming of these pin numbers is at least partially aligned with one of the ways that the Raspberry Pi board names its pins; https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/board/raspi_40pin.py

If you are looking for general help with Blinka, hopefully Adafruit can provide that, or there should be some documentation included with the library?

swarren, 2019-06-18:
@codyrickmanvessel, @romilly,

It looks like you're discussing a separate issue. Perhaps you can file a separate bug for that issue? That will keep each bug focused on a specific topic, and make it much easier for people to concentrate on one thing at a time.

swarren, 2019-08-19:
I think this bug should be closed:
- The original requester hasn't responded to a request for clarification for ~2 months.
- The other issues mentioned in this bug are unrelated, and should be in separate bug reports.

@ladyada do you agree?

ladyada, 2019-08-19:
yah we def have blinka support started, now its just bug squashin time :)

closed 2019-08-19 by ladyada