adafruit/Adafruit_Blinka #110: Jetson NANO SPI fail blinka_test.py == No SCLK

patrickpoirier51, 2019-05-04:
**Loaded blinka**  
pip3 install adafruit-blinka

**launch  test script**
sudo python3 blinka_test.py
import board
import digitalio
import busio
print("hello blinka!")
print("board contents: ", dir(board))
# Try to great a Digital input
pin = digitalio.DigitalInOut(board.D4)
print("Digital IO ok!")
# Try to create an I2C device
i2c = busio.I2C(board.SCL, board.SDA)
print("I2C ok!")
# Try to create an SPI device
spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
print("SPI ok!") 
print("done!")

**here is the result**
hello blinka!
board contents:  ['D10', 'D11', 'D12', 'D13', 'D16', 'D17', 'D18', 'D19', 'D20', 'D21', 'D22', 'D23', 'D24', 'D25', 'D26', 'D27', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'I2C', 'SCL', 'SCL_1', 'SDA', 'SDA_1', 'SPI', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', 'ap_board', 'board_id', 'detector', 'pin', 'sys']
Digital IO ok!
I2C ok!
Traceback (most recent call last):
  File "blinka_test.py", line 20, in <module>
    spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
AttributeError: module 'board' has no attribute 'SCLK'
Exiting... 
Cleaning up pins

ladyada, 2019-05-04:
@tokk-nv

tokk-nv, 2019-06-28:
Hello patrickpoirier51 for the long delay in response.

NVIDIA came out with pinmux configuration table necessary to first expose SPI on the 40pin.
https://developer.nvidia.com/embedded/downloads#?search=Jetson%20Nano%20Developer%20Kit%20Users%20Pinmux%20Configuration%20Table

There is additional setup needed, and the application note to configure SPI is being worked on right now.
Let us update when the application note is published.

ladyada, 2019-07-28:
hi @tokk-nv are there any updates to configure SPI?

tokk-nv, 2019-07-29:
NVIDIA published "[Jetson Nano Developer Kit 40-Pin Expansion
Header Configuration](https://developer.nvidia.com/embedded/downloads#?search=Jetson%20Nano%20Developer%20Kit%2040-Pin%20Expansion%20Header%20Configuration)" application note.

This should cover how to enable SPI, but it is not tested against blinka_test.py.
We are planning to work on verifying this with some SPI devices with Adafruit library now.

32nguyen, 2019-08-12:
Thank you for this update. Looking forward to see any updates on this. 

I have upcoming project that uses Unicorn HAT HD 16x16 LED array with Nano. the LED is functional Raspberry Pi, I really want to control it with Nano. So excited to see that there is Python library provided in the Jetson GPIO Library package. The library has the same API as the RPi.GPIO library for Raspberry Pi. The last thing I consider is how to enable the SPI on Nano to make the LED array compatible with Nano.

Bests and thanks.

swarren, 2019-08-15:
https://github.com/adafruit/Adafruit_Blinka/pull/148 will fix this.

makermelissa, 2019-08-15:
Fixed via #148

closed 2019-08-15 by makermelissa

cscholler, 2019-08-17:
I ran the blink_test.py (latest version 2.3.1):

import board
import digitalio
import busio
print("hello blinka!")
print("board contents: ", dir(board))
pin = digitalio.DigitalInOut(board.D4.OUT)
print("Digital IO ok!")

i2c = busio.I2C(board.SCL, board.SDA)
print("I2C ok!")
i2c = busio.I2C(board.SCL_1, board.SDA_1)
print("I2C_1 ok!")

spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
print("SPI ok!")
print("done!")

**but getting this error:**

Traceback (most recent call last):
  File "/home/jetson/PycharmProjects/servoTest/blink_test.py", line 6, in <module>
    pin = digitalio.DigitalInOut(board.D4.OUT)
  File "/home/jetson/PycharmProjects/servoTest/venv/lib/python3.6/site-packages/digitalio.py", line 74, in __init__
    self._pin = Pin(pin.id)
AttributeError: 'int' object has no attribute 'id'
hello blinka!
board contents:  ['CE0', 'CE0_1', 'CE1', 'CE1_1', 'D10', 'D11', 'D12', 'D13', 'D16', 'D17', 'D18', 'D19', 'D20', 'D21', 'D22', 'D23', 'D24', 'D25', 'D26', 'D27', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'I2C', 'MISO', 'MISO_1', 'MOSI', 'MOSI_1', 'SCK', 'SCK_1', 'SCL', 'SCLK', 'SCLK_1', 'SCL_1', 'SDA', 'SDA_1', 'SPI', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', 'ap_board', 'board_id', 'detector', 'pin', 'sys']
Exiting... 
Cleaning up pins

Process finished with exit code 1

It also seems that the SPI is still not working, is there something additional that I must do?

Thank you.

swarren, 2019-08-19:
@cscholler I ran that code on a Jetson Nano with the latest packages, and did not see that error. Which Jetson board are you using?

cscholler, 2019-08-19:
Hi,

using the latest update Adafruit_Blinka 2.3.1 and using the same code as patrickpoirier51, I now get this error.

hello blinka!
board contents:  ['CE0', 'CE0_1', 'CE1', 'CE1_1', 'D10', 'D11', 'D12', 'D13', 'D16', 'D17', 'D18', 'D19', 'D20', 'D21', 'D22', 'D23', 'D24', 'D25', 'D26', 'D27', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'I2C', 'MISO', 'MISO_1', 'MOSI', 'MOSI_1', 'SCK', 'SCK_1', 'SCL', 'SCLK', 'SCLK_1', 'SCL_1', 'SDA', 'SDA_1', 'SPI', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', 'ap_board', 'board_id', 'detector', 'pin', 'sys']
Traceback (most recent call last):
  File "/home/jetson/PycharmProjects/servoTest/blink_test.py", line 8, in <module>
    pin = digitalio.DigitalInOut(board.D5)
  File "/home/jetson/PycharmProjects/servoTest/venv/lib/python3.6/site-packages/digitalio.py", line 75, in __init__
    self.direction = Direction.INPUT
  File "/home/jetson/PycharmProjects/servoTest/venv/lib/python3.6/site-packages/digitalio.py", line 101, in direction
    self._pin.init(mode=Pin.IN)
  File "/home/jetson/PycharmProjects/servoTest/venv/lib/python3.6/site-packages/adafruit_blinka/microcontroller/tegra/t210/pin.py", line 34, in init
    GPIO.setup(self.id, GPIO.IN)
  File "/home/jetson/PycharmProjects/servoTest/venv/lib/python3.6/site-packages/Jetson/GPIO/gpio.py", line 326, in setup
    ch_infos = _channels_to_infos(channels, need_gpio=True)
  File "/home/jetson/PycharmProjects/servoTest/venv/lib/python3.6/site-packages/Jetson/GPIO/gpio.py", line 121, in _channels_to_infos
    for c in _make_iterable(channels)]
  File "/home/jetson/PycharmProjects/servoTest/venv/lib/python3.6/site-packages/Jetson/GPIO/gpio.py", line 121, in <listcomp>
    for c in _make_iterable(channels)]
  File "/home/jetson/PycharmProjects/servoTest/venv/lib/python3.6/site-packages/Jetson/GPIO/gpio.py", line 104, in _channel_to_info_lookup
    raise ValueError("Channel %s is invalid" % str(channel))
ValueError: Channel CAM_AF_EN is invalid
Exiting... 
Cleaning up pins

Process finished with exit code 1

Any thoughts?

cscholler, 2019-08-19:
@swarren 

I create a new project and here are my packages: 

Adafruit-Blinka | 2.3.1 | 2.3.1
-- | -- | --
Adafruit-PlatformDetect | 1.3.2 | 1.3.2
Adafruit-PureIO | 0.2.3 | 0.2.3
Jetson.GPIO | 1.0.1 | 1.0.1

I am now getting these errors:
Traceback (most recent call last):
  File "/home/jetson/PycharmProjects/adafruit_test/adafruit_test.py", line 8, in <module>
    pin = digitalio.DigitalInOut(board.D5)
  File "/home/jetson/PycharmProjects/adafruit_test/venv/lib/python3.6/site-packages/digitalio.py", line 75, in __init__
    self.direction = Direction.INPUT
  File "/home/jetson/PycharmProjects/adafruit_test/venv/lib/python3.6/site-packages/digitalio.py", line 101, in direction
    self._pin.init(mode=Pin.IN)
hello blinka!
board contents:  ['CE0', 'CE0_1', 'CE1', 'CE1_1', 'D10', 'D11', 'D12', 'D13', 'D16', 'D17', 'D18', 'D19', 'D20', 'D21', 'D22', 'D23', 'D24', 'D25', 'D26', 'D27', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'I2C', 'MISO', 'MISO_1', 'MOSI', 'MOSI_1', 'SCK', 'SCK_1', 'SCL', 'SCLK', 'SCLK_1', 'SCL_1', 'SDA', 'SDA_1', 'SPI', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', 'ap_board', 'board_id', 'detector', 'pin', 'sys']
Exiting... 
Cleaning up pins
  File "/home/jetson/PycharmProjects/adafruit_test/venv/lib/python3.6/site-packages/adafruit_blinka/microcontroller/tegra/t210/pin.py", line 34, in init
    GPIO.setup(self.id, GPIO.IN)
  File "/home/jetson/PycharmProjects/adafruit_test/venv/lib/python3.6/site-packages/Jetson/GPIO/gpio.py", line 326, in setup
    ch_infos = _channels_to_infos(channels, need_gpio=True)
  File "/home/jetson/PycharmProjects/adafruit_test/venv/lib/python3.6/site-packages/Jetson/GPIO/gpio.py", line 121, in _channels_to_infos
    for c in _make_iterable(channels)]
  File "/home/jetson/PycharmProjects/adafruit_test/venv/lib/python3.6/site-packages/Jetson/GPIO/gpio.py", line 121, in <listcomp>
    for c in _make_iterable(channels)]
  File "/home/jetson/PycharmProjects/adafruit_test/venv/lib/python3.6/site-packages/Jetson/GPIO/gpio.py", line 104, in _channel_to_info_lookup
    raise ValueError("Channel %s is invalid" % str(channel))
ValueError: Channel CAM_AF_EN is invalid

Process finished with exit code 1

swarren, 2019-08-19:
@cscholler The errors in your latest posts aren't related to SPI at all. I suggest you open a new bug since this bug is specifically for SPI support. Please state which hardware (which Jetson board) you're using.

cscholler, 2019-08-19:
Thank you, will do.

swarren, 2019-08-19:
FWIW, I believe I've found the issue that caused your problem. Let me know the bug number once you've filed it, and I'll fill in the details there.

cscholler, 2019-08-21:
@swarren Yes, that worked, thank you.

swarren, 2019-08-21:
@patrickpoirier51 can you please retest too; I believe this is now fixed. Thanks.