adafruit/Adafruit_Blinka #114: Board not supported exception - Orange PI R1

kouliss, 2019-05-13:
Hello,

after installation of adafruit-blinka i am trying to run demo script, but getting error message:
File "/usr/local/lib/python3.5/dist-packages/board.py", line 86, in <module> raise NotImplementedError("Board not supported")
NotImplementedError: Board not supported
I am running the script on Orange PI R1 board. Here is the cat /proc/cpuinfo :
 processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 22.85
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 22.85
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 22.85
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 22.85
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : Allwinner sun8i Family
Revision        : 0000
Serial          : 02c0004224ebaf40

Any suggestion or help, please?

ladyada, 2019-05-13:
yeah we only have orange Pi PC right now (thats the only one i have)
first step is to get it recognized by platform detection
https://github.com/adafruit/Adafruit_Python_PlatformDetect
as a unique board

kouliss, 2019-05-13:
I am getting this:
root@orangepi:/scripts# sudo python3 detect.py
Chip id:  SUN8I
Board id:  None
Pi 3B+?  False
BBB?  False
Orange Pi PC?  False
generic Linux PC?  False

ladyada, 2019-05-13:
yep you need to make a new board detection for it, there is code that has to be written to add OrangePi R1 support

kouliss, 2019-05-13:
is there anything I can do on my site to add a detection?

ladyada, 2019-05-13:
yep check the board detection python code and read through how we detect other boards and add it.

kouliss, 2019-05-13:
i am sorry, but i am not very familiar with that. Where can I find th python code for board detection?

ladyada, 2019-05-13:
https://github.com/adafruit/Adafruit_Python_PlatformDetect

kouliss, 2019-05-13:
May be I am missing something, or do not understand right, but should I insert into the board.py if the detection said Board id: None ? Is there something wrong or I just do something bad?

ladyada, 2019-05-13:
the code does know the board, the library itself needs to be updated to know what board it is - it has no knowledge of Orange Pi R1, only the Orange Pi PC.
if you want to add support, please do!

kouliss, 2019-05-13:
Got it and change the files, so board is now properly detected. But since this change, i am not able to reach the IO:
ls /dev/i2c* /dev/spi*
ls: cannot access '/dev/i2c*': No such file or directory
ls: cannot access '/dev/spi*': No such file or directory

This worked before adding a board support. Is there anything else, I need to change?

Thanks.

ladyada, 2019-05-13:
i dont know, that depends on your board linux completely

kouliss, 2019-05-14:
You are right, i got the mistake in boot configuration. I corrected it and it works again - I can reach I2C again.
So I tried the test script and got another error message:
root@orangepi:~# sudo python3 /scripts/test.py
Hello blinka!
Digital IO ok!
Traceback (most recent call last):
  File "/scripts/test.py", line 12, in <module>
    i2c = busio.I2C(board.SCL, board.SDA)
  File "/usr/local/lib/python3.5/dist-packages/busio.py", line 16, in __init__
    self.init(scl, sda, frequency)
  File "/usr/local/lib/python3.5/dist-packages/busio.py", line 23, in init
    from machine import I2C as _I2C
ImportError: No module named 'machine'

Test script is this one:
import board
import digitalio
import busio

print("Hello blinka!")

# Try to great a Digital input
pin = digitalio.DigitalInOut(board.PA6)
print("Digital IO ok!")

# Try to create an I2C device
i2c = busio.I2C(board.SCL, board.SDA)
print("I2C ok!")

# Try to create an SPI device
spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
print("SPI ok!")

print("done!")

kouliss, 2019-05-14:
Sorry, this is the real code:
`import board
import digitalio
import busio

print("Hello blinka!")

# Try to great a Digital input
pin = digitalio.DigitalInOut(board.PA6)
print("Digital IO ok!")

# Try to create an I2C device
i2c = busio.I2C(board.SCL, board.SDA)
print("I2C ok!")

# Try to create an SPI device
spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
print("SPI ok!")

print("done!")
`

ladyada, 2019-05-14:
yep now you have to add support for the orange pi r1 into this repo, look at where orange pi pc stuff is definied, it is a little similar, you need to identify all the GPIO pin names and numbers

kouliss, 2019-05-16:
i solved the problem with core communication, it works at all now and i can detect I2C device.
But still there is a problem with adafruit - i changed the files with pin definition, but I am not sure, if I did right. Can You please check it?

root@orangepi:~# sudo i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

root@orangepi:~# gpio readall
 +-----+-----+---------+------+---+-Model  A-+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 |   IN | 0 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 |   IN | 0 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 0 |  7 || 8  | 0 | IN   | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 0 | IN   | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   IN | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |   IN | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   IN | 0 | 23 || 24 | 0 | IN   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 0 | IN   | CE1     | 11  | 7   |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+-Model  A-+---+------+---------+-----+-----+

I changed orangepipc.py (in "board" folder) and pin.py (in microcontroller/allwinner_h3) - see attahed files.
Is there anything I am doing wrong or missunderstood?
Can You please help me in this definition, I am really lost here...

Thank You!

[orangepipc.py.txt](https://github.com/adafruit/Adafruit_Blinka/files/3188650/orangepipc.py.txt)
[pin.py.txt](https://github.com/adafruit/Adafruit_Blinka/files/3188651/pin.py.txt)

ladyada, 2019-05-16:
hiya we can look if you submit it as a PR

kouliss, 2019-05-16:
what is a PR?

ladyada, 2019-05-16:
here are some tutorials!
https://learn.adafruit.com/an-introduction-to-collaborating-with-version-control
https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

kouliss, 2019-05-16:
Great, thanks. I just updated new branch named orangepi_r1 with updated io files

ladyada, 2019-05-16:
great! please submit a PR to the detector repo and this one too!

makermelissa, 2019-05-23:
@kouliss, I have merged it in. I'm not sure how long PyPI takes to update, but once it does would you like to go ahead and test more thoroughly with hardware hooked up?

closed 2019-05-23 by ladyada