adafruit/Adafruit_Blinka #118: Is it possible to add support to Pine64 board?

ngmacha, 2019-05-22:
Is it possible to add Blinka support for the original Pine64 board running Armbian (https://www.armbian.com/pine64/)? 

I’m currently running Armbian 5.83 Ubuntu 16.04.6 LTS, with Home Assistant HASS.io I’ve installed a RPi.GPIO clone for Pine64 from https://github.com/swkim01/RPi.GPIO-PineA64, and then adafruit-blinka, as per https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi, without errors. However running blinkatest.py gives:

> Traceback (most recent call last):
  File “blinkatest.py”, line 1, in <module>
    import board
  File “/usr/local/lib/python3.5/dist-packages/board.py”, line 86, in <module>
    raise NotImplementedError(“Board not supported”)
NotImplementedError: Board not supported

I saw a similar board, OrangePi, seems already been supported, would it be much more difficult to add Pine64? I’m ready to help and test the implementation, as I have the boards available, and if not much of programming will be required.
Thanks a lot in advance!

ladyada, 2019-05-22:
@makermelissa can help you get going - i think she even has one of the boards!
first step is to add detection working in https://github.com/adafruit/Adafruit_Python_PlatformDetect - we don't have a tutorial at this time but you can check previous commits/PRs to see what is involved :)

makermelissa, 2019-05-22:
Yep, I have a Pine64 A64+ board that I got off the original Kickstarter. Is this the one you’re referring to @ngmacha? Pine64 now has some other boards too, which is why I’m asking.

dastels, 2019-05-22:
I have one of the original kickstarter boards as well. It's working as part of my smarthome system, but I can do some testing on it.

makermelissa, 2019-05-22:
Nice, that's exactly what I'm doing with mine as well.

ngmacha, 2019-05-23:
Yes, @makermelissa , I've got a couple of original Pine64 A64+ boards, with original LCD touch screens, from their first Kickstarter too. Didn't have chances to use them much, though - tried GNU radio, found the performance is a bit too low, used as a portable oscilloscope, with the https://github.com/drandyhaas/Haasoscope, and now started the home automation project... Glad to hear that you and @dastels are using it for the same, maybe will contact if I got into problems...

Now back regarding getting Pine64 into CircuitPython... As said, I'm not much of a programmer, never did any PRs on Github. But eager to try, so please guide me and I'll see how it'd go! So where/how do I start?!

ngmacha, 2019-05-23:
@ladyada I've tried the PlatformDetect, and here is what I got:

> Chip id:  None
Board id:  None
Pi 3B+?  False
BBB?  False
Orange Pi PC?  False
generic Linux PC?  False

Will try to read/test more, but your further advise is very much appreciated. Thank you!

makermelissa, 2019-05-24:
Hi @ngmacha, the platform detect code will need to be updated to detect the Pine64 since it has never been added before, so it will need to check for the correct identifier.

ngmacha, 2019-05-27:
Thanks, @makermelissa , reading a bit around I now understand better. Let me see, once I have more time, if I can make it correctly detect the Pine64 board. BTW I found this: [https://www.hackster.io/wallarug/circuitpython-creating-custom-boards-8e979e](url) - does it correctly depict how to add a custom board into CircuitPython?

makermelissa, 2019-06-05:
Hi, no. That is an article about adding a board to CircuitPython itself rather than Blinka. Right now I'm going through and adding a different board to Blinka that doesn't have the pins already spelled out and will let you know when I know more. :)

makermelissa, 2019-06-06:
Hi @ngmacha. Ok, I know a little more now. Blinka uses libgpiod. First you need to make sure that's installed. You can follow the libgpiod section in this guide: https://learn.adafruit.com/circuitpython-on-orangepi-linux/orange-pi-pc-setup. It's an Orange Pi guide, but it's also running armbian, so the steps should be about the same.

It looks like the Pine64 A64 uses the AllWinner A64 which is very similar to the AllWinner H3. Since the AllWinner H3 is already supported, I would suggest using that which means you only need to really define a board.

You'll want to add detection for the board first in https://github.com/adafruit/Adafruit_Python_PlatformDetect. It looks like there's a Hardware tag in `/proc/cpuinfo` so that makes it easy.

After you can detect board and cpu, submit a PR. You'll then need to define a board and have it use the AllWinner H3 microcontroller. If you get stuck, let me know if you have any questions. Thanks.

yeyeto2788, 2020-01-02:
Hey guys! 

I have the first PineBook and the new PinePhone which should be running the same A64 CPU. 

If you want I can add the detection of that board if you guys open the issue on https://github.com/adafruit/Adafruit_Python_PlatformDetect with the output of `/proc/cpuinfo` so I add you board and, if desired, the PineBook and PinePhone.

Regards.

P.D: just asked about opening the issue there in order to track the changes and also because the output on my device is different.

makermelissa, 2020-01-03:
Sounds great @yeyeto2788!

closed 2020-01-10 by makermelissa