adafruit/Adafruit_Blinka #119: Jetson Nano blinka_test.py fail on GPIO

TarmiziIzzuddin, 2019-05-25:


brtknr, 2019-07-28:
I am having the same issue.

ladyada, 2019-07-30:
@tokk-nv hi please check this out and see why nvidia nano is not working

cscholler, 2019-08-16:
Has this issue been resolved yet?

makermelissa, 2019-08-16:
No. It now says:
```
Hello blinka!
Traceback (most recent call last):
  File "blinka.py", line 8, in <module>
    pin = digitalio.DigitalInOut(board.D4)
  File "/usr/local/lib/python3.6/dist-packages/digitalio.py", line 75, in __init__
    self.direction = Direction.INPUT
  File "/usr/local/lib/python3.6/dist-packages/digitalio.py", line 101, in direction
    self._pin.init(mode=Pin.IN)
  File "/usr/local/lib/python3.6/dist-packages/adafruit_blinka/microcontroller/tegra/t210/pin.py", line 34, in init
    GPIO.setup(self.id, GPIO.IN)
  File "/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/gpio.py", line 326, in setup
    ch_infos = _channels_to_infos(channels, need_gpio=True)
  File "/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/gpio.py", line 121, in _channels_to_infos
    for c in _make_iterable(channels)]
  File "/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/gpio.py", line 121, in <listcomp>
    for c in _make_iterable(channels)]
  File "/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/gpio.py", line 104, in _channel_to_info_lookup
    raise ValueError("Channel %s is invalid" % str(channel))
ValueError: Channel AUD_MCLK is invalid
Exiting...
Cleaning up pins
```

cscholler, 2019-08-19:
I am getting the same error.

swarren, 2019-08-19:
This was caused by the following commit, which I'll investigate partially reverting:
https://github.com/NVIDIA/jetson-gpio/commit/66ef28c50a6c03111ab586523761a8e7a8b58b48.

cscholler, 2019-08-20:
@swarren Thank you for looking into it.

swarren, 2019-08-21:
I have released a new version of Jetson.GPIO which should restore compatibility with Blinka. Try upgrading or re-installing; this issue should be resolved now. Thanks.

cscholler, 2019-08-21:
@swarren Yes, that worked, thank you.

swarren, 2019-08-21:
@TarmiziIzzuddin can you please retest too; I believe this is now fixed. Thanks.