adafruit/Adafruit_Blinka #120: Add Deinit method to PWMOut for Travis

brentru, 2019-05-30:
Travis uses Blinka, so pylint'ing libraries requiring `PWMOut` fails on `no-member` since Blinka's `PWMOut` class does not have the `deinit` method in the CircuitPython API (https://circuitpython.readthedocs.io/en/4.x/shared-bindings/pulseio/PWMOut.html#pulseio.PWMOut.deinit).

@ladyada  would a `deinit()` method be similar to `close()`? https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/generic_linux/sysfs_pwmout.py#L107

Travis log: https://travis-ci.com/brentru/CircuitPython_RGBLED/builds/113806920#L277

ladyada, 2019-05-30:
i dont know, exit calls close
https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/generic_linux/sysfs_pwmout.py#L63
but i dont know the difference
@tannewt do you know the diff between enter/exist and deinit?

sommersoft, 2019-05-30:
> the diff between enter/exist and deinit?

`__enter__` & `__exit__` are used in context management (e.g. `with foo as bar:`). `deinit` is for use outside of context management.

--------------------------------

I put this in Discord, but maybe here is better, with a little more expansion of thought. I will preface it with this however: I am not a pylint expert, nor privy to any specific goals for Blinka wrt to what it _should_ support and when.

I'm not sure things should be added to Blinka, solely for having pylint pass. Blinka being able to serve as a "mock" of sorts for pylint was a happy accident, if I remember correctly. But, establishing an expectation that Blinka will always match the core at any given time when a library is written, seems like it may slow core development. This is not meant as a "core is more important" statement, just that the two exist on separate dev cycles from what I see.

A short-term solution could be to expand the modules listed in `ignore-modules` inside the `.pylint.rc`. Currently only `board` is ignored across the libraries. Or, authors could continue to disable pylint for these particular errors (`no-member`); these are dotted throughout the library landscape.

I don't currently have a longer-term solution. Pylint may have some other approaches that would serve this situation. Since my interest is piqued just discussing, I'll likely start searching anyway. 😄

ladyada, 2019-05-31:
its worth having this library work with Blinka, so if its not working it should be fixed :)

makermelissa, 2019-05-31:
This would be nice to get it consistently linting in Travis correctly.

tannewt, 2019-05-31:
I'd suggest renaming `close` to `deinit` to match CircuitPython's implementation. `deinit` is meant as a way to manually deinitialize and release resources. It is a standard name across our core classes. Other methods on an object must check to see if it has been deinited and raise a ValueError if it has. (It is here in CircuitPython: https://github.com/adafruit/circuitpython/blob/master/shared-bindings/util.c#L38)

`__exit__` is for context managers as @sommersoft points out and it is standard to call deinit in it. `__enter__` simply returns itself.

ladyada, 2019-05-31:
ok understood, i didnt know if any of these were 'fancy' special names
@brentru - please fix my code ;)

sommersoft, 2019-05-31:
> I'd suggest renaming close to deinit

In the PR above, I initially thought about doing it this way, but I chose to just add `deinit` and have it call `close` in case it needed to be backwards compatible. I can update my PR...

PR #121 (2019-06-03): Add 'deinit()' Function To PWMOut
Fixes #120.

I cannot test this, since `pulseio.PWMOut` is only available for the Google Coral and Giant Board. Neither of which I have.
closed 2019-06-03 by brentru