adafruit/Adafruit_Blinka #122: Blinka 1.3.3 won't `import board` on Google Coral

brentru, 2019-05-31:
Seems to be related to https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/23, breaks compatibility with Coral. I haven't tested with other boards to see if this release breaks there, too.

Downgrading to v1.3.2 works properly with Coral. 

```
  import board
  File "/usr/local/lib/python3.5/dist-packages/board.py", line 64, in <module>
    elif board_id == ap_board.ORANGE_PI_R1:
AttributeError: module 'adafruit_platformdetect.board' has no attribute 'ORANGE_PI_R1'
```

brentru, 2019-05-31:
@makermelissa  any ideas as to what this might be caused by?

makermelissa, 2019-05-31:
I’ll take a look

makermelissa, 2019-05-31:
It's almost as if this was not being with the most current version of platform detect. It's defined here: https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/72bd91fccaa83a93a2d51fcc5d06b905a55b8876/adafruit_platformdetect/board.py#L29

brentru, 2019-05-31:
@makermelissa You're right, it looks like an inconsistency:
```pip3 freeze | grep PlatformDetect
Adafruit-PlatformDetect==1.0.2```

brentru, 2019-05-31:
@makermelissa  Reopening, I'm getting the same issue with an updated platformdetect and latest blinka installed via pypi.

```
mendel@king-orange:~$ pip3 freeze | grep PlatformDetect
Adafruit-PlatformDetect==1.0.3
mendel@king-orange:~$ pip3 freeze | grep Blinka
Adafruit-Blinka==1.3.3
mendel@king-orange:~$ sudo python3 blinka.py
Traceback (most recent call last):
  File "blinka.py", line 1, in <module>
    import board
  File "/usr/local/lib/python3.5/dist-packages/board.py", line 64, in <module>
    elif board_id == ap_board.ORANGE_PI_R1:
AttributeError: module 'adafruit_platformdetect.board' has no attribute 'ORANGE_PI_R1'
```

Full pip package listing
```
mendel@king-orange:~$ pip3 freeze
Adafruit-Blinka==1.3.3
adafruit-circuitpython-74hc595==1.0.2
adafruit-circuitpython-bme280==2.3.0
adafruit-circuitpython-busdevice==2.2.11
adafruit-circuitpython-charlcd==3.2.0
adafruit-circuitpython-crickit==2.1.4
adafruit-circuitpython-framebuf==1.1.0
adafruit-circuitpython-gps==3.2.3
adafruit-circuitpython-is31fl3731==2.3.1
adafruit-circuitpython-max31855==3.0.5
adafruit-circuitpython-mcp230xx==1.2.0
adafruit-circuitpython-motor==1.4.2
adafruit-circuitpython-rgbled==0.1.dev37+g7353f76
adafruit-circuitpython-seesaw==1.4.0
adafruit-circuitpython-simpleio==1.1.3
Adafruit-PlatformDetect==1.0.3
Adafruit-PureIO==0.2.3
chardet==3.0.4
edgetpu==1.2.0
edgetpuvision==1.0
httplib2==0.12.3
numpy==1.12.1
Pillow==4.0.0
protobuf==3.0.0
pycurl==7.43.0
pygobject==3.22.0
PyJWT==1.4.2
pyserial==3.4
python-apt==1.4.0b3
python-debian==0.1.30
python-debianbts==2.6.1
python-periphery==1.1.1
reportbug==7.1.7
requests==2.12.4
RPi.GPIO==0.6.5
six==1.10.0
spidev==3.4
sysv-ipc==1.0.0
unattended-upgrades==0.1
urllib3==1.19.1
```

makermelissa, 2019-05-31:
Ok, I'm going to run it and see if I get similar results.

makermelissa, 2019-05-31:
It seems to work for me. Did you remove the corresponding __pycache__ folder? It might still be running the old one.

makermelissa, 2019-05-31:
It's also possible that it's working for me because I'm running it on the Orange Pi R1.

brentru, 2019-05-31:
@makermelissa  I'm not sure if I removed the `__pycache__` folder, where is that located?

Could you try testing 1.3.3 on a RasPi?

makermelissa, 2019-05-31:
Yes, but I'll have to alter the else if statements in board.py as it would never reach that line of code otherwise.

makermelissa, 2019-05-31:
Ok, I altered `board.py` and moved the `else if` statements for the Raspberry Pi right next to the check for Coral and it had no problems. Can you try uninstalling and reinstalling? Perhaps you have it installed in more than one place. You can check your path by entering python and typing:
```
$ python3
>>> import sys
>>> sys.path
```

brentru, 2019-06-03:
@makermelissa Tried uninstalling/reinstalling and deleting the `__pycache__` folder, same issue. 

```
>>> sys.path
['', '/home/mendel', '/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-aarch64-linux-gnu', '/usr/lib/python3.5/lib-dynload', '/home/mendel/.local/lib/python3.5/site-packages', '/usr/local/lib/python3.5/dist-packages', '/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_rgbled-0.1.dev37+g7353f76-py3.5.egg', '/usr/local/lib/python3.5/dist-packages/Adafruit_Blinka-1.3.2-py3.5.egg', '/usr/lib/python3/dist-packages']
```

brentru, 2019-06-03:
Looks like a v1.0.2 of PlatformDetect was installed in `/usr/local/lib/python3.5/dist-packages/Adafruit_PlatformDetect`. Deleting and reinstalling Blinka (which sources latest Platformdetect) seems to have solved the issue.

closed 2019-06-03 by brentru

makermelissa, 2019-06-03:
Ok cool. :)