adafruit/Adafruit_Blinka #138: CM3 "Board not supported"

jm4r7in, 2019-07-14:
Hello,

When I do : "import board"
I get a message telling me the board is not supported.
Here is the result of detect.py:

> Chip id:  BCM2XXX
> Board id:  None
> Is this a DragonBoard 410c? False
> Is this a Pi 3B+? False
> Is this a Pi 4B? False
> Is this a 40-pin Raspberry Pi? False
> Is this a BBB? False
> Is this a Giant Board? False
> Is this a Coral Edge TPU? False
> Is this a SiFive Unleashed?  False
> Is this an embedded Linux system? True
> Is this a generic Linux PC? False
> Raspberry Pi detected.

Is it because of the CM3 ?
How to make it work ?

Thank you.

ladyada, 2019-07-14:
The CM3 is not yet supported! would you like to help add it?

jm4r7in, 2019-07-16:
Oh no :'(
What is needed to do to add it ?

ladyada, 2019-07-16:
@makermelissa heya any suggestions for the CM3?

makermelissa, 2019-07-16:
Hi @jm4r7in. It should have the same processor as the regular Raspberry Pi 3, but with more pins exposed. Let's start with platform detection. This is a Raspberry Pi, but not a 40-pin RasPi. So in the repo https://github.com/adafruit/Adafruit_Python_PlatformDetect we need to add the following to board.py:
```
_RASPBERRY_PI_CM_IDS = (
    RASPBERRY_PI_CM1,
    RASPBERRY_PI_CM3
)
```
Then add a Property:
```
    @property
    def any_raspberry_pi_cm(self):
        """Check whether the current board is any Compute Module Raspberry Pi."""
        return self.id in _RASPBERRY_PI_CM_IDS
```

Then you'll need to update the src/adafruit_blinka/microcontroller/bcm283x/pin.py file to add the additional pins that are exposed. The number inside of `Pin()` refers to the GPIO number. Here's a data sheet for the CM3 to help you: https://www.raspberrypi.org/documentation/hardware/computemodule/datasheets/rpi_DATA_CM_1p0.pdf Looking at the board though, it appears to have GPIO 0-45, so it's probably a matter of filling in the blanks.

Then once that's done, you'll create a file inside of src/adafruit_blinka/ and let's call it raspi_cm.py since it should work with all 3 compute modules and probably future ones. I would start with copying the contents of the raspi_40pin.py and then add the additional GPIO pins and make any necessary changes.

After you have the microcontroller and board done, you will need to add an else if statement in for the new board in src/board.py:
```
elif detector.board.any_raspberry_pi_cm:
    from adafruit_blinka.board.raspi_cm import *
```

That should get you most if not all the way there, but there's probably something I missed. Please let me know if I can clarify something.

jm4r7in, 2019-07-17:
@makermelissa Thank you very much for your answer.
Do you think you can update the repo today or tomorrow ? Or at least before the weekend ?

ladyada, 2019-07-17:
@jm4r7in the instructions above are to help you do the detection, please try them out today or tomorrow

jm4r7in, 2019-07-17:
@ladyada In the instructions above, I read:

> So in the repo https://github.com/adafruit/Adafruit_Python_PlatformDetect we need to add the following to board.py

> Then you'll need to update

So, as I understand the instructions, you need to update the repo. Am I wrong ?

makermelissa, 2019-07-17:
You can do the update and submit a Pull Request

jm4r7in, 2019-07-17:
Hmm... So, I guess I need to learn how to do the update and how to do a Pull Request.
You assume that I have the same knowledge, experience and tools as you, but it's not the case.
Thanks for your help anyway.

makermelissa, 2019-07-17:
You could take a look at https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

jm4r7in, 2019-07-18:
@makermelissa Since you know exactly what to do, and have all the environment set, could you just do a commit, please ?
Thank you very much.

makermelissa, 2019-07-18:
Hi, you could always take a stab at it. Sometimes that's the best way to learn.

If not, I'll see what I can do, but I only have the CM3+. Also, I would want to do a little testing before committing and need to set up my CM3 image, so it may not be done before you need it.

makermelissa, 2019-07-18:
The steps you need to make changes are basically:
1. Create a Fork of the Repo into your Github account
2. Clone the Repo Locally (usually something like `git clone https://github.com/jm4r7in/Adafruit_Python_PlatformDetect.git`)
3. Make your changes
4. Add the files to be committed (usually `git add -A`)
5. Commit (usually `git commit -m "Some message"`)
6. Push changes to your github (usually `git push`)
7. Got to the repo you forked from and submit a PR.

The first time I forked/submitted a PR I just watch some youtube videos first. :)

makermelissa, 2019-07-18:
Hi @jm4r7in, fortunately my RPi4 sd card was able to work in my CM3+, so I added a commit for the platform detect, which went smoothly. I'll release once it's reviewed/merged. However, you'll need to do do the Blinka code if you need it in by this weekend. Otherwise, no eta at this point. That's a little more extensive and will require research/testing.

jm4r7in, 2019-07-19:
@makermelissa I'm actually using a CM3+ Lite. Sorry, for the confusion.
I didn't what you said, and I have errors.
Can I have access to your commit ? Or, if you can see mine and tell me what's wrong...

I didn't expect your library to not work with CM3, and I really need to make it work asap.

Thanks for your help.

jm4r7in, 2019-07-19:
@makermelissa detect.py gives me the error :

`  File "Adafruit_Python_PlatformDetect/bin/detect.py", line 3, in <module>
    import adafruit_platformdetect
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_platformdetect/__init__.py", line 28, in <module>
    from adafruit_platformdetect.board import Board
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_platformdetect/board.py", line 350
    def any_raspberry_pi_cm(self):
    ^
IndentationError: unexpected indent
`

I don't know Python, so any help would be appreciated.
Thank you.

jm4r7in, 2019-07-19:
I installed the library without "sudo", and I don't have those errors, but now I have :+1: 

> NotImplementedError: Board not supported

makermelissa, 2019-07-19:
Hi @jm4r7in, Could you type `cat /proc/cpuinfo` and post what the revision number is? We may need to add that for your board to work.

makermelissa, 2019-07-19:
I went ahead and added CM3+ Support and released. If you are still not able to run Blinka on it, feel free to reopen, type `cat /proc/cpuinfo`, and let us know your Revision code and I'll add it.

closed 2019-07-19 by makermelissa

jm4r7in, 2019-07-20:
@makermelissa I saw, a bit late, that you published your commits.
Thank you very much, because I couldn't make it work on my own. 
:)