adafruit/Adafruit_Blinka #141: Got “NotImplementedError: Board not supported” on Jetson Nano

Hongbo-Miao, 2019-07-30:
Originally posted on [Stack Overflow](https://stackoverflow.com/questions/57278681/notimplementederror-board-not-supported).

___

I am trying to follow [this tutorial](https://learn.adafruit.com/reading-a-analog-in-and-controlling-audio-volume-with-the-raspberry-pi/script) to read analog input.

But instead of using Raspberry Pi, I am trying to use a Jetson Nano to read MCP3008 channel 0 analog input only.

I am following the pin layout in the tutorial on Jetson Nano since it shares same layout with Raspberry Pi.

    import busio
    import digitalio
    import board
    import adafruit_mcp3xxx.mcp3008 as MCP
    from adafruit_mcp3xxx.analog_in import AnalogIn
    
    # create the spi bus
    spi = busio.SPI(clock=board.SCK, MISO=board.MISO, MOSI=board.MOSI)
    
    # create the cs (chip select)
    cs = digitalio.DigitalInOut(board.D22)
    
    # create the mcp object
    mcp = MCP.MCP3008(spi, cs)
    
    # create an analog input channel on pin 0
    chan0 = AnalogIn(mcp, MCP.P0)
    
    print('Raw ADC Value: ', chan0.value)
    print('ADC Voltage: ' + str(chan0.voltage) + 'V')

>         import board
>       File "/usr/local/lib/python3.6/dist-packages/board.py", line 98, in <module>
>         raise NotImplementedError("Board not supported")
>     NotImplementedError: Board not supported

I checked the source code, it points to [this line](https://github.com/adafruit/Adafruit_Blinka/blob/a7bcbc4bc896176070770e9d7834402b4afa3020/src/board.py#L98).

However, I looked up of the **board** library source, it does support Jetson Nano because of the line

```
board_id == ap_board.JETSON_NANO
```

Any guide can help me further locate the issue will be very helpful. Thanks!

ladyada, 2019-07-30:
@tokk-nv hi please check this out and see why nvidia nano stopped working?

cscholler, 2019-08-23:
Greetings,

I am trying to follow this article (http://blog.whatgeek.com.pt/2015/04/raspberry-pi-and-adafruits-neopixel-stick/) but running into "Board not supported". I am using the Jetson Nano.
Here is the error below:

Exiting... 
Cleaning up pins
Traceback (most recent call last):
  File "/home/jetson/PycharmProjects/servoTest/venv/ledTest.py", line 3, in <module>
    from neopixel import *
  File "/home/jetson/PycharmProjects/servoTest/venv/lib/python3.6/site-packages/neopixel.py", line 34, in <module>
    from neopixel_write import neopixel_write
  File "/home/jetson/PycharmProjects/servoTest/venv/lib/python3.6/site-packages/neopixel_write.py", line 19, in <module>
    raise NotImplementedError("Board not supported")
NotImplementedError: Board not supported

Process finished with exit code 1

Here is a list of my packages:

Adafruit-Blinka | 2.3.1 | 2.3.1
-- | -- | --
Adafruit-GPIO | 1.0.3 | 1.0.3
Adafruit-PlatformDetect | 1.3.2 | 1.3.2
Adafruit-PureIO | 0.2.3 | 0.2.3
Jetson.GPIO | 2.0.0 | 2.0.0
RPi.GPIO | 0.7.0 | 0.7.0
adafruit-circuitpython-busdevice | 3.0.0 | 3.0.0
adafruit-circuitpython-motor | 2.0.0 | 2.0.0
adafruit-circuitpython-neopixel | 3.3.7 | 3.3.7
adafruit-circuitpython-pca9685 | 3.2.5 | 3.2.5
adafruit-circuitpython-register | 1.7.1 | 1.7.1
adafruit-circuitpython-servokit | 1.1.0 | 1.1.0
board | 0.0.0.post0 | 0.0.0.post0

Any help will greatly be appreciated.