adafruit/Adafruit_Blinka #143: Can you add Support for the Odroid N2.

rpineau, 2019-08-04:
I'm trying to use the i2c backpack Adafruit product id 292) with a RGB backlight negative LCD 16x2 (Adafruit product id 399) on an Odroid N2 but this SBC is not yet supported (Get the "NotImplementedError: Board not supported").
Thank you.

paulcrawford, 2019-08-05:
I have done the implementation of board and chip detection required for the Odroid N2 to be recognized by the CircuitPython code. My specific application was for the Adafruit GPIO Expander Bonnet for the Raspberry Pi (Adafruit product id 4132) but the new code should work with any Adafruit end use and the Odroid N2. I need to work with makermelissa to review my changes and additions before implementing a pull request for merging.

ladyada, 2019-08-05:
sweet!

makermelissa, 2019-08-05:
Nice! Would you like to submit a PR and I can review it?

paulcrawford, 2019-08-05:
@makermelissa, I would appreciate your help in developing the PR (I have actually never done it before). Before that is even started I would like you to review my two personal repositories, paulcrawford/Adafruit_Blinka and paulcrawford/Adafruit_Python_PlatformDetect, to ensure that the new code is OK. Essentially I simply copied the approach you used for the Odroid C2 and made the necessary changes for the Odroid N2. I have invited you to become a collaborator on my fork so you should easily be able to see the changes. Please advise if there are any issues with either viewing my fork or understanding the changes I made.

Thanks.

makermelissa, 2019-08-05:
Ok, I found the invitations. I wasn't sure what those were about when you sent them. Ok, first you'll want to pull in the latest changes to both of them as they're each 2 commits behind. Next, go to adafruit/Adafruit_Python_PlatformDetect and adafruit/Adafruit_Blinka andclick the New Pull Request button. Then click Compare across forks. On the right side, choose your fork and Create a Pull Request. That's really all that's needed.

rpineau, 2019-08-06:
thank you all for the quick response on this.

paulcrawford, 2019-08-06:
Rodolphe,

It was entirely coincidence that I just started working on this last week because I needed the code for the Adafruit Bonnet connected to my Odroid N2. When I completed the development and testing today, I thought I should check the “Issues” before advising Adafruit. Lo and behold, there was your request for the very same thing...

rpineau, 2019-08-06:
Yep. If nobody had replied I would probably have done it myself. I already started digging in the python code to see what needed to be done and though .. let's ask before I start doing something in case it's already being worked on.

makermelissa, 2019-08-06:
Added! It might take a few minutes for the release to update on PyPI.

closed 2019-08-06 by makermelissa

rpineau, 2019-08-07:
tested. works after installing libgpiod 1.1 from git to get the python bindings and reinstalling some Adafruit libs. I have a hello world on the LCD using the GPIOX_17 and GPIOX_18 for SDA,SCL
Thanks.

makermelissa, 2019-08-07:
Awesome!

paulcrawford, 2019-08-07:
Rodolphe,

That’s fantastic!

Melissa, 

Is there a place on the AdaFruit website that it can be made clear that the Python libraries for all sorts of AdaFruit devices now work with Odroid N2?

makermelissa, 2019-08-07:
Yep, I wrote a guide for adding it to circuitpython.org. https://learn.adafruit.com/how-to-add-a-new-board-to-the-circuitpython-org-website

danpe, 2023-11-19:
@makermelissa 
This warning is not correct then right?
[image]
https://learn.adafruit.com/circuitpython-libaries-linux-odroid-c2/initial-setup

The library also supports Odroid N2 and N2+ right?

paulcrawford, 2023-11-19:
Yes it does support both N2 and  N2+ as well as N2L. Sent from my iPhone

https://learn.adafruit.com/circuitpython-libaries-linux-odroid-c2/initial-setup

danpe, 2023-11-20:
TBH @paulcrawford I followed the guide and it did not work for me 😢
First of all `gpiod` doesn't support Python 3.11 which is installed by default now running `sudo apt-get install -y python3 git python3-pip`.

So I had to compile Python 3.8 manually.

After doing so, running blinkatest.py still fails for I2C hardware.

paulcrawford, 2023-11-21:
@Dan,

The following is my python3 version and blinkatest.py output:

***@***.***:~/src$ python3 -V
Python 3.10.12

***@***.***:~/src$ sudo python3 blinkatest.py
Hello blinka!
I2C ok!
Traceback (most recent call last):
  File "/home/daco/src/blinkatest.py", line 16, in <module>
    spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
  File "/usr/local/lib/python3.10/dist-packages/busio.py", line 301, in __init__
    self._spi = _SPI(portId)
  File "/usr/local/lib/python3.10/dist-packages/adafruit_blinka/microcontroller/generic_linux/spi.py", line 25, in __init__
    self._spi = spi.SPI(device=(portid, 0))
  File "/usr/local/lib/python3.10/dist-packages/Adafruit_PureIO/spi.py", line 167, in __init__
    raise IOError("{} does not exist".format(device))
OSError: /dev/spidev0.0 does not exist

I only run I2C on my N2L so there no SPI output.

Thanks,

Paul

danpe, 2023-11-21:
@paulcrawford did you try controlling WS281X LED strip with Adafruit library?