adafruit/Adafruit_Blinka #144: RFM9x Library Incompatible - SPI writebytes2, Pi 3

brentru, 2019-08-05:
I'm unable to run the following code on a Raspberry Pi 3 running Raspbian Buster:
https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/master/pi_radio/rfm9x_check.py

This change (writebytes -> writebytes2) seems to have broken it: https://github.com/adafruit/Adafruit_Blinka/commit/9b432b0734eb915686568a50942203065771a519

Full code with traceback:
```
>>> CS = DigitalInOut(board.CE1)
>>> RESET = DigitalInOut(board.D25)
>>> spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
>>> rfm-x = adafruit_rfm9x
KeyboardInterrupt
>>> rfm = adafruit_rfm9x.RFM9x(spi, CS, RESET, 915.0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/adafruit_rfm9x.py", line 365, in __init__
    version = self._read_u8(_RH_RF95_REG_42_VERSION)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_rfm9x.py", line 415, in _read_u8
    self._read_into(address, self._BUFFER, length=1)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_rfm9x.py", line 410, in _read_into
    device.write(self._BUFFER, end=1)
  File "/home/pi/.local/lib/python3.7/site-packages/busio.py", line 138, in write
    return self._spi.write(buf, start, end)
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_blinka/microcontroller/generic_linux/spi.py", line 56, in write
    self._spi.writebytes2(buf[start:end])
AttributeError: 'SpiDev' object has no attribute 'writebytes2'
```

jerryneedell, 2019-08-05:
hmm -- works for me on a Raspberry Pi ZeroW with Buster

brentru, 2019-08-05:
@jerryneedell  on Blinka 2.2.1?

jerryneedell, 2019-08-05:
ah -- was 2.2.0 not sure I want to update ;-)

makermelissa, 2019-08-05:
Ok, good to know. I think we should place it in a try block so that it will fall back to writebytes

jerryneedell, 2019-08-05:
hmm -- seems to be there: https://github.com/doceme/py-spidev/blob/master/spidev_module.c#L416

caternuson, 2019-08-05:
Check your version of spidev. Looks like `writebytes2` was added in 3.4.
```console
pi@raspberrypi:~ $ python3 -m pip show spidev
Name: spidev
Version: 3.4
Summary: Python bindings for Linux SPI access through spidev
Home-page: http://github.com/doceme/py-spidev
Author: Volker Thoms
Author-email: unconnected@gmx.de
License: GPLv2
Location: /home/pi/.local/lib/python3.7/site-packages
Requires: 
Required-by: unicornhathd
pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import spidev
>>> spi = spidev.SpiDev()
>>> dir(spi)
['__class__', '__delattr__', '__dir__', '__doc__', '__enter__', '__eq__', '__exit__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', 'bits_per_word', 'close', 'cshigh', 'fileno', 'loop', 'lsbfirst', 'max_speed_hz', 'mode', 'no_cs', 'open', 'readbytes', 'threewire', 'writebytes', 'writebytes2', 'xfer', 'xfer2', 'xfer3']
>>>
```

jerryneedell, 2019-08-05:
I have spider 3.4 -- I just updated to blinka 2.2.1 and it still works - yay!

brentru, 2019-08-05:
Updated to spidev 3.4, works now.

ladyada, 2019-08-05:
hiya you can put the rev-need in the requirements.txt :)

makermelissa, 2019-08-05:
Ok, I'll look up how to do that.

PR #145 (2019-08-23): Update requirements for SpiDev >= 3.4
This fixes #144.

I'm not sure if this is the best approach or if we should just tell the user to update SPIDEV to latest version in the error.
closed 2019-08-23 by makermelissa