adafruit/Adafruit_Blinka #149: rpi_ws281x stopping installation of Blinka on NVidia Jetson TX2

CodeMaterial, 2019-08-19:
@makermelissa Creating a new issue here off the back of 
https://github.com/adafruit/Adafruit_CircuitPython_BNO055/pull/35

Here are my following attempts at installing the Blinka / BNO055 module.

I've truncated a lot of the outputs to just the errors which I hope is okay.

I'm really not sure what's going on but it's not critical for me. Just in case other people were having similar issues as it is more than likely the errors are due to how I've set up my Jetson.

I've tried the below in and out of a venv with the same issues.

All the best and thank you for your time,

Sam.

```
(myenv) ~/git/Adafruit_CircuitPython_BNO055$ pip3 install adafruit-circuitpython-bno055

Building wheels for collected packages: rpi-ws281x
  Building wheel for rpi-ws281x (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/nvidia/git/Adafruit_CircuitPython_BNO055/myenv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-7qbin8mm/rpi-ws281x/setup.py'"'"'; __file__='"'"'/tmp/pip-install-7qbin8mm/rpi-ws281x/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-s2j9eyn3 --python-tag cp35
       cwd: /tmp/pip-install-7qbin8mm/rpi-ws281x/
  Complete output (18 lines):
  running bdist_wheel
  running build
  running build_py
  Compiling ws281x library...
  b'lib/rpihw.c: In function \xe2\x80\x98rpi_hw_detect\xe2\x80\x99:\nlib/rpihw.c:398:5: error: ignoring return value of \xe2\x80\x98fread\xe2\x80\x99, declared with attribute warn_unused_result [-Werror=unused-result]\n     fread(&rev, sizeof(uint32_t), 1, f);\n     ^\ncc1: all warnings being treated as errors\nmake: *** [lib-built/rpihw.o] Error 1\n'
  creating build
  creating build/lib.linux-aarch64-3.5
  creating build/lib.linux-aarch64-3.5/rpi_ws281x
  copying rpi_ws281x/__init__.py -> build/lib.linux-aarch64-3.5/rpi_ws281x
  copying rpi_ws281x/rpi_ws281x.py -> build/lib.linux-aarch64-3.5/rpi_ws281x
  running build_ext
  building '_rpi_ws281x' extension
  creating build/temp.linux-aarch64-3.5
  aarch64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Ilib/ -I/usr/include/python3.5m -I/home/nvidia/git/Adafruit_CircuitPython_BNO055/myenv/include/python3.5m -c rpi_ws281x_wrap.c -o build/temp.linux-aarch64-3.5/rpi_ws281x_wrap.o
  aarch64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-aarch64-3.5/rpi_ws281x_wrap.o -Llib-built/ -lws2811 -o build/lib.linux-aarch64-3.5/_rpi_ws281x.cpython-35m-aarch64-linux-gnu.so
  /usr/bin/ld: cannot find -lws2811
  collect2: error: ld returned 1 exit status
  error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for rpi-ws281x
  Running setup.py clean for rpi-ws281x

```
```
(myenv) ~/git/Adafruit_CircuitPython_BNO055$ sudo python setup.py install
[success]
```
```
(myenv) ~/git/Adafruit_CircuitPython_BNO055$ python examples/bno055_simpletest.py 
Traceback (most recent call last):
  File "examples/bno055_simpletest.py", line 2, in <module>
    import board
ImportError: No module named 'board'
```
```
(myenv) ~/git/Adafruit_CircuitPython_BNO055$ sudo pip3 install adafruit-circuitpython-bno055
Installing collected packages: rpi-ws281x
  Running setup.py install for rpi-ws281x ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-ccj9mymh/rpi-ws281x/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-f9kd9sro-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    Compiling ws281x library...
    b'lib/rpihw.c: In function \xe2\x80\x98rpi_hw_detect\xe2\x80\x99:\nlib/rpihw.c:398:5: error: ignoring return value of \xe2\x80\x98fread\xe2\x80\x99, declared with attribute warn_unused_result [-Werror=unused-result]\n     fread(&rev, sizeof(uint32_t), 1, f);\n     ^\ncc1: all warnings being treated as errors\nmake: *** [lib-built/rpihw.o] Error 1\n'
    creating build
    creating build/lib.linux-aarch64-3.5
[... 48 lines cut ...]

  Running setup.py install for rpi-ws281x ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-39k7960y/rpi-ws281x/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-tcf_ti30-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    Compiling ws281x library...
    b'lib/rpihw.c: In function \xe2\x80\x98rpi_hw_detect\xe2\x80\x99:\nlib/rpihw.c:398:5: error: ignoring return value of \xe2\x80\x98fread\xe2\x80\x99, declared with attribute warn_unused_result [-Werror=unused-result]\n     fread(&rev, sizeof(uint32_t), 1, f);\n     ^\ncc1: all warnings being treated as errors\nmake: *** [lib-built/rpihw.o] Error 1\n'
    creating build
    creating build/lib.linux-aarch64-3.5
    creating build/lib.linux-aarch64-3.5/rpi_ws281x
    copying rpi_ws281x/__init__.py -> build/lib.linux-aarch64-3.5/rpi_ws281x
    copying rpi_ws281x/rpi_ws281x.py -> build/lib.linux-aarch64-3.5/rpi_ws281x
    running build_ext
    building '_rpi_ws281x' extension
    creating build/temp.linux-aarch64-3.5
    aarch64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Ilib/ -I/usr/include/python3.5m -c rpi_ws281x_wrap.c -o build/temp.linux-aarch64-3.5/rpi_ws281x_wrap.o
    aarch64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-aarch64-3.5/rpi_ws281x_wrap.o -Llib-built/ -lws2811 -o build/lib.linux-aarch64-3.5/_rpi_ws281x.cpython-35m-aarch64-linux-gnu.so
    /usr/bin/ld: cannot find -lws2811
    collect2: error: ld returned 1 exit status
    error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-39k7960y/rpi-ws281x/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-tcf_ti30-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-39k7960y/rpi-ws281x/
```

swarren, 2019-08-21:
This issue doesn't happen for me.

Are you using the sample root filesystem that's shipped with L4T/JetPack, or are you using a custom root filesystem? Adafruit_Blinka depends on rpi_ws281x ONLY on armv6l or armv7l, whereas a default Jetson system is aarch64. Perhaps you're using a 32-bit root filesystem or 32-bit Python binary? If so, that's the root cause of the issue.

(Of course, an alternative fix might be to fix the rpi_ws281x library's compilation to succeed with the compilers provided by L4T, but that shouldn't be necessary since the library shouldn't be installed in the first place. Still, such a fix would be useful for if/when the RPi upgrades its compilers.)

swarren, 2019-08-21:
https://github.com/jgarff/rpi_ws281x/pull/370 should fix the compilation error, once it's merged into Pimoroni's fork of rpi_ws281x.

ladyada, 2019-08-21:
hiya...gcc 5 seems really old. even on microcontrollers we're up to v7. why is it so old on the nvidia? :/

swarren, 2019-08-21:
Oops. I ran `gcc --version` on the wrong system (my Ubuntu 16.04 development machine). Jetson has Ubuntu 18.04 with gcc 7.4.0.

ladyada, 2019-08-21:
dya mind trying to update to gcc 8 to match Buster? just cause...im not sure if/when we'll get that PR merged :/

swarren, 2019-08-21:
L4T ships an essentially unmodified Ubuntu 18.04 as its sample root filesystem. We are at the mercy of whatever compiler Canonical/Ubuntu chooses. I could maybe try a different gcc locally out of curiosity, but our users aren't going to be upgraded until if/when we decide to switch our sample root filesystem to a new Ubuntu release (which I'm sure wouldn't be at least until 20.04).

ladyada, 2019-08-21:
well, we'll see what we can do open source is aWeZome :)

ladyada, 2019-08-21:
@makermelissa im tempted to say we should drop rpi-ws from requirements if there's no way to install it only for Pi's

makermelissa, 2019-08-21:
I think it's possible to specify it by processor or architecture so that it only installs on Raspberry Pi Broadcom chips.

ladyada, 2019-08-21:
lets!

makermelissa, 2019-08-21:
If that doesn't work, we can just update the guides with another step.

ladyada, 2019-08-21:
agreed!

swarren, 2019-08-21:
As far as I can tell from reading setup.py/requirements.txt, the ws281x library IS already only installed on Pis and not Jetson (since Pis use 32-bit SW and Jetson 64-bit by default). I can't work out why it is getting installed on Jetson for @CodeMaterial. Perhaps a 32-bit-vs-64-bit issue that I asked about in my earlier comment?

makermelissa, 2019-08-21:
Oh yeah, you're right. It's already set to install only on armv7l or armv6l. I'm not sure if we can get more specific than that because it would still install on lots of non-RPi systems.

CodeMaterial, 2019-08-23:
I'm going to prefix this with the fact that I am now completely out of my depth with this.

I think it must just be something buggy with my system if @swarren isn't having any issues.

```
~$ python
Python 2.7.12 (default, Nov 12 2018, 14:36:49) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import platform, os
>>> platform.architecture()
('64bit', 'ELF')
>>> os.uname()
('Linux', 'tegra-ubuntu', '4.4.38-tegra-leopard', '#2 SMP PREEMPT Wed Oct 25 12:11:46 PDT 2017', 'aarch64')
```
```
~$ python3
Python 3.5.2 (default, Nov 12 2018, 13:43:14) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import platform, os
>>> platform.architecture()
('64bit', 'ELF')
>>> os.uname()
posix.uname_result(sysname='Linux', nodename='tegra-ubuntu', release='4.4.38-tegra-leopard', version='#2 SMP PREEMPT Wed Oct 25 12:11:46 PDT 2017', machine='aarch64')
``

CodeMaterial, 2019-08-23:
Ah I've just realised I'm running 16.04.

```
~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04 LTS
Release:	16.04
Codename:	xenial
```

```
~$ gcc --version
gcc (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

swarren, 2019-08-23:
Ah yes, we've only tested Jetson.GPIO and Blinka on L4T r32.x. It sounds like you're running a prior release; r28 perhaps? You can check using `head -1 /etc/nv_tegra_release`.

swarren, 2019-08-24:
Also, what version of pip do you have (run `pip --version`), and are you using a Python virtual environment or not (if so, please activate it before querying the pip version). Thanks.

closed 2019-09-11 by makermelissa

CodeMaterial, 2019-09-16:
Hi @makermelissa.

Just to update you here, I've managed to install CircuitPython so thank your work.

All the best,

Sam.