adafruit/Adafruit_Blinka #150: Pi 4B  "RuntimeError: Not running on a RPi!" coming from RPi.GPIO

Eirikur, 2019-08-23:
Hmmm. The latest version of RPi.GPIO (0.7.0) claims to be updated for Pi4B.  What's really annoying is that the bug appears to be in the Cython binary. So it's probably not your fault at all, but I can't fix it. You're requiring this other library RPi.GPIO and it's failing. I'm a developer and happy to help.

eh@pi4 ~/eyelert $ python3 leds.py 
Traceback (most recent call last):
  File "leds.py", line 29, in <module>
    pixel_order=ORDER)
  File "/usr/local/lib/python3.7/dist-packages/neopixel.py", line 91, in __init__
    self.pin = digitalio.DigitalInOut(pin)
  File "/usr/local/lib/python3.7/dist-packages/digitalio.py", line 75, in __init__
    self.direction = Direction.INPUT
  File "/usr/local/lib/python3.7/dist-packages/digitalio.py", line 101, in direction
    self._pin.init(mode=Pin.IN)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pin.py", line 32, in init
    GPIO.setup(self.id, GPIO.IN)
RuntimeError: Not running on a RPi!

ladyada, 2019-08-23:
hihi, thats definitely a new error 
can you provide a minimal example that breaks, taht's just the GPIO library? e.g. sepearate from blinka
also maybe you can open up a ticket here?
https://sourceforge.net/p/raspberry-gpio-python/tickets/

Eirikur, 2019-08-23:
Sounds like a good plan, Limor. Will do in a bit.

ladyada, 2019-09-05:
did this get any resolution? i think you're the only person to bring this up, makes us think something else is amiss

Eirikur, 2019-09-05:
Ah, I think I sent email but didn't report back here. I reported it against the Sourceforge project. I will retest with a fully updated Raspbian and report back tonight.  I was a very early recipient of Pi4s.

ladyada, 2019-09-05:
if you got a pre-release board you probably have a unique/unknown/unburnt ID number which is why you'd be seeing this error - not much we can do about it, could pick up a new PI 4 :)

Eirikur, 2019-09-05:
Not pre-release, but ordered on first day! Mine are real Pis. I have a herd of them now.
It's still happening.
Code:
# Minimal code example. Happens with Python2 and Python3.
# Looks to be in the Cython binary.
# Produces the message "Not running on an RPi" on the Pi4B.

import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM) # Use BCM pins D4 = GPIO #4
GPIO.setwarnings(True) # shh!
id = 18 # BCM GPIO 18 I assume this doesn't matter because we don't get far.
GPIO.setup(id, GPIO.IN) # Hits exception: RuntimeError: Not running on a RPi!
# That's all folks!  We don't get this far.

Terminal session:
Raspbian GNU/Linux 10 (buster)
Thu Sep  5 18:33:17 EDT 2019
eh@pi4 ~ $ python3 not-running-on-a-pi.py 
Traceback (most recent call last):
  File "not-running-on-a-pi.py", line 11, in <module>
    GPIO.setup(id, GPIO.IN) # Hits exception: RuntimeError: Not running on a RPi!
RuntimeError: Not running on a RPi!
eh@pi4 ~ $

makermelissa, 2019-09-05:
I just tested this on a fresh Blinka install on the RPi 4B and do not get this error. Please try running:
```
pip3 install --upgrade adafruit_blinka
```
and then try again.

Eirikur, 2019-09-05:
It goes away when I use sudo. Since I haven't done anything with setuid or groups, I use sudo in production (embedded wearable device). So, this can be closed as it is basically a user error, but a horrid error message. I manually removed the underlying rpi.gpio and reinstalled it after a reinstall of adafruit_blinka.  rpi.gpio isn't a very nice package. It doesn't have a __version__ attribute. Sigh. Anyway, this was never a blocker for me and it's just an error message problem in a package that is not yours.

ladyada, 2019-09-05:
ok - thanks for the report, i think we'll close because there's no fix we can perform, hopefully will slowly be fixed by our partner packages :)

closed 2019-09-05 by ladyada

doctorbigtime, 2020-01-05:
In case other people run into this, I also had this problem with an Arch Linux ARM installation on a Pi4. It was resolved by changing the rights on /dev/gpiomem to be readable/writable to the RPi.GPIO user (for a more permanent solution, changing the udev rules).

robin48gx, 2020-01-09:
Also add the user to the gpio group:

sudo adduser <username> gpio

adding your user to the spi group is also useful.

ehrenbrav, 2020-01-13:
The previous comment solves the issue. After making the change to groups, log out and log in again.

LongLiveCHIEF, 2020-04-04:
On some distros, there may not be a gpio group, in which case /dev/gpiomem would be owned by root instead of gpio. (for example, ubutu server for rpi 3+)

In this case, you'll need to add a gpio group, add your user to the gpio group, and then create a udev rule for gpio.  The RPI.GPIO library has a [good python script for this](https://sourceforge.net/p/raspberry-gpio-python/code/ci/default/tree/create_gpio_user_permissions.py), and you just need to change the name of the user from 'pi' to whatever your user is

Eminlin, 2020-06-17:
> Also add the user to the gpio group:
> 
> sudo adduser username gpio
> 
> adding your user to the spi group is also useful.

it works for me

LongLiveCHIEF, 2020-06-17:
The other thing I've discovered, is that if you're using ubuntu server image for rpi4, is that it uses the new character device library for gpio instead of the sysfs library that RPi.GPIO uses. 

If this is the case, you'll also need different set of udev rules than those used in the past. I wrote a python script that will add the appropriate udev rules based on a user/group you can specify with args.  https://github.com/LongLiveCHIEF/docker-pi-gpio/blob/master/gpio_udev_setup.py

mruz, 2020-06-23:
On openSUSE I don't have `/dev/gpiomem` at all, so RPi.GPIO throws `Not running on a RPi!`. Any idea how to add it?

geoffschultz, 2021-01-02:
I'm running into s similar variant of this problem.  I just got a Raspberry Pi 4 and have the latest version of NOOBgpio installed with the Apache web server. I have CGI enabled in Apache and Python 3 installed.  I have a Python program which reads the temperature from a DHT11.  This code works fine when I run it from command line or Thonny.  However, when it runs via CGI, I get the "Not running on a RPi!" error.

Any help getting this to work would be appreciated.

geoffschultz, 2021-01-02:
To provide a somewhat answer to my own problem, I found that doing a "sudo chmod og+rwx /dev/gpio*" solved my problem.  However, I suspect that this isn't exactly the best solution.

combrs, 2021-01-16:
On Raspbian 10.7 adding user to gpio group definitely helps, because 
```
crw-rw---- 1 root gpio 246, 0 Jan 16 18:17 /dev/gpiomem
```
But error message is misleading, and do not show real problem source.

MarkProminic, 2021-04-09:
I was able to fix the  pip3 install RPi.gpio with this:

export CFLAGS=-fcommon
pip3 install RPi.GPIO

https://www.raspberrypi.org/forums/viewtopic.php?t=289084

mathieuMorichon, 2021-04-09:
Hi,
If you run your python script through a apache process (ie: a web page calls php, and php calls python ...), you may add the "**www-data**" user to the **gpio** group. You can see that error if the script runs well in command line (ie: with "pi" or "root" user) but fails in case of a web call (with the "Not running on a RPi" error). 
To make apache belongs to the correct group, run the following command as root user (on the raspberry-pi) :
`root@raspberrypi:~# adduser www-data gpio`

To check (you can see that the gpio group has been added) : 
`root@raspberrypi:~# groups www-data
www-data : www-data gpio`

Then restart the Apache2 process (or reboot the raspberry-pi) and the error will be fixed.

jeison1946, 2024-01-31:
To solve this error the following way worked for me:

`sudo chmod 660 /dev/gpiomem`

`sudo chown root:gpio /dev/gpiomem`