adafruit/Adafruit_Blinka #153: raspi_1a support (26 pin model a)

schustykd, 2019-09-07:
looking for help with blinka support for board id: RASPBERRY_PI_A (26 pin, raspberry pi 1, model a, 2011.12).

am thinking that it is not currently supported, but i'm new at this, so would appreciate advice if i'm missing something.

detect.py correctly identifies my board as RASPBERRY_PI_A but blinkatest.py fails with "Board not supported".

thanks in advance.

makermelissa, 2019-09-11:
Thanks for posting this @schustykd. I'm gonna try and trace down why it's not working on the Pi 1 A in the code, but I don't currently have one so I may need you to test it.

schustykd, 2019-09-11:
I can do that, no problem.  Let me know how I can help.

makermelissa, 2019-09-11:
I have a PR #155 in if you want to try that out.

makermelissa, 2019-09-12:
Ok, it was merged and I just pushed a release. Please test at your convenience.

makermelissa, 2019-09-12:
Sounds good. If you run into other issues, let’s open a new request.

closed 2019-09-12 by makermelissa