adafruit/Adafruit_Blinka #156: Failed to access I2C on NVidia Jetson TX2

CodeMaterial, 2019-09-16:
Good Morning, me again!

I've now managed to install Adafruit_Blinka thanks to the resolution in issue #149

However, I've been hitting the same errors when trying to use the CircuitPython libraries for the BNO055, FXOS8700 and FXAS21002C boards.
I've been mainly following this tutorial here: https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano/initial-setup

as it's the closest I could find to the NVidia Jetson TX2
Blinka seems to install correctly and the libraries I'm using also install correctly but on running any of their test scripts with the `import board` statement, I get the following errors.

Any help would be apreciated but again, if this is only happening to me then I wouldn't be suprised if there was something odd going on with my system and the library is fine how it is.

All the best,

Sam.
```
~/git/Adafruit_CircuitPython_FXOS8700/examples$ python3 fxos8700_simpletest.py 
Traceback (most recent call last):
  File "fxos8700_simpletest.py", line 5, in <module>
    import board
  File "/home/nvidia/.local/lib/python3.5/site-packages/board.py", line 80, in <module>
    from adafruit_blinka.board.jetson_tx2 import *
  File "/home/nvidia/.local/lib/python3.5/site-packages/adafruit_blinka/board/jetson_tx2.py", line 3, in <module>
    from adafruit_blinka.microcontroller.tegra.t186 import pin
  File "/home/nvidia/.local/lib/python3.5/site-packages/adafruit_blinka/microcontroller/tegra/t186/pin.py", line 3, in <module>
    import Jetson.GPIO as GPIO
  File "/home/nvidia/.local/lib/python3.5/site-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/home/nvidia/.local/lib/python3.5/site-packages/Jetson/GPIO/gpio.py", line 33, in <module>
    raise RuntimeError("The current user does not have permissions set to "
RuntimeError: The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this
```
And when run with sudo
```
~/git/Adafruit_CircuitPython_FXOS8700/examples$ sudo python3 fxos8700_simpletest.py 
Traceback (most recent call last):
  File "fxos8700_simpletest.py", line 5, in <module>
    import board
  File "/home/nvidia/.local/lib/python3.5/site-packages/board.py", line 80, in <module>
    from adafruit_blinka.board.jetson_tx2 import *
  File "/home/nvidia/.local/lib/python3.5/site-packages/adafruit_blinka/board/jetson_tx2.py", line 3, in <module>
    from adafruit_blinka.microcontroller.tegra.t186 import pin
  File "/home/nvidia/.local/lib/python3.5/site-packages/adafruit_blinka/microcontroller/tegra/t186/pin.py", line 3, in <module>
    import Jetson.GPIO as GPIO
  File "/home/nvidia/.local/lib/python3.5/site-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/home/nvidia/.local/lib/python3.5/site-packages/Jetson/GPIO/gpio.py", line 70, in <module>
    model, JETSON_INFO, _channel_data_by_mode = gpio_pin_data.get_data()
  File "/home/nvidia/.local/lib/python3.5/site-packages/Jetson/GPIO/gpio_pin_data.py", line 270, in get_data
    for fn in os.listdir(gpio_chip_gpio_dir):
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/3160000.i2c/i2c-0/0-0074/gpio'
```

On upgrading adafruit_blinka:
```
~$ sudo -H pip3 install --upgrade adafruit_blinka
Requirement already up-to-date: adafruit_blinka in /usr/local/lib/python3.5/dist-packages (2.4.1)
Requirement already satisfied, skipping upgrade: spidev in /usr/local/lib/python3.5/dist-packages/spidev-3.4-py3.5-linux-aarch64.egg (from adafruit_blinka) (3.4)
Requirement already satisfied, skipping upgrade: sysv-ipc in /usr/local/lib/python3.5/dist-packages/sysv_ipc-1.0.0-py3.5-linux-aarch64.egg (from adafruit_blinka) (1.0.0)
Requirement already satisfied, skipping upgrade: Jetson.GPIO in /usr/local/lib/python3.5/dist-packages (from adafruit_blinka) (2.0.1)
Requirement already satisfied, skipping upgrade: Adafruit-PureIO in /usr/local/lib/python3.5/dist-packages/Adafruit_PureIO-0.2.3-py3.5.egg (from adafruit_blinka) (0.2.3)
Requirement already satisfied, skipping upgrade: Adafruit-PlatformDetect in /usr/local/lib/python3.5/dist-packages (from adafruit_blinka) (1.3.3)
```
on checking the i2c rail, just in case...
```
~/git/Adafruit_CircuitPython_FXOS8700/examples$ sudo i2cdetect -r -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 1f 
20: -- 21 -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
[... 77 lines cut ...]
Searching for sysv-ipc==1.0.0
Best match: sysv-ipc 1.0.0
Processing sysv_ipc-1.0.0-py3.5-linux-aarch64.egg
sysv-ipc 1.0.0 is already the active version in easy-install.pth

Using /usr/local/lib/python3.5/dist-packages/sysv_ipc-1.0.0-py3.5-linux-aarch64.egg
Searching for Adafruit-PlatformDetect==1.3.3
Best match: Adafruit-PlatformDetect 1.3.3
Adafruit-PlatformDetect 1.3.3 is already the active version in easy-install.pth

Using /usr/local/lib/python3.5/dist-packages
Searching for Adafruit-PureIO==0.2.3
Best match: Adafruit-PureIO 0.2.3
Processing Adafruit_PureIO-0.2.3-py3.5.egg
Adafruit-PureIO 0.2.3 is already the active version in easy-install.pth

Using /usr/local/lib/python3.5/dist-packages/Adafruit_PureIO-0.2.3-py3.5.egg
Searching for spidev==3.4
Best match: spidev 3.4
Processing spidev-3.4-py3.5-linux-aarch64.egg
spidev 3.4 is already the active version in easy-install.pth

Using /usr/local/lib/python3.5/dist-packages/spidev-3.4-py3.5-linux-aarch64.egg
Finished processing dependencies for adafruit-circuitpython-bno055==3.1.1.dev2+g7269b86
```

swarren, 2019-09-16:
@CodeMaterial you'll need to follow the instructions here:
https://github.com/NVIDIA/jetson-gpio/#setting-user-permissions

swarren, 2019-09-16:
The previous comment was to solve the permission errors when not running as `root`. Could you please run the following commands to diagnose the issue when running as `root`:

```
ls -l /sys/devices
ls -l /sys/devices/3160000.i2c
```

Thanks.

makermelissa, 2019-09-17:
The guide has been updated with the instructions added

ladyada, 2019-09-26:
hiya closing cause no response from OP - reopen if still an issue!

closed 2019-09-26 by ladyada

codemaster97, 2019-10-03:
I also have this issue on the Jetson TX2. Note that I am using an Orbitty Carrier board from Connecttech, but the issue is exactly the same. In 

`ls /sys/devices/3160000.i2c/i2c-0`

There is no `0-0074`

swarren, 2019-10-03:
At least at present, Jetson.GPIO only supports the Jetson TX2 developer kit, and not the Jetson TX2 module installed in an alternative baseboard.

It might be best to file a new bug against Jetson.GPIO with title something like "Orbitty baseboard for Jetson TX2 not supported". At least initially, this bug is an issue in Jetson.GPIO. Some work might be required on Adafruit Blinka after fixing that; I'd have to check.

In that other bug, can you please tell me:
1) The URL of the code (L4T release) you downloaded to flash the system.
2) The flash.sh command you used to flash the system.
3) Do you know where I can get a schematic for the Orbitty baseboard?

Thanks.

swarren, 2019-10-03:
For my later reference: I haven't found Orbitty's schematics, but the following page lists the GPIO mapping:
http://connecttech.com/resource-center/kdb342-using-gpio-connect-tech-jetson-tx1-carriers/

I note that Orbitty doesn't have a 40-pin GPIO expansion connector, but rather, something custom. This might make it difficult/impossible to fit into Jetson.GPIO's pin naming convention.

swarren, 2019-10-04:
I've released Jetson.GPIO v2.0.2. This detects when the carrier board isn't the carrier in the Jetson Developer Kit and prints a warning. I've been directed not to proactively add support for 3rd-party carrier boards into Jetson.GPIO; vendors can submit patches to add support if they wish.