adafruit/Adafruit_Blinka #159: have a problem useing PWM on pocketbeagle

fclyyxy1999, 2019-09-23:
**Board:** pocket beagle
Test:  Test Servo on pocket beagle,which should use PWM
**Code:**`
import time

import board

import pulseio

from adafruit_motor import servo

# pwm = pulseio.PWMOut(board.P1_36,duty_cycle=(2 ** 15),frequency=50)
freqency = (6000 * 1000)  # 6MHz

pwm = pulseio.PWMOut(board.P1_36,duty_cycle=(2 ** 15),frequency=freqency,variable_frequency=True)

my_servo = servo.Servo(pwm)

while True:
    for angle in range(0, 180, 5):
        my_servo.angle = angle
        time.sleep(0.05)
    for angle in range(180, 0, -5):
        my_servo.angle = angle
        time.sleep(0.05)`

**Error: AttributeError: module 'pulseio' has no attribute 'PWMOut'.**

I have seen a similar test on https://github.com/s-light/pocketbeagle_python_tests/blob/master/cp_blinka/blinka_pwm_test.py
And I can use PWM on https://github.com/adafruit/adafruit-beaglebone-io-python successfully.

Is it because there is no pocket beagle PWM support？ I saw #101 but I couldn't find any details.

I'm really confused about these Adafruit libraries.Either python3 or I2C is not supported.Could you please help me solve this problem?

Sincerely thanks!!

pdp7, 2019-10-07:
Sorry, I missed this when it was assigned.  I'll take a look and update.

pdp7, 2019-10-08:
I was able to reproduce that error with `cp_blinka/blinka_pwm_test.py` by @s-light:  
```
debian@beaglebone:~/pocketbeagle_python_tests$ sudo python3 cp_blinka/blinka_test.py 
Hello blinka!
Digital IO ok!
I2C ok!
I2C_2 ok!
SPI ok!
SPI_1 ok!
done!
debian@beaglebone:~/pocketbeagle_python_tests$ sudo python3 cp_blinka/blinka_pwm_test.py 
******************************************
test for pwm frequency
Traceback (most recent call last):
  File "cp_blinka/blinka_pwm_test.py", line 14, in <module>
    pwm = pulseio.PWMOut(
AttributeError: module 'pulseio' has no attribute 'PWMOut'
```

pdp7, 2019-10-25:
@makermelissa @ladyada @sommersoft @Groguard:
I'm looking at the Blink implementation of PWM output for generic linux boards:
https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/generic_linux/sysfs_pwmout.py#L26

The naming scheme used does not match what the BeagleBone/PocketBeagle uses so I'm trying to figure out how to handle that.

For example, PocketBeagle has:
```
/sys/class/pwm/pwmchip0/pwm-0:0/period
/sys/class/pwm/pwmchip0/pwm-0:1/period
/sys/class/pwm/pwmchip2/pwm-2:0/period
/sys/class/pwm/pwmchip2/pwm-2:1/period
/sys/class/pwm/pwmchip4/pwm-4:0/period
/sys/class/pwm/pwmchip4/pwm-4:1/period
```

This is a result of this kernel patch that makes the pwm naming scheme work with uevents so that udev can properly set the permissions.  Here is the patch that @robertcnelson is applying to the BeagleBoard.org kernel:
https://github.com/RobertCNelson/bb-kernel/blob/am33x-v4.11/patches/drivers/pwm/0001-pwm-Create-device-class-for-pwm-channels.patch
> Pwm channels don't send uevents when exported, this change adds the channels to a pwm class and set their device type to pwm_channel so uevents are sent. 
> To do this properly, the device names need to change to uniquely identify a channel.
> This change is from pwmN to pwm-(chip->base):N

This does not match the pattern the [Blinka code is looking for](https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/generic_linux/sysfs_pwmout.py#L26
):
```
class PWMOut(object):
    # Sysfs paths
    _sysfs_path = "/sys/class/pwm/"
    _channel_path = "pwmchip{}"

    # Channel paths
    _export_path = "export"
    _unexport_path = "unexport"
    _pin_path = "pwm{}"
```
I could add code to detect the board type and use a different pattern if Beagle but I wanted to check if there is a better approach that I am not thinking of.

makermelissa, 2019-10-25:
You could copy the PWMOut file into the microprocessor folder for the BBB and modify it there. I think that's basically what we did when stuff differed before.

ladyada, 2019-10-25:
i agree with mel! the 'generic sysfs PWM' is for simple implementations, its fine to fork that to your board def :)

pdp7, 2019-10-25:
@makermelissa thanks.  so I think I would change https://github.com/adafruit/Adafruit_Blinka/blob/master/src/pulseio.py 
to add:
```
if detector.board.any_beagle:
    from adafruit_blinka.microcontroller.am335x.sysfs_pwmout import PWMOut
```
However, wouldn't I need to copy all the code in `src/adafruit_blinka/microcontroller/generic_linux/sysfs_pwmout.py` to a new file: `src/adafruit_blinka/microcontroller/am335x/sysfs_pwmout.py`?

That seems like I would then create the situation where I've essentially forked the file and it could fall out of sync.  I really just want change:
```
debian@beaglebone:~/Adafruit_Blinka$ grep -n _pin_path src/adafruit_blinka/microcontroller/generic_linux/sysfs_pwmout.py 
26:    _pin_path = "pwm{}"
81:        pin_path = os.path.join(channel_path, self._pin_path.format(self._pwmpin))
133:            self._pin_path.format(self._pwmpin),
147:            self._pin_path.format(self._pwmpin),
```
Would maybe be better to add logic to handle this other naming pattern in `src/adafruit_blinka/microcontroller/generic_linux/sysfs_pwmout.py`?

pdp7, 2019-10-25:
@makermelissa @ladyada  thanks.  so I think I would change https://github.com/adafruit/Adafruit_Blinka/blob/master/src/pulseio.py 
to add:
```
if detector.board.any_beagle:
    from adafruit_blinka.microcontroller.am335x.sysfs_pwmout import PWMOut
```
However, I will need to copy `src/adafruit_blinka/microcontroller/generic_linux/sysfs_pwmout.py` to a new file: `src/adafruit_blinka/microcontroller/am335x/sysfs_pwmout.py`?

That seems like I would then create the situation where I've essentially forked the file and it could fall out of sync.  I really just need to change:
```
debian@beaglebone:~/Adafruit_Blinka$ grep -n _pin_path src/adafruit_blinka/microcontroller/generic_linux/sysfs_pwmout.py 
26:    _pin_path = "pwm{}"
81:        pin_path = os.path.join(channel_path, self._pin_path.format(self._pwmpin))
133:            self._pin_path.format(self._pwmpin),
147:            self._pin_path.format(self._pwmpin),
```
Would maybe be better to add logic to handle this other naming pattern in `src/adafruit_blinka/microcontroller/generic_linux/sysfs_pwmout.py`?

makermelissa, 2019-10-25:
Yes, that sounds right on the changes. You could always inherit from the generic class and add your overriding functions

pdp7, 2019-10-26:
I have created PR #168 which adds PWMOut support for the PocketBeagle

pdp7, 2019-12-01:
@fclyyxy1999 are you able to successfully use PWM on the PocketBeagle now?

makermelissa, 2020-05-07:
Closing due to lack of user response and the fact that a fix was pushed.

closed 2020-05-07 by makermelissa