adafruit/Adafruit_Blinka #161: Add frequency property to busio.SPI

caternuson, 2019-10-04:
From CircuitPython API:
https://circuitpython.readthedocs.io/en/4.x/shared-bindings/busio/SPI.html#busio.SPI.frequency

On a CircuitPython board:
```python
Adafruit CircuitPython 4.1.0 on 2019-08-02; Adafruit ItsyBitsy M4 Express with samd51g19
>>> import board
>>> spi = board.SPI()
>>> spi.frequency
250000
>>> 
```

On a Blinka board (RPi):
```python
pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> spi = board.SPI()
>>> spi.frequency
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'SPI' object has no attribute 'frequency'
>>> 
```

ladyada, 2019-10-04:
woops - yeah, you can initialize the frequency but not set/get it after - wanna do that?

caternuson, 2019-10-04:
Just to clarify - should be a read only property.
```python
Adafruit CircuitPython 4.1.0 on 2019-08-02; Adafruit ItsyBitsy M4 Express with samd51g19
>>> import board
>>> spi = board.SPI()
>>> spi.frequency
250000
>>> spi.frequency = 6000000
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'SPI' object has no attribute 'frequency'
>>> 
```
I think setting requires using `configure()`, via `baudrate`:
https://circuitpython.readthedocs.io/en/4.x/shared-bindings/busio/SPI.html#busio.SPI.configure

ladyada, 2019-10-04:
yep!

caternuson, 2019-10-11:
Added with #164

closed 2019-10-11 by caternuson