adafruit/Adafruit_Blinka #163: Add _bleio support

rdagger, 2019-10-05:
Any plans to add BLE support?  I’d like to be able to use CircuitPython BLE code to communicate between Raspberry Pi and nRF52840 Feather.

ladyada, 2019-10-06:
no plans any time soon, bluez is completely different. you would want to try https://learn.adafruit.com/bluefruit-le-python-library instead!

closed 2019-10-06 by ladyada

rdagger, 2019-10-06:
Thanks!  I tried Adafruit_Python_BluefruitLE but it appears to be broken with respect to the latest version of BlueZ that comes preloaded with Raspbian Buster.  I tried patching as recommended in https://github.com/adafruit/Adafruit_Python_BluefruitLE/issues/23 but it just wasn’t stable.

ladyada, 2019-10-06:
yah that's not surprising, if you have any PR reviews or suggestions, let us know - bluez is kinda wretched :/