adafruit/Adafruit_Blinka #165: Jetson Nano blinka library error

cetinkayafaruk, 2019-10-15:
Hi,

I just tried to test blinka libraries on my Jetson Nano. But, I had this error ;

Traceback (most recent call last):
  File "nano_lib_test.py", line 1, in <module>
    import digitalio
  File "/usr/local/lib/python3.6/dist-packages/digitalio.py", line 23, in <module>
    from adafruit_blinka.microcontroller.tegra.t210.pin import Pin
  File "/usr/local/lib/python3.6/dist-packages/adafruit_blinka/microcontroller/tegra/t210/pin.py", line 3, in <module>
    import Jetson.GPIO as GPIO
  File "/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/gpio.py", line 70, in <module>
    model, JETSON_INFO, _channel_data_by_mode = gpio_pin_data.get_data()
  File "/usr/local/lib/python3.6/dist-packages/Jetson/GPIO/gpio_pin_data.py", line 286, in get_data
    revision = f.split('-')[-1]
AttributeError: '_io.TextIOWrapper' object has no attribute 'split'

How can I solve it ?  I made initial setup base on https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano?view=all

makermelissa, 2019-10-15:
Looks like it may be an issue with Jetson.GPIO. @swarren do you have any ideas about this?

swarren, 2019-10-15:
This was a bug in Jetson.IO that's since been fixed. Update to the latest version and the problem should go away.

cetinkayafaruk, 2019-10-15:
Thank you, that worked.

closed 2019-10-15 by cetinkayafaruk