adafruit/Adafruit_Blinka #171: Aaeon Upboard support

FPSychotic, 2019-11-03:
Hi, Could uou confirm if Blinka is supported by Upboard, the are compatible with many RPi HATs as it share the shame gpio interface. To be compatible, I think that internally they used the same system than Blinka, a software layer to correspond different physical layers in the CPu and Chipsets.
If it is not compatible could you add support for this board, and in consequence probably to the incoming Banana PiX.

makermelissa, 2019-11-25:
@FPSychotic, I'm almost certain that it is not supported. Blinka uses PlatformDetect to figure out the processor that it is running on and the Aaeon Upboard uses an Intel Atom x5-z8350 processor and that doesn't appear to be one we have inside of https://github.com/adafruit/Adafruit_Blinka/tree/master/src/adafruit_blinka/microcontroller. I'll leave this issue open for now as a request to support that board.

astumpf, 2021-06-21:
Still not supported? I'm wondering as the Aaeon boards are widely used for robotics application.

ladyada, 2021-06-22:
we're happy to take a PR from someone who owns one - we do not have any plans to do it ourselves.