adafruit/Adafruit_Blinka #172: FT232h only supports SPI mode 0/2, but all are allowed

alexwhittemore, 2019-11-11:
I was trying to prototype with the Max31865 and FT232h but couldn't get it to work, while the Metro M4 Express had no trouble. Then I saw the note [here](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/faq-troubleshooting) and suddenly I remembered I'd been hit by this CPHA=1 issue before and REALLY hate chip designers that use modes =/= 0. 

What popped into my head is [this FTDI appnote](https://www.ftdichip.com/Support/Documents/AppNotes/AN_114_FTDI_Hi_Speed_USB_To_SPI_Example.pdf) that, on page 3, describes SPI modes 0-3 and proceeds to explain that ONLY mode 0 and 2 are supported, because only CPHA=0 is supported by their serial engine. Then I suddenly remembered the stupid looking glitch I just captured at the start of the CLK burst looked the same as the last time this bit me. Like Dej

Anyway, long story short, it looks like @caternuson originally wrote a mode check + ValueError raise. That code actually error'd on SPI mode 2, which per above isn't correct - SPI modes 1 and 4 should error out. But then the check was removed completely:
https://github.com/adafruit/Adafruit_Blinka/commit/cbaee77564e37047937a9c642d13f20b53821845

I propose that the mode check be re-added and a ValueError raised for modes 1+3. Perhaps with a slightly saltier error message along the lines of "SPI Mode <X> Not Supported. The chip you're using is dumb, and unfortunately, FTDI agrees."

alexwhittemore, 2019-11-11:
Just because I went to all the effort and might as well, here's the offending clock waveform. You CAN command the FT232h into CPHA=1, it just doesn't do the right thing and your device won't talk right.
[image]

caternuson, 2019-11-11:
Thought this might resurface. I'll take a look.

caternuson, 2019-11-14:
Here's where the SPI mode 2 confusion came from:
https://eblot.github.io/pyftdi/api/spi.html#pyftdi.spi.SpiController.get_port
but actual code looks more inline with FTDI app note:
https://github.com/eblot/pyftdi/blob/master/pyftdi/spi.py#L332
and of course the FTDI app note should be ultimate answer:
[image]

**But why were the checks removed?**
An early approach (the commit you linked) obtained the FT232H SPI port (controller) in the read/write functions. That was a much cleaner approach since whatever the current SPI port configurations settings were could be used at that time. This was changed to obtaining the port (controller) in `__init__` and keeping it around. Changing the SPI port settings were then done by essentially cheating and directly setting the `_cpol` and `_cpha` members. By doing that, the pyftdi check is unfortunately circumvented:
https://github.com/eblot/pyftdi/blob/master/pyftdi/spi.py#L349

**Why was that done?**
That was driven by the way GPIO works. GPIO control is not a separate item, but is provided through the I2C and SPI controllers. But from a user point of view, it should be a separate item. What does an I2C or SPI port have to do with setting a GPIO pin high/low? So an attempt was made to make it work like that here in Blinka. This required some behind the scenes magic that I was never super happy about. See 5th bullet in PR comment: https://github.com/adafruit/Adafruit_Blinka/pull/158

**Can we still have a check?**
Absolutely. PR'd #173

alexwhittemore, 2019-11-14:
Two lines of fix, three lines of explanatory comment in-line with that code, and 21 lines plus a screenshot of background - you have no idea how much I appreciate that kind of effort for documentation. 

Thank you, what a hero!

caternuson, 2019-11-14:
@alexwhittemore No prob. Want to try the 3.0.1 version when it becomes available:
https://github.com/adafruit/Adafruit_Blinka/releases/tag/3.0.1
and let us know how that works out?

alexwhittemore, 2019-11-14:
I mean, if it "works" I'll run my current code and get immediately yelled at for trying to do a bad thing, but I can certainly see if I do indeed get yelled at :) I'll test and close when I'm back at the bench tonight.

ladyada, 2019-11-15:
plz re-open if still an issue

closed 2019-11-15 by ladyada

alexwhittemore, 2019-11-15:
I have tested, and it does indeed tell me I can't do that! Now that I experience it in person, it seems like an indication of WHY I get the error would be nice. For instance, "The Max31865 lib is requesting this, but I can't!" I'll have to think a little harder about where it might make sense to implement that.

Thanks!