adafruit/Adafruit_Blinka #177: Cant update spidev to 3.4 from 3.3

Padduzaj, 2019-11-22:
When using the max31865 converter with a raspberry pi 4, I cannot use the code as follows:

import time

import board
from busio import SPI
from digitalio import DigitalInOut

import adafruit_max31865
SDI = 10 # Pin19
SDO = 9 # Pin21
CLK = 11 # Pin23
CS = 5 # Pin29

#Initialize SPI bus and sensor
spi = SPI(board.SCLK, MOSI=board.MOSI, MISO=board.MISO)
cs = DigitalInOut(board.D5)
sensor = adafruit_max31865.MAX31865(spi, cs, wires = 3, rtd_nominal = 100, ref_resistor = 430)

temp = sensor.temperature
print('Temperature: %0.3f*C' %temp)
time.sleep(1)

Running this code gives the traceback of:
Traceback (most recent call last):
  File "/home/pi/Python Code/Main code/Temp.py", line 16, in <module>
    sensor = adafruit_max31865.MAX31865(spi, cs, wires = 3, rtd_nominal = 100, ref_resistor = 430)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_max31865.py", line 107, in __init__
    config = self._read_u8(_MAX31865_CONFIG_REG)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_max31865.py", line 123, in _read_u8
    device.write(self._BUFFER, end=1)
  File "/usr/local/lib/python3.7/dist-packages/busio.py", line 170, in write
    return self._spi.write(buf, start, end)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/spi.py", line 54, in write
    self._spi.writebytes2(buf[start:end])
AttributeError: 'SpiDev' object has no attribute 'writebytes2'

After finding out it is because I have spidev 3.3 and writebytes2 is included in spidev 3.4, I cannot upgrade from 3.3 to 3.4. Many attempts have been made and none have worked.

Please help before I lose my mind.

caternuson, 2019-11-22:
See here:
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/faq-troubleshooting#faq-3047724

If that does not work, please paste whatever error you are getting here.

Also - if you have any information as to why version 3.4 did not get installed when you did the Blinka install, please share that as well.

makermelissa, 2019-11-25:
@Padduzaj, were you able to get this working by following @caternuson's suggestion?

Padduzaj, 2019-11-25:
Yes it worked, apologies I was away for a few days without Internet. 
Blinka came with 3.3 installed and then the sudo update and upgrade wouldn't update the version to 3.4. Forcing the install has worked and I appreciate the assistance.

closed 2019-11-25 by Padduzaj

makermelissa, 2019-11-25:
Awesome. Glad to hear.

caternuson, 2019-11-26:
Hopefully this: #182 will fix this for good. Still not sure why I couldn't recreate the issue when trying a Blinka install with a fresh Raspbian. But fine with that being something I was doing wrong or overlooking.

Thanks @makermelissa for that PR!