adafruit/Adafruit_Blinka #179: Raspi 4B not detected

dimk1, 2019-11-23:
Hello,

I'm trying to use the `ads1x15` library with my raspi 4B, but when I run the `blinkatest.py` I get the following error:

```
pi@rpi4:~/readTemp $ python3 blinkatest.py 
Traceback (most recent call last):
  File "blinkatest.py", line 1, in <module>
    import board
  File "/usr/local/lib/python3.7/dist-packages/board.py", line 107, in <module>
    raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported None
```

I tried the `detect.py` script I found in another forum/page and I get the following:
```
pi@rpi4-measure:~/readTemp $ python3 detect.py 
Chip id:  BCM2XXX
Board id:  None
Is this a DragonBoard 410c? False
Is this a Pi 3B+? False
Is this a Pi 4B? False
Is this a 40-pin Raspberry Pi? False
Is this a Raspberry Pi Compute Module? False
Is this a BBB? False
Is this a Giant Board? False
Is this a Coral Edge TPU? False
Is this a SiFive Unleashed?  False
Is this an embedded Linux system? True
Is this a generic Linux PC? False
Is this an OS environment variable special case? False
Raspberry Pi detected.
```

Further info of my system:
```
pi@rpi4:~ $ pip3 list
Package                          Version  
-------------------------------- ---------
Adafruit-Blinka                  3.0.1    
adafruit-circuitpython-ads1x15   2.1.1    
adafruit-circuitpython-busdevice 4.0.0    
Adafruit-GPIO                    1.0.3    
Adafruit-PlatformDetect          1.3.5    
Adafruit-PureIO                  0.2.3    
asn1crypto                       0.24.0   
certifi                          2018.8.24
chardet                          3.0.4    
cryptography                     2.6.1    
entrypoints                      0.3      
idna                             2.6      
keyring                          17.1.1   
keyrings.alt                     3.1.1    
mock                             3.0.5    
pi-ina219                        1.3.0    
pip                              19.3.1   
pycrypto                         2.6.1    
PyGObject                        3.30.4   
python-apt                       1.8.4    
pyxdg                            0.25     
requests                         2.21.0   
rpi-ws281x                       4.2.2    
RPi.GPIO                         0.7.0    
SecretStorage                    2.3.1    
setuptools                       40.8.0   
six                              1.12.0   
spidev                           3.4      
ssh-import-id                    5.7      
sysv-ipc                         1.0.0    
urllib3                          1.24.1   
wheel                            0.32.3   
```

```
pi@rpi4:~ $ cat /proc/cpuinfo 
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: c03112
Serial		: 100000002c91a017
Model		: Raspberry Pi 4 Model B Rev 1.2
```

Ideas?

ladyada, 2019-11-23:
that would stem from here
https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/board.py#L281
looks like we need to add a new revision `c03112` to the end of that list
wanna try that and see if works?

dimk1, 2019-11-24:
Yes, it works. I simply added my revision to the list and code works for
the ne. Thanks for the quick response. Cheers. We can close this.

ladyada, 2019-11-24:
ok please submit a PR so it will be fixed for others!

makermelissa, 2019-11-25:
I submitted a PR to PlatformDetect, so this should now be good.

closed 2019-11-25 by makermelissa